espressif/esp-idf #10081: ESP32-S3 BLE LoadProhibited when calling esp_ble_gatts_add_char() from ESP_GATTS_ADD_CHAR_EVT

spkurtzweil, 2022-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3-Devkit-C1

### Power Supply used.

USB

### What is the expected behavior?

I expect it to add the characteristic, complete the function, and call the GATT callback again with a ADD_CHAR_EVT.

### What is the actual behavior?

Crashes after finishing execution of the callback and before executing the next callback.

### Steps to reproduce.

1. Enable Bluetooth in SDKConfig
2. Add target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-error=missing-braces) to CMakeLists.txt
3. Build, flash, and monitor code. It will happen almost immediately after beginning execution.
4. Data structures to define GATT structure are above app_main, GATT callback is at the bottom of main.c. I will split them into separate .h and .c files once I get this sorted out.
[https://github.com/spkurtzweil/Senior-Design-Project/blob/39ddd83f439d7b89f54378004723c76781c0be72/Microcontroller/code/main/main.c](url)

### Debug Logs.

```plain
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375850
0x40375850: esp_restart_noos at C:/Users/spkur/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1650
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 17:46:30
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 code             factory app      00 00 00010000 00100000
I (84) boot:  3 flight_data      Unknown data     01 06 00110000 0058c000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1c0b8h (114872) map
I (124) esp_image: segment 1: paddr=0002c0e0 vaddr=3fc95470 size=039d0h ( 14800) load
I (128) esp_image: segment 2: paddr=0002fab8 vaddr=40374000 size=00560h (  1376) load
I (130) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=62ea0h (405152) map
I (211) esp_image: segment 4: paddr=00092ec8 vaddr=40374560 size=10f04h ( 69380) load
I (227) esp_image: segment 5: paddr=000a3dd4 vaddr=50000000 size=00010h (    16) load
I (234) boot: Loaded app from partition at offset 0x10000
I (234) boot: Disabling RNG early entropy source...
I (247) cpu_start: Pro cpu up.
I (247) cpu_start: Starting app cpu, entry point is 0x40375288
0x40375288: call_start_cpu1 at C:/Users/spkur/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (225) cpu_start: App cpu up.
I (261) cpu_start: Pro cpu start user code
I (261) cpu_start: cpu freq: 160000000
I (261) cpu_start: Application information:
I (264) cpu_start: Project name:     code
I (269) cpu_start: App version:      d42a3db-dirty
I (274) cpu_start: Compile time:     Oct 30 2022 21:50:01
I (280) cpu_start: ELF file SHA256:  e2adf6ee14e03f7a...
I (286) cpu_start: ESP-IDF:          v4.4.2
I (291) heap_init: Initializing. RAM available for dynamic allocation:
I (298) heap_init: At 3FC9C600 len 00043A00 (270 KiB): D/IRAM
I (305) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (311) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (318) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (325) spi_flash: detected chip: generic
I (329) spi_flash: flash io: dio
I (333) sleep: Configure to isolate all GPIO pins in sleep state
I (340) sleep: Enable automatic switching of GPIO sleep configuration
I (347) coexist: coexist rom version e7ae62f
I (352) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (372) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (382) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (382) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (392) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
flash ID: 123b0c151c80b5c7
I (412) BT_INIT: BT controller compile version [05195c9]
I (412) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (452) system_api: Base MAC address is not set
I (462) system_api: read default base MAC address from EFUSE
I (462) BT_INIT: Bluetooth MAC: 60:55:f9:f5:25:fa

creating service 0
Started serv 0x28
adding char 0
entered add char evt
Added char 0x2a
adding char 1
exited add char evt
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40056f9c  PS      : 0x00060f30  A0      : 0x82027e3e  A1      : 0x3fced2b0
A2      : 0x3fced8d4  A3      : 0x00000000  A4      : 0x00000002  A5      : 0x3fced8d4  
A6      : 0x69744353  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fced280
A10     : 0x3fced8d4  A11     : 0x00000002  A12     : 0x00060120  A13     : 0x00060123  
A14     : 0x69744353  A15     : 0xc0da7363  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x40056f99:0x3fced2b0 |<-CORRUPTED

ELF file SHA256: e2adf6ee14e03f7a
```

xiewenxiang, 2022-10-31:
Hi @spkurtzweil 
I cann't access this URL https://github.com/spkurtzweil/Senior-Design-Project/blob/39ddd83f439d7b89f54378004723c76781c0be72/Microcontroller/code/main/main.c

spkurtzweil, 2022-10-31:
Apologies. Having permissions issues with that repository. Copied the code directly. I'm very inexperienced with github, so sorry in advance for formatting or procedural errors.

xiewenxiang, 2022-10-31:
@spkurtzweil You can paste your code directly into the attachment.

xiewenxiang, 2022-10-31:
The attachment is sample code.

I found a problems in the code you provided, the `attr_value` point will be `NULL`:

```c
esp_attr_value_t sdt_flt_ctrl_value = {
    .attr_max_len = 0x02,
    .attr_len = 0x02,
    .attr_value = 0x0000
};
```

```c
typedef struct
{
    uint16_t attr_max_len;                                  /*!<  attribute max value length */
    uint16_t attr_len;                                      /*!<  attribute current value length */
    uint8_t  *attr_value;                                   /*!<  the pointer to attribute value */
} esp_attr_value_t;
```

[gatts_demo.c.zip](https://github.com/espressif/esp-idf/files/9899804/gatts_demo.c.zip)

spkurtzweil, 2022-10-31:
Gah so simple. I was looking in the completely wrong place.

That fixed it, thanks.

closed 2022-10-31 by spkurtzweil