espressif/esp-idf #10084: Build failure with release/v5.0 (v5.0-beta1-791-g2635a18df2)

oursland, 2022-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-791-g2635a18df2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### What is the expected behavior?

`esp-idf` should build cleanly.

### What is the actual behavior?

Compiler fails due to undeclared symbol `ESP_TASK_PING_STACK`.

### Steps to reproduce.

1. Checkout `esp-idf` at version `v5.0-beta1-791-g2635a18df2`
2. Build a project that has LWIP enabled

### Build or installation Logs.

```plain
[3/422] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/ping/ping.c.obj
FAILED: esp-idf/lwip/CMakeFiles/__idf_lwip.dir/apps/ping/ping.c.obj 
In file included from /Users/joursland/esp/esp-idf/components/lwip/apps/ping/ping.c:58:
/Users/joursland/esp/esp-idf/components/lwip/apps/ping/ping.c: In function 'ping_init':
/Users/joursland/esp/esp-idf/components/lwip/include/apps/ping/ping_sock.h:83:28: error: 'ESP_TASK_PING_STACK' undeclared (first use in this function); did you mean 'ESP_TASK_MAIN_STACK'?
   83 |         .task_stack_size = ESP_TASK_PING_STACK,         \
      |                            ^~~~~~~~~~~~~~~~~~~
/Users/joursland/esp/esp-idf/components/lwip/apps/ping/ping.c:281:30: note: in expansion of macro 'ESP_PING_DEFAULT_CONFIG'
  281 |   esp_ping_config_t config = ESP_PING_DEFAULT_CONFIG();
      |                              ^~~~~~~~~~~~~~~~~~~~~~~
/Users/joursland/esp/esp-idf/components/lwip/include/apps/ping/ping_sock.h:83:28: note: each undeclared identifier is reported only once for each function it appears in
   83 |         .task_stack_size = ESP_TASK_PING_STACK,         \
      |                            ^~~~~~~~~~~~~~~~~~~
/Users/joursland/esp/esp-idf/components/lwip/apps/ping/ping.c:281:30: note: in expansion of macro 'ESP_PING_DEFAULT_CONFIG'
  281 |   esp_ping_config_t config = ESP_PING_DEFAULT_CONFIG();
      |                              ^~~~~~~~~~~~~~~~~~~~~~~
[14/422] Building C object esp-idf/esp_netif/CMakeFiles/__idf_esp_netif.dir/lwip/esp_netif_lwip.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "/opt/homebrew/bin/zsh '-c', 'ninja '" terminated with exit code: 1.
```

### More Information.

As of commit `9948de101d9c22538fd8740c836ec89f40e13b88` the source depends upon the symbol `ESP_TASK_PING_STACK` which is defined in `esp_task.h`, however `esp_task.h` is not included by the source file.

xueyunfei998, 2022-11-01:
hi @oursland 
Which example do you use to compile this problem?

MaxwellAlan, 2024-01-11:
Hi @oursland 

We will close this issue since so long no reply. If still have this issue, welcome to reopen.

closed 2024-01-11 by MaxwellAlan