espressif/esp-idf #10085: pppos_example: Modem Disconnect from PPP Server

ForgetCSX, 2022-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.2-388-g755ce1077d-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

LILYGO TTGO T-Call

### Power Supply used.

USB

### What is the expected behavior?

运行这个例子应该可以获取到IP地址，并且可以连接到mqtt服务器

### What is the actual behavior?

无法获取到IP地址，在这之前就断开了PPP Serve

### Steps to reproduce.

1. 由于这块开发板将SIM800L的电源交给ESP32控制，我首先初始化了SIM800L的电源控制IO，并使能输出
2. 在menuconfig里面选择模块为SIM800L
3. 设置串口IO，这个模块是没有将RTS和CTS引出，所以我将这两个IO设置为0
4. 然后idf.py build & idf.py flash monitor
...

### Debug Logs.

```plain
ELF file SHA256: 621778f89009a651

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4.2-388-g755ce1077d-dirty 2nd stage bootloader
I (27) boot: compile time 11:06:41
I (28) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1922ch (102956) map
I (149) esp_image: segment 1: paddr=00029254 vaddr=3ffb0000 size=02590h (  9616) load
I (153) esp_image: segment 2: paddr=0002b7ec vaddr=40080000 size=0482ch ( 18476) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=64930h (411952) map
I (312) esp_image: segment 4: paddr=00094958 vaddr=4008482c size=09174h ( 37236) load
I (327) esp_image: segment 5: paddr=0009dad4 vaddr=50000000 size=00010h (    16) load
I (334) boot: Loaded app from partition at offset 0x10000
I (334) boot: Disabling RNG early entropy source...
I (347) cpu_start: Pro cpu up.
I (348) cpu_start: Starting app cpu, entry point is 0x40081264
0x40081264: call_start_cpu1 at /home/xiang/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (335) cpu_start: App cpu up.
I (362) cpu_start: Pro cpu start user code
I (362) cpu_start: cpu freq: 160000000
I (362) cpu_start: Application information:
I (367) cpu_start: Project name:     pppos_client
I (372) cpu_start: App version:      1
I (376) cpu_start: Compile time:     Nov  1 2022 11:22:39
I (382) cpu_start: ELF file SHA256:  621778f89009a651...
I (388) cpu_start: ESP-IDF:          v4.4.2-388-g755ce1077d-dirty
I (395) heap_init: Initializing. RAM available for dynamic allocation:
I (402) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (409) heap_init: At 3FFB3F00 len 0002C100 (176 KiB): DRAM
I (415) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (421) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (428) heap_init: At 4008D9A0 len 00012660 (73 KiB): IRAM
I (435) spi_flash: detected chip: generic
I (438) spi_flash: flash io: dio
W (442) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (457) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (467) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (477) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (487) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1797) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (1797) uart: queue free spaces: 30
E (1827) esp-modem: handle line failed
W (1827) pppos_example: Unknown line received: +COPS: 0
I (1837) pppos_example: Module: SIMCOM_SIM800L
I (1837) pppos_example: Operator: 
I (1837) pppos_example: IMEI: 867372059787494
I (1847) pppos_example: IMSI: 460028380727552
I (1847) pppos_example: rssi: 0, ber: 0
I (1857) pppos_example: Battery voltage: 3983 mV
W (1867) sim800: enter ppp mode ok
I (1967) pppos_example: Modem PPP Started
I (1967) pppos_example: PPP state changed event 259
I (1967) pppos_example: PPP state changed event 262
I (1997) pppos_example: PPP state changed event 263
I (2007) pppos_example: PPP state changed event 265
I (2017) pppos_example: PPP state changed event 262
I (8017) pppos_example: PPP state changed event 268
I (8017) pppos_example: PPP state changed event 256
I (8017) esp-netif_lwip-ppp: Connection lost
I (8017) pppos_example: Modem Disconnect from PPP Server

Done
```

### More Information.

模块IO：
[image]

ForgetCSX, 2022-11-15:
> Hi @ForgetCSX
> 
> From the log it seems that you have a very poor signal or incorrectly configured cellular network connection. The device wasn't even able to read the operator's name. Have you correctly set the APN (Access Point Name in menuconfig)?

您好，我将流控引脚接上就可以了，经过多测测试都可以连接到mqtt服务器，看起来问题解决了，流控引脚是必须的嘛？我开始也怀疑是信号差，我接上SIM天线问题依旧如此，接上流控才可以的。

david-cermak, 2022-11-18:
Are you talking about the UART control flow? You can enable it directly in menuconfig for that example and it should normally work. Do you see any trouble getting this to run? (note that you need to connect the wires and set the device -- modem to the appropriate flow control mode).

Something similar what's done here:

https://github.com/espressif/esp-protocols/blob/045009bb828d229e651200532f9dd14ba5db3625/components/esp_modem/examples/modem_console/main/modem_console_main.cpp#L170-L176

(which is a command like `AT+IFC=2,2`, but I think it's easier to preconfigure the settings and save it, so you won't need to set it up every reset)

ForgetCSX, 2022-11-19:
感谢您的回复，我是在这个例子中不明白为什么会断开，这让我无法定位问题，从目前测试来看，应该是串口流控没有设置正确，谢谢了，我将关闭这个问题。

closed 2022-11-19 by ForgetCSX