espressif/esp-idf #10086: [usb host] HUB: Root port reset failed

qazyn951230, 2022-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1498-g2d9e75bf05

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S3 DevKitC-1 N8R8

### Power Supply used.

USB

### What is the expected behavior?

Print logs as the example output.

### What is the actual behavior?

Prints "E (5090) HUB: Root port reset failed".

### Steps to reproduce.

1. Clone esp-idf, `cd examples/peripherals/usb/host/usb_host_lib`
2. `idf.py set-target esp32s3`
3. set `CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE` to 255, according to https://github.com/espressif/esp-idf/issues/7799
4. Build and upload to a esp32s3 board.
5. Connent to a deli(得力) DL-581P(new) printer.

### Debug Logs.

```plain
Executing action: monitor
Running idf_monitor in directory /.../esp-idf/examples/peripherals/usb/host/usb_host_lib
Executing "/.../.espressif/python_env/idf5.1_py3.10_env/bin/python /.../esp-idf/tools/idf_monitor.py -p /dev/cu.wchusbserial54CE0249671 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /.../esp-idf/examples/peripherals/usb/host/usb_host_lib/build/usb_host_lib_example.elf -m '/.../.espressif/python_env/idf5.1_py3.10_env/bin/python' '/.../esp-idf/tools/idf.py' '-p' '/dev/cu.wchusbserial54CE0249671'"...
--- idf_monitor on /dev/cu.wchusbserial54CE0249671 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x16ac
load:0x403c9700,len:0xbc8
load:0x403cc700,len:0x2d64
entry 0x403c98fc
I (25) boot: ESP-IDF v5.1-dev-1498-g2d9e75bf05 2nd stage bootloader
I (25) boot: compile time Nov  1 2022 13:55:40
I (25) boot: chip revision: V001
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : DIO
I (46) boot.esp32s3: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0b86ch ( 47212) map
I (118) esp_image: segment 1: paddr=0001b894 vaddr=3fc92400 size=02e0ch ( 11788) load
I (120) esp_image: segment 2: paddr=0001e6a8 vaddr=40374000 size=01970h (  6512) load
I (127) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=20c80h (134272) map
I (159) esp_image: segment 4: paddr=00040ca8 vaddr=40375970 size=0ca50h ( 51792) load
I (170) esp_image: segment 5: paddr=0004d700 vaddr=50000000 size=00010h (    16) load
I (176) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (190) cpu_start: Pro cpu up.
I (190) cpu_start: Starting app cpu, entry point is 0x4037526c
0x4037526c: call_start_cpu1 at /.../esp-idf/components/esp_system/port/cpu_start.c:146

I (0) cpu_start: App cpu up.
I (205) cpu_start: Pro cpu start user code
I (205) cpu_start: cpu freq: 160000000 Hz
I (205) cpu_start: Application information:
I (208) cpu_start: Project name:     usb_host_lib_example
I (214) cpu_start: App version:      v5.1-dev-1498-g2d9e75bf05
I (221) cpu_start: Compile time:     Nov  1 2022 13:55:28
I (227) cpu_start: ELF file SHA256:  a43428b82c29ce81...
I (233) cpu_start: ESP-IDF:          v5.1-dev-1498-g2d9e75bf05
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC95C48 len 00053AC8 (334 KiB): D/IRAM
I (253) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (260) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (266) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (273) spi_flash: detected chip: generic
I (277) spi_flash: flash io: dio
W (281) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (294) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (315) DAEMON: Installing USB Host Library
I (345) CLASS: Registering Client
E (21885) HUB: Root port reset failed
E (22195) HUB: Root port reset failed
E (22505) HUB: Root port reset failed
E (22815) HUB: Root port reset failed
E (23125) HUB: Root port reset failed
E (23435) HUB: Root port reset failed
```

### More Information.

Executing action: monitor
Running idf_monitor in directory /.../esp-idf/examples/peripherals/usb/host/usb_host_lib
Executing "/.../.espressif/python_env/idf5.1_py3.10_env/bin/python /.../esp-idf/tools/idf_monitor.py -p /dev/cu.wchusbserial54CE0249671 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /.../esp-idf/examples/peripherals/usb/host/usb_host_lib/build/usb_host_lib_example.elf -m '/.../.espressif/python_env/idf5.1_py3.10_env/bin/python' '/.../esp-idf/tools/idf.py' '-p' '/dev/cu.wchusbserial54CE0249671'"...
--- idf_monitor on /dev/cu.wchusbserial54CE0249671 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1808
load:0x403c9700,len:0xd6c
load:0x403cc700,len:0x2fd4
entry 0x403c9928
I (24) boot: ESP-IDF v5.1-dev-1498-g2d9e75bf05 2nd stage bootloader
I (25) boot: compile time Nov  1 2022 14:06:37
I (25) boot: chip revision: V001
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : QIO
I (51) boot.esp32s3: SPI Flash Size : 8MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0b86ch ( 47212) map
I (122) esp_image: segment 1: paddr=0001b894 vaddr=3fc92400 size=02e0ch ( 11788) load
I (125) esp_image: segment 2: paddr=0001e6a8 vaddr=40374000 size=01970h (  6512) load
I (133) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=20c80h (134272) map
I (160) esp_image: segment 4: paddr=00040ca8 vaddr=40375970 size=0ca50h ( 51792) load
I (170) esp_image: segment 5: paddr=0004d700 vaddr=50000000 size=00010h (    16) load
I (177) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (190) cpu_start: Pro cpu up.
I (191) cpu_start: Starting app cpu, entry point is 0x4037526c
0x4037526c: call_start_cpu1 at /.../esp-idf/components/esp_system/port/cpu_start.c:146

I (0) cpu_start: App cpu up.
I (205) cpu_start: Pro cpu start user code
I (205) cpu_start: cpu freq: 160000000 Hz
I (205) cpu_start: Application information:
I (208) cpu_start: Project name:     usb_host_lib_example
I (214) cpu_start: App version:      v5.1-dev-1498-g2d9e75bf05
I (221) cpu_start: Compile time:     Nov  1 2022 14:06:28
I (227) cpu_start: ELF file SHA256:  23ae45fc3b89706d...
I (233) cpu_start: ESP-IDF:          v5.1-dev-1498-g2d9e75bf05
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC95C48 len 00053AC8 (334 KiB): D/IRAM
I (253) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (259) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (266) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (273) spi_flash: detected chip: generic
I (277) spi_flash: flash io: qio
I (281) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (301) DAEMON: Installing USB Host Library
I (331) CLASS: Registering Client
E (2841) HUB: Root port reset failed
E (3151) HUB: Root port reset failed
E (3461) HUB: Root port reset failed
E (3771) HUB: Root port reset failed
E (4081) HUB: Root port reset failed
E (4391) HUB: Root port reset failed
E (4641) HUB: Root port reset failed

yourchanges, 2022-11-01:
same problem

Dazza0, 2022-11-01:
Could you double check that

- D+/D- signals are connected the right way around
- D+/D- signals of the port are connected to the correct GPIOs of the target
- No pull-ups are being applied to D+ or D-

qazyn951230, 2022-11-01:
@Dazza0 I use a standard USB port to connect the board and the printer, so there should be no problem.

[image]
[image]

yourchanges, 2022-11-01:
it works well with the gp-3120 gp-3150  epson and etc usb printer, but with no luck with the deli-581p printer

Dazza0, 2022-11-01:
@qazyn951230 Any chance you could use a logic analyzer/oscilloscope to analyze what's going on with the D+/D- on connection?

chegewara, 2022-11-01:
I see similar message, which is not error in my case, when USB device is esp32-s3 JTAG port and i am pressing reset on it fairly quickly. Im not saying it is related, but may be a clue.

qazyn951230, 2022-11-02:
I could try it, but I don't have the oscilloscope on hand...

qazyn951230, 2022-11-16:
I've given up on tracking this problem as I don't know much about using an oscilloscope and there are other printers available.

closed 2022-11-16 by qazyn951230