espressif/esp-idf #10088: Building for linux fails when explicitly adding component dependencies

ljden, 2022-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Adding `idf_component.yml` will fetch specified dependencies and build

### What is the actual behavior?

Fails to build when non-empty `idf_component.yml` is added

### Steps to reproduce.

```bash
cd /path/to/esp/esp-idf/examples/build_system/cmake/linux_host_app
cat << EOF > main/idf_component.yml 
dependencies:
  example/cmp: "*"
EOF
idf.py reconfigure
```

### Build or installation Logs.

```plain
$ idf.py reconfigure
Executing action: reconfigure
Running cmake in directory /home/luke.dennis/esp/esp-idf/examples/build_system/cmake/linux_host_app/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=linux -DCCACHE_ENABLE=0 /home/luke.dennis/esp/esp-idf/examples/build_system/cmake/linux_host_app"...
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target linux
Processing 2 dependencies:
[1/2] example/cmp (3.3.8)
[2/2] idf (4.4.2)
-- building FREERTOS MOCKS (only task and queue)
CMake Error at /home/luke.dennis/esp/esp-idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'esp_system_protocols_linux'.
Call Stack (most recent call first):
  /home/luke.dennis/esp/esp-idf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  /home/luke.dennis/esp/esp-idf/tools/cmake/build.cmake:233 (__build_expand_requirements)
  /home/luke.dennis/esp/esp-idf/tools/cmake/build.cmake:228 (__build_expand_requirements)
  /home/luke.dennis/esp/esp-idf/tools/cmake/build.cmake:509 (__build_expand_requirements)
  /home/luke.dennis/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/home/luke.dennis/esp/esp-idf/examples/build_system/cmake/linux_host_app/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

-- Configuring incomplete, errors occurred!
See also "/tmp/host_test/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```
```

0xjakob, 2022-11-02:
@ljden Sorry for the inconvenience! The linux target isn't well supported until including IDF v4.4. From v5.0, however, there is much better support and we are running unit tests on applications that also use the component manager. I'm afraid you have to switch to v5.0 to make component manager and linux target work.

Sorry we can't offer a better alternative in this case!

ljden, 2022-11-02:
@0xjakob Any idea what the timeline is for an official release of v5.0? When would stable move from v4.4.2 -> v5.*

0xjakob, 2022-11-03:
@ljden It should be during the next weeks, but I'm personally not so involved in the release process, hence my information is more hearsay, rather than concrete.

ljden, 2022-11-09:
@0xjakob any chance you can point out the relevant commits so I can try patch in a fix in our fork of v4.4.2?

0xjakob, 2022-11-10:
@ljden I'm not sure about everything necessary for component manager. Finding out the necessary commits for Linux target and component manager would be tedious and error-prone due to the dependenies. I suggest relying on the [`https://github.com/espressif/esp-idf/tree/release/v5.0](v5.0 release branch) or [v5.0 beta-1](https://github.com/espressif/esp-idf/releases/tag/v5.0-beta1) for the time being. 

If you try using  [v5.0 beta-1](https://github.com/espressif/esp-idf/releases/tag/v5.0-beta1), does that fix your issue, at least?

I also wish we had a more reliable release schedule.

0xjakob, 2022-12-02:
@ljden [v5.0 has been released](https://github.com/espressif/esp-idf/releases/tag/v5.0).

Does you code work when using the IDF v5.0 release?

ljden, 2022-12-05:
In the process of migrating, but I have just tested this and it seems to work with v5.0. 
Closing as v5.0 is now released as stable

closed 2022-12-05 by ljden

0xjakob, 2022-12-05:
@ljden Happy to hear it works! Please keep us posted if something still breaks.