espressif/esp-idf #10089: In two-way communication, the AP receives the packet at time t1, and then the AP returns the packet at time t2. I need the difference between t1 and t2 to be within the microsecond level

cquptJiang, 2022-11-01:
### Is your feature request related to a problem?

  ESP32, IDF V4.4.1
  The sta sends a packet to the AP, which then sends another packet to the sta after receiving the packet. The size of the difference between t1 and t2 directly affects the error of the carrier phase of the obtained bidirectional CSI. The approach I've tried is based on the many receive callbacks, where I add code to send the packet. Examples are wifi_csi_rx_cb, wifi_promiscuous_rx_cb, and esp_now_recv_cb. However, I found that most of these functions work with a delay of more than a millisecond from the actual timestamp. As a result, I cannot return the packet as soon as I receive it. Part of the output log is given below, and time_csi represents the timestamp at which the receiving callback function wifi_csi_rx_cb starts working.
`time_csi:265.875,
CSI_DATA,155,10:97:bd:e3:c3:d5,-30,11,1,0,0,1,1,0,0,0,1,-93,0,1,0,264790851,0,44,0,256,1,[ 44 -64 2 0 16 -8 17 -7 16 -8 17 -8 16 -8 15 -9 15 -9 14 -9 13 -10 13 -10 13 -10 13 -10 13 -11 12 -10 13 -11 13 -10 13 -10 13 -10 12 -9 12 -9 12 -8 13 -7 13 -7 13 -6 14 -5 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 11 -6 12 -6 12 -5 13 -4 13 -4 14 -4 15 -3 14 -3 15 -2 15 -2 16 -2 16 -2 16 -2 16 -2 16 -2 15 -3 15 -3 15 -3 17 -4 16 -4 16 -5 17 -5 16 -6 17 -7 18 -7 17 -7 0 0 37 3 34 5 34 5 33 3 33 4 33 2 33 1 31 0 31 -3 32 -3 33 -1 32 -2 32 -2 32 -4 30 -5 31 -5 30 -3 32 -5 31 -5 31 -3 30 -3 28 0 27 1 27 3 28 4 27 6 23 9 21 14 0 0 0 0 0 0 0 0 0 0 0 0 0 0 22 -5 23 -1 25 1 26 3 23 5 25 6 24 8 27 8 27 9 27 10 27 10 26 11 27 12 28 13 26 14 28 13 28 12 28 12 29 11 27 9 29 10 32 9 31 7 31 7 32 6 33 8 33 6 34 6 ]
I (266270) ESPnowbase CSI (PAS): received 1
time_csi:267.881,
CSI_DATA,156,10:97:bd:e3:c3:d5,-29,11,1,0,0,1,1,0,0,0,1,-92,0,1,0,266796831,0,44,0,256,1,[ 44 -64 2 0 -25 7 -25 8 -23 8 -24 8 -23 9 -22 10 -22 10 -21 10 -20 11 -21 12 -20 11 -20 11 -20 13 -19 12 -20 12 -20 12 -20 12 -20 12 -19 11 -20 11 -20 8 -20 8 -20 6 -20 5 -20 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -17 7 -18 6 -19 4 -20 3 -20 3 -21 2 -22 2 -22 2 -22 1 -23 0 -23 -1 -23 0 -23 -1 -23 -1 -23 0 -23 1 -23 0 -22 1 -23 2 -23 4 -24 5 -24 5 -24 5 -25 5 -26 6 -25 7 0 0 -49 0 -49 -1 -47 -2 -48 0 -45 1 -47 4 -44 5 -44 5 -44 8 -45 9 -45 8 -43 9 -43 9 -42 10 -44 12 -43 11 -43 9 -44 10 -43 11 -42 9 -43 7 -40 4 -41 2 -40 -1 -39 -3 -38 -5 -37 -10 -33 -15 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -31 10 -33 5 -36 1 -36 -1 -37 -3 -37 -4 -38 -7 -39 -8 -40 -8 -40 -10 -41 -11 -41 -14 -41 -15 -41 -16 -40 -15 -40 -15 -40 -13 -42 -12 -40 -11 -41 -11 -42 -10 -43 -6 -44 -5 -44 -5 -47 -5 -47 -6 -47 -4 -47 -3 ]
I (268276) ESPnowbase CSI (PAS): received 1
time_csi:269.878,
CSI_DATA,157,10:97:bd:e3:c3:d5,-29,11,1,0,0,1,1,0,0,0,1,-92,0,1,0,268794297,0,44,0,256,1,[ 44 -64 2 0 13 20 14 21 13 19 14 19 14 18 15 18 15 17 15 17 15 15 16 14 17 14 17 15 18 15 16 14 17 15 17 14 16 15 15 16 15 15 14 15 13 16 13 15 12 16 11 16 9 17 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 11 13 10 15 9 15 9 17 8 17 8 19 7 19 7 19 6 20 6 20 7 21 7 21 7 21 7 21 7 21 7 20 6 20 7 19 8 20 9 20 10 20 11 21 11 20 13 21 14 21 14 21 0 0 -27 40 -27 38 -29 40 -24 36 -25 39 -22 40 -20 39 -18 39 -19 39 -15 40 -16 41 -15 41 -15 42 -14 43 -15 38 -15 41 -15 40 -16 41 -15 40 -15 38 -17 38 -19 34 -20 34 -23 32 -24 31 -25 30 -28 21 -32 17 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -7 32 -12 32 -17 30 -19 30 -22 27 -25 28 -25 27 -27 28 -29 29 -29 28 -31 27 -31 26 -33 26 -33 27 -34 27 -34 27 -33 29 -33 25 -33 26 -30 26 -31 31 -31 32 -27 34 -30 33 -26 34 -28 36 -28 38 -27 38 ]
I (270273) ESPnowbase CSI (PAS): received 1`
Although I know the actual timestamp from wifi_pkt_rx_ctrl_t, I can't do anything about it at this moment because I don't know where the timestamp comes from (including the precondition it generates), nor can I see the underlying code.
  

### Describe the solution you'd like.

1. Provides an API that is called as soon as data is received(us).
2. Provide a parameter that can be used as a tag value that changes every time a particular packet is received. And i can monitor its value with a while loop, such as
 while(tag){
send(); 
tag=0; }
3.If I know the conditions under which the real timestamp is obtained, I can also do some other operations under the same conditions.

### Describe alternatives you've considered.

  I learned about the ESPNOW communication protocol in the official document, when the ap receives the espnow packet, it will reply with an ACK (the delay is about 10us), so I want to get bidirectional CSI data in this way. However, at present, the sender cannot directly use the ACK of ESPNOW to obtain CSI (I wonder if there is a way to obtain CSI through ACK), and I even do not know whether the sender has received the ACK from the receiver, because I cannot read the ACK without knowing the frame structure of ACK and no other information.

Alvin1Zhang, 2025-03-12:
Thanks for reporting.

zhangyanjiaoesp, 2025-03-12:
@cquptJiang 
1. Is this time `264790851`  referring to the `timestamp` inside `wifi_pkt_rx_ctrl_t`? If so, its explanation is as follows :
https://github.com/espressif/esp-idf/blob/94cfe394fe94af5e53da8398af5b886016e7fc11/components/esp_wifi/include/local/esp_wifi_types_native.h#L64

2. 
> However, at present, the sender cannot directly use the ACK of ESPNOW to obtain CSI (I wonder if there is a way to obtain CSI through ACK)

We have supported to dump 802.11 ACK frame for CSI since 571c99bda20ff13c59eec11bdf3bf895e36fa979. Please update your IDF version to at least v5.1