espressif/esp-idf #10091: Bluetooth Controller: Unexpected Connect Request for existing ACL BR/EDR Secure Connection

mringwal, 2022-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 Thing

### Power Supply used.

USB

### What is the expected behavior?

Pairing with Windows 10 works flawless.

### What is the actual behavior?

After pairing and enabling AES-CCM encryption for an BR/EDR connection, the EPS32 Bluetooth Controller emits a HCI Connection Request event for the actual connection. 

As far as I know, there can only be one BR/EDR ACL link between devices.

### Steps to reproduce.

Run the spp_counter example from the BTstack example folder.https://github.com/bluekitchen/btstack/tree/master/port/esp32

Use a USB Bluetooth Controller that supports BR/EDR Secure Connection. The issue does not occur if BR/EDR Secure Connections are disabled on the ESP32 or on the the remote side.

Pair to the ESP32.

### Debug Logs.

```plain
Please have a look at the full HCI log from the ESP32 with Wireshark attached in the "More Information" block.

...
Jan 01 11:00:36.710  HCI Event        Connection Request - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:36.726  HCI Command      [0409] Accept Connection Request - A4:C3:F0:23:16:93  SEND  
...
Jan 01 11:00:37.239  HCI Event        IO Capability Response - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:37.257  HCI Event        IO Capability Request - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:37.271  HCI Command      [042B] IO Capability Request Reply - A4:C3:F0:23:16:93  SEND  
Jan 01 11:00:37.518  HCI Event        User Confirmation Request - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:37.521  HCI Command      [042C] User Confirmation Request Reply - A4:C3:F0:23:16:93  SEND  
Jan 01 11:00:40.027  HCI Event        Simple Pairing Complete - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:40.038  HCI Event        Link Key Notification - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:40.075  HCI Event        Encryption Change Complete - Encryption Enabled AES-CCM For BR/EDR  RECV  
...
Jan 01 11:00:41.261  HCI Event        Connection Request - A4:C3:F0:23:16:93  RECV  
Jan 01 11:00:41.269  HCI Command      [0409] Accept Connection Request - A4:C3:F0:23:16:93  SEND  
Jan 01 11:00:41.275  HCI Event        Command Status - Accept Connection Request - Command Disallowed  RECV  

The wasn't a disconnect event between the encryption complete and the connection request.
```

### More Information.

The issue was reported by a BTstack user here: https://groups.google.com/g/btstack-dev/c/YhVUENTNWXY

I haven't been able to reproduce it yet. With a CYW20704 Controller (which supports BR/EDR Secure Connections) plugged into the virtual Windows 10 machine, I couldn't get it to setup a BR/EDR Secure Connection. 

Could you check the Controller code to see, if there's any condition in which a Connection Request might be emitted although a connection already exists?

[teraterm_intel3.pklg.zip](https://github.com/espressif/esp-idf/files/9910597/teraterm_intel3.pklg.zip)

BetterJincheng, 2022-11-02:
@mringwal This is similar to one of our previous issues, Please test with latest v4.4.3.

mringwal, 2022-11-02:
@BetterJincheng  Thanks for your suggestion and sorry for the not testing with current before. The user tried with esp-idf master and the new log don't show the unexpected connection request.

closed 2022-11-02 by mringwal