espressif/esp-idf #10093: Please explain ledc_cb_register‘s callback return bool mean in LEDC controller.

WangShuoran, 2022-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I try to use the ledc example ledc_fade with ESP32C3 and IDF v4.4.2, I find the ISR function:
```c
static bool cb_ledc_fade_end_event(const ledc_cb_param_t *param, void *user_arg) {
    portBASE_TYPE taskAwoken = pdFALSE;
    if (param->event == LEDC_FADE_END_EVT) {
        SemaphoreHandle_t counting_sem = (SemaphoreHandle_t)user_arg;
        xSemaphoreGiveFromISR(counting_sem, &taskAwoken);
    return (taskAwoken == pdTRUE);
}
```
I don't know return `return (taskAwoken == pdTRUE);` mean. Exactly, I don't know return true or false will happen what.

Some information I searched:
https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.2/esp32c3/api-reference/peripherals/ledc.html
https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0-beta1/esp32c3/api-reference/peripherals/ledc.html

suda-morris, 2022-11-03:
The return value indicates whether a high-priority task has been woken up by the callback function. We will explain this in the doc, thanks for reporting!

WangShuoran, 2022-11-07:
Thank you. I think if only use `return taskAwoken;` can let us understand the return mean.

closed 2022-11-11 by espressif-bot