espressif/esp-idf #10094: Jenkins SSL: CERTIFICATE_VERIFY_FAILED for api.components.espressif.com

ddomnik, 2022-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

01d014c42def8d0c19e1ce55c07de6761e092ffa

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Resolving components managed by idf component manager.

### What is the actual behavior?

SSL: CERTIFICATE_VERIFY_FAILED error for api.components.espressif.com

### Steps to reproduce.

1. Set up jenkins environment
2. Add managed components
3. build

### Build or installation Logs.

```plain
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/XXX/.espressif/python_env/idf5.0_py3.10_env/bin/python
Checking if Python packages are up to date...
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt
Destination: /home/XXX/.espressif/espidf.constraints.v5.0.txt.tmp

0%
100%
Done
Python requirements are satisfied.
Constraint file: /home/XXX/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /home/XXX/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/XXX/.espressif/python_env/idf5.0_py3.10_env/bin/python
Added the following directories to PATH:

	[...]
	
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

-- Found Git: /usr/bin/git (found version "2.17.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/XXX/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/XXX/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
[... 100 lines cut ...]
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/__main__.py", line 3, in <module>
      main()
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py", line 126, in main
      args.func(args)
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py", line 39, in prepare_dep_dirs
      ComponentManager(args.project_dir).prepare_dep_dirs(
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_manager/core.py", line 404, in prepare_dep_dirs
      downloaded_component_paths = download_project_dependencies(
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_manager/dependencies.py", line 128, in download_project_dependencies
      download_paths = fetcher.download()
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_tools/sources/fetcher.py", line 51, in download
      return self.source.download(self.component, self.managed_path)
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_tools/sources/web_service.py", line 129, in download
      component_manifest = self.api_client.component(component.name, component.version)
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 225, in component
      response = self._base_request(
    File "/home/XXX/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 178, in _base_request
      raise APIClientError('HTTP request error')

  idf_component_tools.api_client_errors.APIClientError: HTTP request error
```
```

kumekay, 2022-11-04:
Hi @ddomnik 
it looks like the root certificates of your distro are outdated. What distribution are you using in your CI?

Please try upgrading `ca-certificates` package. 

The source of the problem is explained here https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

ddomnik, 2022-11-04:
> Hi @ddomnik it looks like the root certificates of your distro are outdated. What distribution are you using in your CI?
> 
> Please try upgrading `ca-certificates` package.
> 
> The source of the problem is explained here https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

Thank you!
Was just looking into the root certificate / x3 expiration as cause before I noticed that you already replied. But now I am sure.

closed 2022-11-04 by ddomnik

ddomnik, 2022-11-08:
For others that face the same problem:

In fact, it was not the ubuntu distribution that caused that error. It was python itself. Idf.py uses urllib3 which uses certifi's certificate bundle. 
You can tell python to use the systems certificates by adding this env 
`export REQUESTS_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt`

@kumekay you may want to consider using the systems' bundle as well, or tell idf.py to update pythons bundle once in a while.

kumekay, 2022-11-08:
@ddomnik Thank you for the comment, 
could you please help me to reproduce your issue? What version of what distro are you using (or what docker image?) And how did you install esp-idf there?

ddomnik, 2022-11-09:
@kumekay I am sorry, I don't have free access to our build server. The distro was an Ubuntu (16.04. or 18.04) VM. While setting IDF up we had some problems installing virtualenv beacause ensurepip was not available, but got it running somehow. So maybe python 3.4 was installed.