espressif/esp-idf #10095: ESP-IDF Littlefs + esp_vfs_readdir

ifoxbr, 2022-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Can someone explain to me what this routine does? When debugging my program, the debug always stops at this routine. I use littlefs, does anyone know what I'm doing wrong and how to avoid this error?

```
struct dirent* esp_vfs_readdir(DIR* pdir)
{
    const vfs_entry_t* vfs = get_vfs_for_index(pdir->dd_vfs_idx);
    struct _reent* r = __getreent();
    if (vfs == NULL) {
       __errno_r(r) = EBADF;
        return NULL;
    }
    struct dirent* ret;
    CHECK_AND_CALLP(ret, r, vfs, readdir, pdir);
    return ret;
}
```

dobairoland, 2022-11-03:
Calls to `readdir()` are redirected to `esp_vfs_readdir()` which will redirect the call to readdir implementation of the driver. It decides which driver to use, i.e. fatfs if the directory is in fatfs. 

You can find more information about this topic on this site: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html

dobairoland, 2022-11-03:
The callback for littlefs is registered here: https://github.com/joltwallet/esp_littlefs/blob/master/src/esp_littlefs.c#L255

ifoxbr, 2022-11-03:
Thank you Dobairoland

I´m using Littlefs and instantiate as

```
     esp_vfs_littlefs_conf_t conf = 
    {
        .base_path = "/littlefs",
        .partition_label = "littlefs", // este nome é o Name do arquivo "partition.csv"
        .format_if_mount_failed = true,
        .dont_mount = false,
    };
```
Regards