espressif/esp-idf #10096: CoreDump (IllegalInstructionCause) when rtc_clk_cal is called

ztefanjo, 2022-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2 with https://glab.espressif.cn/technical_support/husqvarnagroup/esp-idf-patch

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

No core dumps

### What is the actual behavior?

The attached coredumps has started to occur, possibly after we raised max CPU frequency to 240 MHz (used to be 80 MHz in combination when light sleep).
IllegalInstructionCause happens inside rtc_clk_cal which we do not have source code to.
A variant of the core dump has instead InstFetchProhibitedCause and no readable source addresses for the IDLE task.

[core-dump_51.1_Wifi-BLE-App_43.9-build-36_221102-070001.elf.txt](https://github.com/espressif/esp-idf/files/9921405/core-dump_51.1_Wifi-BLE-App_43.9-build-36_221102-070001.elf.txt)

[core-dump_51.1_Wifi-BLE-App_43.9-build-36_221101-130139.elf.txt](https://github.com/espressif/esp-idf/files/9921420/core-dump_51.1_Wifi-BLE-App_43.9-build-36_221101-130139.elf.txt)

### Steps to reproduce.

Happens on our test fleet that is harvested for core dumps automatically.
From the core dump, it can be seen that it happens when entering LightSleep.

### Debug Logs.

```plain
#0  0x40002547 in ?? ()
#1  0x4037df06 in wait_dig_dbias_valid (rtc_cycles=2) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_hw_support/port/esp32s3/rtc_clk.c:134
#2  rtc_clk_cpu_freq_to_xtal (freq=10, div=4) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_hw_support/port/esp32s3/rtc_clk.c:483
#3  0x4037e091 in rtc_clk_cpu_freq_set_config_fast (config=0x3fcf473c) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_hw_support/port/esp32s3/rtc_clk.c:449
#4  0x40379e32 in do_switch (new_mode=PM_MODE_LIGHT_SLEEP) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_pm/pm_impl.c:508
#5  esp_pm_impl_switch_mode (mode=<optimized out>, lock_or_unlock=<optimized out>, now=<optimized out>) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_pm/pm_impl.c:399
#6  0x40379c70 in esp_pm_lock_release (handle=0x3fcf3154) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_pm/pm_locks.c:151
#7  0x4037f4e0 in esp_pm_impl_idle_hook () at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_pm/pm_impl.c:794
#8  0x403804b4 in esp_vApplicationIdleHook () at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/esp_system/freertos_hooks.c:60
#9  0x40381ccb in prvIdleTask (pvParameters=<optimized out>) at D:/Jenkins/workspace/Releases/Release_43/51.1_Wifi-BLE-App/Ext/esp-idf/components/freertos/tasks.c:3973
```

### More Information.

Max frequency is configured to 240 MHz and Min frequency is 10 MHz. We are using external 32 kHz XTAL for bluetooth.

wmy-espressif, 2022-11-04:
Hi @ztefanjo , thank you for the detailed information. This issue is likely to be related to SoC power management. 

The backtrace shown in the coredump files are dubious, the address and corresponding functions should not occur in the chain of function call. We doubt that the issue may be hardware related.

I would like to ask what is the possibility of this issue, how many pieces of ESP32-S3 chips do you use in your test, and what is the occurance ratio of this issue?

ztefanjo, 2022-11-06:
Hi @wmy-espressif,
Units updating to a later SW where we increased the max frequency when allowing light sleep seem to get these problems several times a day. But there may be other changes also affecting, so hard to say what is triggering this error. But maybe the devices can sleep slightly longer with a higher max frequency (they get their tasks done quicker).

SoC power management as you suggest could be a reason. I read in [gpio-wakeup-light-sleep-only](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32s3/api-reference/system/sleep_modes.html?highlight=esp_sleep_pd_config#gpio-wakeup-light-sleep-only) that:

> Before entering light sleep mode, check if any GPIO pin to be driven is part of the VDD_SPI power domain. If so, this power domain must be configured to remain ON during sleep. For example, on ESP32-WROOM-32 board, GPIO16 and GPIO17 are linked to VDD_SPI power domain. If they are configured to remain high during light sleep, the power domain should be configured to remain powered ON. This can be done with [esp_sleep_pd_config()](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32s3/api-reference/system/sleep_modes.html?highlight=esp_sleep_pd_config#_CPPv419esp_sleep_pd_config21esp_sleep_pd_domain_t21esp_sleep_pd_option_t):

We are currently not doing any call to esp_sleep_pd_option and I don't quite understand if this could be the root of our problem. Are we required to call `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON);` to get proper SPI functionality during light sleep? We are indeed using ESP32-S3-WROOM-1-N4R2. We do also use some GPIO, mainly a UART and some digital input/output.

Could there be other configs that are required to make light sleep work properly together with ESP32-S3-WROOM? Like the ones listed in [sleep-config](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#sleep-config)? Please advice.

ztefanjo, 2022-11-10:
@wmy-espressif : We have tried `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON)` without success. Still the same error.

lennyJL, 2022-11-11:
Hi @ztefanjo, I am looking into this issue, can you provide the probability of this issue? This will help us analyze the cause of this issue.

ztefanjo, 2022-11-11:
Hi @lennyJL ,
Status right now (might change):
Of some reason, this seem to be related to individuals. We have one device that gives this issue one to several times a day and another device that has given it twice so far (since approx last week). Our other devices seem to be free from the error so far. So either some difference in HW, in the config of our SW or in how the devices are used. The users of these two devices use Wifi more than many other users, could be a reason.

As we have not seen the error on the problematic device before, it is likely related to some change in SW we have done recently. I have a list of eight suspected commits and we are trying to find which one of them that causes the issue (takes some time to build and let the user update his machine + another day or two to know for sure if the error exist or not).
* One still suspected commit is the change in clock frequency mentioned above. We changed from 80 to 240 MHz max frequency.
* Another suspected commit (which is the one we are testing right now) is that we lowered CONFIG_BT_ACL_CONNECTIONS from default to 3. 
* The rest of the commits are more related to how we use the BLE API and perhaps therefore less suspect for giving this error.

I will keep you updated when I know more.

lennyJL, 2022-11-14:
Hi @ztefanjo Can you please do some tests:

- **test1**: change `#define DIG_DBIAS_240M      RTC_CNTL_DBIAS_1V25` to `#define DIG_DBIAS_240M      RTC_CNTL_DBIAS_1V30`, the macro is in esp32s3's `rtc.h`
- **test2**: change `#define DIG_DBIAS_XTAL      RTC_CNTL_DBIAS_1V10` to `#define DIG_DBIAS_XTAL      RTC_CNTL_DBIAS_1V20`, the macro is in esp32s3's `rtc.h`
- **test3**: replace `wait_dig_dbias_valid(2);` with `esp_rom_delay_us(200);`(esp32s3's `rtc_clk.c`)
- **test4**: all changes in `test1`, `test2` and `test3`

And also, If you have an oscilloscope, can you monitor whether the power supply of your board is stable?

ztefanjo, 2022-11-14:
Hi @lennyJL ,
OK, we will do the tests you suggest, starting with test1. It will take some days for each test, depending on the frequency of errors and time to get the SW to the device under test. We will also measure the power supply as you suggest.
Reverting the CONFIG_BT_ACL_CONNECTIONS change had no effect on this error.

ztefanjo, 2022-11-14:
@lennyJL , Are there any fixes in the new v4.4.3 that could solve this error? Is "PSRAM: Fixed 40MHz PSRAM CS signal glitch issue on ESP32" affecting also ESP32S3? We are using the default 40MHz (but planning to change).

ztefanjo, 2022-11-18:
Hi @lennyJL , thanks for following up.
We have been running test1 since Tuesday morning and have not seen the error yet so it looks very promising. Before applying the test1 patch, the error happened at least once a day. My plan is to let it keep on running over the weekend to know for sure that test1 solves the issue. If it turns out that test1 works, do you still want us to continue with the other tests? Are there any risks with introducing these patches?
While running test1, we have however seen one other core dump on one occasion that possibly could be related to the test1 patch, see 
[attachement](https://github.com/espressif/esp-idf/files/10039497/core-dump_51.1_Wifi-BLE-App_pipeline_221114-221756_221115-084435.elf.txt). This dump is not complete, indicating that the chip is not feeling well at all. We do get other similar crashes from time to time.

lennyJL, 2022-11-20:
Hi @ztefanjo. If test1 works, I hope you can try `test2`. Test3 and test4 are no longer need.

What `test1` does is boost the voltage of the digital regulator while the chip is running at 240MHz. It is not difficult to understand that the chip needs higher voltage when running at higher frequency. Due to the inconsistency of semiconductor manufacturing, some chips have relatively good physique, and there are also **very very very very very** few chips with very poor physique. This means that **very very very very very** few chips that require a higher voltage(higher than default value in IDF) when running at 240MHz. It is worth noting that the voltage is not the higher the better. We will assess the risks of the boost if `test1` works.

Don't worry that this issue will affect your mass production plan, because we have adopted a more advanced voltage regulation method for esp32s3 recently(s3 may not have similar issues anymore).

ztefanjo, 2022-11-25:
@lennyJL : To give you an update, we have been running test2 since Wednesday. There have been snow now in Sweden so a little colder than before and the automower is not able to run in the snow. So far, we have not seen the rtc_clk_cal error with test2 either, but there have been two other similar coredumps that we have not seen before:
[core-dump_51.1_Wifi-BLE-App_H880794_221122-072952_221123-161903.elf.txt](https://github.com/espressif/esp-idf/files/10091758/core-dump_51.1_Wifi-BLE-App_H880794_221122-072952_221123-161903.elf.txt)
[core-dump_51.1_Wifi-BLE-App_H880794_221122-072952_221125-101700.elf.txt](https://github.com/espressif/esp-idf/files/10091762/core-dump_51.1_Wifi-BLE-App_H880794_221122-072952_221125-101700.elf.txt)
Crash in the Wifi task as you see. Could these be caused by the test2 patch. And could they appear instead of the rtc_clk_cal error?
The plan is to let test2 keep on running during the weekend.

ztefanjo, 2022-12-01:
Hi @lennyJL ,
test2 has now been running for about two weeks, but test conditions are a bit different than before with some snow some days and colder temperatures.
The original coredump has not been reported on test2, but we have instead gotten a new coredump variant that have been reported (with some variations) five times (but may have occurred more frequently as there have been some problems with our service that fetches the dumps). They all have in common that the esp_timer task executes ieee80211_timer_process:
[ieee80211_timer_process.zip](https://github.com/espressif/esp-idf/files/10131087/ieee80211_timer_process.zip)

Can this new error be explained by the lower DBIAS? How do we proceed from here? Is test1 a safe patch to apply on all devices?

lennyJL, 2022-12-02:
Hi @ztefanjo. Just from the crash log, I am not quite sure whether this is caused by DBIAS, but it's very possible. And according to my colleague's conclusion, `test1` is not a safe patch to apply on all devices(it may affect RF calibration).

Based on discussions with our digital team colleagues, I hope you can do some more test:

- **test5**: change #define DIG_DBIAS_XTAL RTC_CNTL_DBIAS_1V10 to #define DIG_DBIAS_XTAL RTC_CNTL_DBIAS_1V25, the macro is in esp32s3's rtc.h. This test aims to further increase DBIAS.
- The previously mentioned **test3**. I'm so sorry I didn't let you go ahead with this test. This test aims to test for instability during DBIAS regulation. Since `esp_rom_delay_us` is a very simple implementation.
- **test6**: all changes in `test5` and `test3`.
- **test7**: apply this change:
   ```
--- a/components/esp_hw_support/port/esp32s3/rtc_clk.c
+++ b/components/esp_hw_support/port/esp32s3/rtc_clk.c
@@ -474,13 +474,6 @@ void rtc_clk_cpu_freq_set_xtal(void)
 void rtc_clk_cpu_freq_to_xtal(int freq, int div)
 {
     ets_update_cpu_frequency(freq);
-    /* set digital voltage for different cpu freq from xtal */
-    if (freq <= 2) {
-        REGI2C_WRITE_MASK(I2C_DIG_REG, I2C_DIG_REG_EXT_DIG_DREG, DIG_DBIAS_2M);
-    } else {
-        REGI2C_WRITE_MASK(I2C_DIG_REG, I2C_DIG_REG_EXT_DIG_DREG, DIG_DBIAS_XTAL);
-    }
-    wait_dig_dbias_valid(2);
     /* Set divider from XTAL to APB clock. Need to set divider to 1 (reg. value 0) first. */
     REG_SET_FIELD(SYSTEM_SYSCLK_CONF_REG, SYSTEM_PRE_DIV_CNT, 0);
     REG_SET_FIELD(SYSTEM_SYSCLK_CONF_REG, SYSTEM_PRE_DIV_CNT, div - 1);
@@ -488,6 +481,13 @@ void rtc_clk_cpu_freq_to_xtal(int freq, int div)
     /* switch clock source */
     REG_SET_FIELD(SYSTEM_SYSCLK_CONF_REG, SYSTEM_SOC_CLK_SEL, DPORT_SOC_CLK_SEL_XTAL);
     rtc_clk_apb_freq_update(freq * MHZ);
+    /* set digital voltage for different cpu freq from xtal */
+    if (freq <= 2) {
+        REGI2C_WRITE_MASK(I2C_DIG_REG, I2C_DIG_REG_EXT_DIG_DREG, DIG_DBIAS_2M);
+    } else {
+        REGI2C_WRITE_MASK(I2C_DIG_REG, I2C_DIG_REG_EXT_DIG_DREG, DIG_DBIAS_XTAL);
+    }
+    esp_rom_delay_us(200);
 }
   ```
`test7` aims to adjust the order of DBIAS reduction and frequency reduction. Since `reduce frequency first, then reduce DBIAS` is a safer order.

lennyJL, 2022-12-02:
Besides, I highly recommend using our new s3 chips(with new DBIAS param in **efuse**, and with new DBIAS regulation method in IDF branch **v4.4.3** and later). You can contact our `business department` to get the new s3 chips.

ztefanjo, 2022-12-02:
Hi @lennyJL ,
Thanks for the reply.
Regarding test 3, 5, 6 and 7: Is it the same with these as with test1 that they are not safe or recommended to apply on all devices? If so, I think we can skip them, especially if it is as you say that the problem only affects our first samples. We are getting close to wintertime here in Huskvarna and if we want to run all these tests, we must likely arrange another setup for the device (it is currently running in a garden, but there is no more grass to be cut and it might be snow soon).

> Besides, I highly recommend using our new s3 chips(with new DBIAS param in efuse, and with new DBIAS regulation method in IDF branch v4.4.3 and later). 
* Will all new S3 chips have this new DBIAS param in efuse or do we need to change how we order our chips?
* We will likely not switch to v4.4.3 on customer releases for quite some time as the testing for 2023 releases are done (we typically release automower SW only once per year). 
    * Will new S3 chips with DBIAS param in efuse work with v4.4.2 of the IDF?
    * Will older S3 chips without DBIAS param in efuse work with v4.4.3?
    * Do we still risk getting these kinds of crashes also for new S3 chips if we do not update to v4.4.3?
    * Does v4.4.3 require the DBIAS param in efuse?

lennyJL, 2022-12-02:
Hi @ztefanjo,

> Is it the same with these as with test1 that they are not safe or recommended to apply on all devices?

**test7** and **test3** are safe. **test5** and **test6** are used to help analyze the root cause. If you have some limits on doing more tests, you can do test7 first. Make some supplements: **when doing test7, the min CPU frequency should be set to 40M(DFS range: 40M~240M)**.

> Will all new S3 chips have this new DBIAS param in efuse or do we need to change how we order our chips?

Yeah, S3 have new DBIAS param in efuse(espressif's engineers will get the job done). You can contact our business department with this issue link(https://github.com/espressif/esp-idf/issues/10096), I communicated with my colleagues in business department, thet know this issue.

v4.4.3 need the DBIAS param in efuse. So:

> Will new S3 chips with DBIAS param in efuse work with v4.4.2 of the IDF?

no(no software in v4.4.2 read param from efuse)

> Will older S3 chips without DBIAS param in efuse work with v4.4.3?

no(In this case the default voltage parameters will be used)

> Do we still risk getting these kinds of crashes also for new S3 chips if we do not update to v4.4.3?

If you are currently inconvenient to rebase to 4.4.3, you can consider using `patch` to support the new voltage regulation method in v4.4.2.

lennyJL, 2022-12-02:
@ztefanjo

Your understanding is mostly correct. A little supplementary explanation: we have a stricter chip screening for s3 than before.

ztefanjo, 2022-12-06:
@lennyJL , we have just decided to include v4.4.3 in our first release, so no `patch` is required. `test7` has now been deployed on the troublesome mower.

lennyJL, 2023-01-04:
Hi @ztefanjo, any new feedback about `test7` or `new s3 chip with v4.4.3`?

ztefanjo, 2023-01-04:
Hi @lennyJL , 
Thanks for checking, I am sorry that I have not kept you up-to-date.

Unfortunately, winter in Sweden has made further testing a bit cumbersome and we have also had some problems with our server that collects core dumps from the test fleet. The DUT ran `test7` for about 2 weeks, and we detected no related core dump during that time, but we also had problems with the server during that time. After that, we had to switch to another FW with v4.4.3 on that device and after that it has been running fine with no core dumps indicating DBIAS problems. But due to the winter, the mower is not running at all in the same way as it were when having the problems. 

The plan is to switch the electronics board on the mower and continue the testing indoors in a more controlled way and focused only on this problem. Hopefully, we will then also get some shorter feedback loop. On the other hand, it might be that the error is related to temperature, so we will see.

It is however a good sign that it seems to be working with v4.4.3. Could it be that it is a chip that already has the calibration values stored in efuses? Or perhaps other changes in v4.4.3 that makes it more stable? It could of course also be changes in the test environment that makes the error to not appear.

ztefanjo, 2023-01-19:
@lennyJL :
Just to give you an update: I have now been running the previously failing board with the original failing SW (to see if I can reproduce) on my desk for a week without any problems what so ever. I suspect either a relation to temperature or use-case. I will try to set up a test in a colder temperature to see if I can reproduce.

lennyJL, 2023-01-30:
Hi @ztefanjo. Synchronize some information with you:
Another customer have reported similar issue, and changes in `test7` makes sense.

> Or perhaps other changes in v4.4.3 that makes it more stable?
`v4.4.3` has some improvements to frequency switching, even though there is no latest voltage parameter in eufuse:
1. use `esp_rom_delay_us` instead of `wait_dig_dbias_valid`
2. synchronous regulation of `RTC regulator` and `DIG regulator`

These changes makes chip more stable when do frequency switching.

ztefanjo, 2023-01-30:
Hi @lennyJL ,
An update from my side: 
During last two weeks, I have been running the failing board outside in my garage. With the original SW, the coredump repeated itself after a few days, acknowledging that the error is temperature dependent.
After that, I have been running `test7` for a little over a week and have not seen the error since then.
Two other core-dumps though. Probably not related to the DBIAS:
[51.1_Wifi-BLE-App_H880794_221202-152421_watchdog_during_BT.txt](https://github.com/espressif/esp-idf/files/10533434/51.1_Wifi-BLE-App_H880794_221202-152421_watchdog_during_BT.txt)
[51.1_Wifi-BLE-App_H880794_221202-152421.elf.txt](https://github.com/espressif/esp-idf/files/10533435/51.1_Wifi-BLE-App_H880794_221202-152421.elf.txt)

`v4.4.3` working better could perhaps also be explained by that the chip has the e-fuse calibration data programmed (but not used with `v4.4.2`)? Is there a way to check this? Shall we run "vanilla v.4.4.3" for a week more as the next test to confirm that the error is not repeating with this? If that seems to work, maybe no patch is required.

lennyJL, 2023-01-30:
@ztefanjo
> v4.4.3 working better could perhaps also be explained by that the chip has the e-fuse calibration data programmed (but not used with v4.4.2)? Is there a way to check this?

I'm not sure about the exact starting time to use the programmed efuse. However, I think if your chip can not read dbias data from efuse, you can get the log containing `efuse read fail, xxx`.

> Shall we run "vanilla v.4.4.3" for a week more as the next test to confirm that the error is not repeating with this?

In addition to improving the stability of frequency switching, `v4.4.3` also includes other bugfixes. I recommend you to use this version to do more test.

ztefanjo, 2023-02-06:
Hi @lennyJL ,
The failing board has now been running v4.4.3 for a week without any of clock-related crashes. I have also confirmed that it does not have the calibration data programmed into its e-fuses. So, at least for this individual, one of the changes you mentioned above must have helped.
If you don't want to work further with the patches you provided (they may make things even more stable?), it is fine from our side to close this issue. Thanks for your support!

Alvin1Zhang, 2023-03-03:
Thanks for reporting and sharing updates, feel free to reopen if the issue still happens. Thanks.

closed 2023-03-03 by Alvin1Zhang