espressif/esp-idf #10098: QRcode not readable

diplfranzhoepfinger, 2022-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### Development Kit.

Atom M5

### Power Supply used.

USB

### What is the expected behavior?

QR-Code: readable.

### What is the actual behavior?

QR-Code: not readable.
however: only the QR on the Console not readable. 
the QR-Code 
[link](https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_6CB5C4","pop":"abcd1234","transport":"ble"})
readable fine. 

### Steps to reproduce.

1. Step build example https://github.com/espressif/esp-idf/tree/master/examples/provisioning
2. Step Read the QR-Code: not readable.

### Debug Logs.

```plain
Executing action: monitor
Running idf_monitor in directory /home/franz/git/hr/LOGIBUS_integration_Hutschienenmoped/Application
Executing "/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/franz/esp-idf-v4.4.2/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 3 --print_filter  /home/franz/git/hr/LOGIBUS_integration_Hutschienenmoped/Application/build/Application.elf -m '/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/franz/esp-idf-v4.4.2/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7168
load:0x40078000,len:15580
load:0x40080400,len:4572
0x40080400: _init at ??:?

entry 0x400806c4
I (60) HOOK: This hook is called BEFORE bootloader initialization
I (35) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (35) boot: compile time 20:50:46
I (35) boot: chip revision: 3
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 8MB
I (57) boot: Enabling RNG early entropy source...
I (62) HOOK: This hook is called AFTER bootloader initialization
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 0036b000
I (103) boot:  3 storage          Unknown data     01 82 0037b000 000f0000
I (110) boot: End of partition table
I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4e420h (320544) map
I (211) esp_image: segment 1: paddr=0005e448 vaddr=3ffbdb60 size=01bd0h (  7120) load
I (214) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=184b8ch (1592204) map
I (656) esp_image: segment 3: paddr=001e4bb4 vaddr=3ffbf730 size=03fc8h ( 16328) load
I (661) esp_image: segment 4: paddr=001e8b84 vaddr=40080000 size=1e9c0h (125376) load
I (702) esp_image: segment 5: paddr=0020754c vaddr=50000000 size=00010h (    16) load
I (718) boot: Loaded app from partition at offset 0x10000
I (718) boot: Disabling RNG early entropy source...
I (729) psram: This chip is ESP32-PICO-V3-02
I (729) spiram: Found 16MBit SPI RAM device
I (730) spiram: SPI RAM mode: flash 80m sram 80m
I (735) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (742) cpu_start: Pro cpu up.
I (746) cpu_start: Starting app cpu, entry point is 0x40081918
0x40081918: call_start_cpu1 at /home/franz/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (1008) spiram: SPI SRAM memory test OK
I (1018) cpu_start: Pro cpu start user code
I (1018) cpu_start: cpu freq: 240000000
I (1018) cpu_start: Application information:
I (1021) cpu_start: Project name:     Application
I (1027) cpu_start: App version:      513c8fb-dirty
I (1032) cpu_start: Compile time:     Nov  2 2022 20:50:44
I (1038) cpu_start: ELF file SHA256:  42a015f3884a852a...
I (1044) cpu_start: ESP-IDF:          v4.4.2-dirty
I (1050) heap_init: Initializing. RAM available for dynamic allocation:
I (1057) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1063) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1070) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1076) heap_init: At 3FFC9930 len 000166D0 (89 KiB): DRAM
I (1082) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1088) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1095) heap_init: At 4009E9C0 len 00001640 (5 KiB): IRAM
I (1101) spiram: Adding pool of 1998K of external SPI memory to heap allocator
I (1110) spi_flash: detected chip: generic
I (1114) spi_flash: flash io: qio
I (1127) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1130) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
isobus_sample!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 3, 8MB embedded flash
Minimum free heap size: 2220691 bytes
CONFIG_FREERTOS_HZ: 1000 must be 1000 !!!! (200Hz at least for 5ms )
app_main starting 
app_main 
I (1183) wifi:wifi driver task: 3ffd81e8, prio:23, stack:6656, core=0
I (00:00:00.163) system_api: Base MAC address is not set
I (00:00:00.164) system_api: read default base MAC address from EFUSE
I (1193) wifi:wifi firmware version: eeaa27d
I (1195) wifi:wifi certification version: v7.0
I (1199) wifi:config NVS flash: enabled
I (1203) wifi:config nano formating: disabled
I (1207) wifi:Init data frame dynamic rx buffer num: 32
I (1212) wifi:Init management frame dynamic rx buffer num: 32
I (1217) wifi:Init management short buffer num: 32
I (1222) wifi:Init static tx buffer num: 16
I (1226) wifi:Init tx cache buffer num: 32
I (1230) wifi:Init static rx buffer size: 1600
I (1234) wifi:Init static rx buffer num: 10
I (1237) wifi:Init dynamic rx buffer num: 32
I (00:00:00.222) wifi_init: rx ba win: 6
I (00:00:00.226) wifi_init: tcpip mbox: 32
I (00:00:00.231) wifi_init: udp mbox: 6
I (00:00:00.236) wifi_init: tcp mbox: 6
I (00:00:00.240) wifi_init: tcp tx win: 5744
I (00:00:00.245) wifi_init: tcp rx win: 5744
I (00:00:00.250) wifi_init: tcp mss: 1440
I (00:00:00.255) wifi_init: WiFi/LWIP prefer SPIRAM
I (00:00:00.260) wifi_init: LWIP IRAM OP enabled
I (00:00:00.266) wifi_prov_scheme_ble: BT memory released
I (00:00:00.272) app: Starting provisioning
I (00:00:00.277) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1390) wifi:mode : sta (ac:0b:fb:6c:b5:c4)
I (1391) wifi:enable tsf
I (00:00:00.373) BTDM_INIT: BT controller compile version [bf9c077]
I (00:00:00.541) protocomm_nimble: BLE Host Task Started
I (00:00:00.544) wifi_prov_mgr: Provisioning started with service name : PROV_6CB5C4 
I (00:00:00.544) app: Provisioning started
I (00:00:00.549) app: Scan this QR code from the provisioning application for Provisioning.
I (00:00:00.559) QRCODE: Encoding below text with ECC LVL 0 & QR Code Version 10
I (00:00:00.566) QRCODE: {"ver":"v1","name":"PROV_6CB5C4","pop":"abcd1234","transport":"ble"}
I (00:00:00.570) NimBLE: GAP procedure initiated: advertise; 
I (00:00:00.582) NimBLE: disc_mode=2
I (00:00:00.586) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (00:00:00.597) NimBLE: 

                                      
  █▀▀▀▀▀█ ▀▀    ▀ ▀▄█▀▀█ ▄▄ █▀▀▀▀▀█   
  █ ███ █ █▄ █▀█ ██▄▄   ▀▀▄ █ ███ █   
  █ ▀▀▀ █ ▀█▀▀▄ ▀▀▀▀▀█  ▀▀▀ █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ ▀▄▀▄▀ █▄█▄█▄█ █ █ ▀▀▀▀▀▀▀   
  ▀▀█▀▄▄▀▄▀▄   ▄▄▀█▀ █▄▄ ▄█▀▄ ▀▀█▄█   
  ▄█▀▄▀▄▀ ▄  ▄▄█▀ ▄  ▄ ▀ ██▀ ▀▄▀▀▀▀   
  ▄█▀▄██▀▄▀▀▄█ ▀▄████▄  ▄█▄▀ █▀▄ ▀    
  ▄▀ ▄ ▀▀▀▄▀▄ █    ▀ █ ▄ ▄ ▄  ▄ █ ▄   
   ▄▀█▀█▀▀ ▄█▀▄█ ▄▀ ▄ ▀▀▄▀█▄█▄█ █▀█   
   █▀ ██▀ ▀▀▄▀█▀▄█▄▄ ▄▄   ▄  ▀▀▄ ▀    
  █▀█▄▄▀▀ ▄▀▀▀█▄▀▀█▀▄▄▀▀▀▀ █▄▄  █▀▄   
   ▄▄█▀▄▀▀█▀█▀▀▄  █   ▀ ██ ▄▄▄ █▀ █   
  ▀ ▀   ▀▀█▀▄▄█▀██▄▀▀▄▄   █▀▀▀█▀ ▄▀   
  █▀▀▀▀▀█  █ █ ▀█ ▀ █▀ █▀▄█ ▀ █ █     
  █ ███ █ ▄ █  █▀▄▀   ▄▀▀▀▀█▀▀▀ ▀█▀   
  █ ▀▀▀ █ █▄█▄▄▀▄▀▄█▀▄▀▀  ▀ █▀▄█▀▀▄   
  ▀▀▀▀▀▀▀ ▀  ▀▀ ▀  ▀▀  ▀ ▀ ▀▀ ▀ ▀     
                                      

I (00:00:00.781) app: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_6CB5C4","pop":"abcd1234","transport":"ble"}
```

allard-potma, 2022-11-03:
Did you try another terminal? I had the same problem with the DPP example, the builtin terminal of eclipse gave this problem, termterm did not.

diplfranzhoepfinger, 2022-11-03:
i did try:
* Eclipse
* The Terminal of https://github.com/esphome/esp-web-tools

both same Problem.

idea--list, 2022-11-03:
This bug is known for a while. If you absolutely need the QR code you can try putty. Until now that has always rendered a readable QR code on my side. However the whole issue can be easily solved by opening the link generated below the QR code.

By opening that link you will find the same QR code properly displayed in your internet browser. If you need the QR code for a commercial product, you can just print out that QR code and put it into the packaging of the product so that the end user can provision their devices easily.

diplfranzhoepfinger, 2022-11-03:
>:

> This bug is known for a while.
any chance to help espressif ?

> If you absolutely need the QR code you can try putty. Until now that has
> always rendered a readable QR code on my side. However the whole issue can
> be easily solved by opening the link generated below the QR code.
true, and this works fine. as i stated.

> By opening that link you will find the same QR code properly displayed in
> your internet browser. If you need the QR code for a commercial product,
> you can just print out that QR code and put it into the packaging of the
> product so that the end user can provision their devices easily.
true, but in my case it should be used by a "online" installer.