espressif/esp-idf #10100: Add an approximate release timeline and/or roadmap

ljden, 2022-11-02:
### Is your feature request related to a problem?

When beginning a new project it is useful to have an idea of how close a beta is to release to gauge whether the new features are worth waiting for (if waiting is even feasible) or if the current stable should be used.

### Describe the solution you'd like.

A clear indication of approximate timelines for version releases, eg when will v5.0 be released, when will stable move from v4.4.x to v5.x

### Describe alternatives you've considered.

(Others) asking on the forum hasn't seemed to generate much clarification of timelines https://esp32.com/viewtopic.php?t=26934

### Additional context.

In particular, support for a linux target is of interest to us so that we can have thorough, host based, unit testing. This does not seem to be properly supported in v4.4 and the current advise is to "switch to v5.0" https://github.com/espressif/esp-idf/issues/10088#issuecomment-1300250055. As we are developing for a production project this is not feasible unless v5.0 will be officially released in the short term. A formal timeline or road map indicating what is blocking the release and when the release may occur would aid in this decision greatly

dobairoland, 2022-12-05:
ESP-IDF v5.0 has been released 3 days ago.

I'm sorry that no one shared a timeline with you. This is not lack of transparency but honestly nobody knows. Even though we have a soft schedule but it is often delayed for several reasons (more bugs discovered, not every feature of the chip bounded with the release is ready). We are sorry for that. There are discussions about how to improve this and be able to release more aligned to the schedule (which then could be shared with you all).

closed 2022-12-05 by dobairoland

ljden, 2022-12-05:
It is not so much this specific release as the project more generally. I think having even a rough timeline of which quarter/month releases are planned, what noteworthy features are planning on being added etc would be very useful. 

Having a rough timeline from igrr https://github.com/espressif/esp-idf/issues/10103#issuecomment-1303072766 gave me a good indication the v5.0 release was around a month away. Before that it could have been 3-6 months away for all I knew, which would mean there was no way it would be ready in time for our project and we would have to try cherry-pick the features we needed. Even if it's just an esp32 forum post that has semi-regular updates on progress and timelines that would be better than the current situation.

Glad to hear there are discussions on how to improve this, sounds like it is something that is being considered. If that's the case I thing the ticket should be reopened