espressif/esp-idf #10101: ESP32-S3 Espressif IDE FreeRTOS task debugging using Eclipse FreeRTOS plugin

krupis, 2022-11-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I am trying to setup FreeRTOS debugging (monitoring each task stack, priority and etc) via the Eclipse IDE plugin.

I use ESP32-S3 Devkit and I am debugging using built in USB JTAG. I setup the debug configuration as following (selected the right cfg):
[image]

and I can debug the ESP32-S3 without any issues (placing breakpoints and etc...)

I have read about FreeRTOS debuging plugin from the NXP:
https://mcuoneclipse.com/2016/07/06/freertos-kernel-awareness-for-eclipse-from-nxp/

I have downloaded this plugin and added it into my view. I have run the debugging session and placed a breakpoint in one of my FreeRTOS tasks. The FreeRTOS task view seemed to work as it loaded some usefull information about currently running tasks. 

[image]

In order to use full potential of this plugin, I should enable GENERATE_RUN_TIME_STATS for my project in the sdkconfig. After doing so, I would expect the FreeRTOS task view to automatically refresh and show live statistics. After enabling this and building project again, It does not seem to work as expected:

When there are no breakpoints placed, the FreeRTOS task view shows nothing. As soon as I place a breakpoint, it fetches data and displays freertos task information. When I remove the breakpoint and run the code again, the data dissapears. I cant seem to understand how can I debug freertos task in real time. For some reason enabling the GENERATE_RUN_TIME_STATS did not work and it still warns me that I need to enable:

[image]

But you can see from my sdkconfig that it is enabled:
[image]

Did anyone else have similar issues with ESP32 FreeRTOS task debugging? I really hope to get some help regarding this as I feel this is a cool debugging feature and could help me out.

[image]

ChrisJP84, 2023-01-20:
Also waiting for this to be sorted out. Seems like a core feature required for professional development. Any workarounds?

krupis, 2023-01-20:
Still waiting for any feedback regarding real time RTOS debugging from espressif team. I didint find any workarrounds yet.

Just need confirmation if its possible to achieve or not. Surely if they have option for enabling real time FREERTOS stats it should be capable?

ammaree, 2023-01-20:
+1 please!!!

krupis, 2023-01-20:
@zikalino  Please shed some light :)

igrr, 2023-01-20:
> After doing so, I would expect the FreeRTOS task view to automatically refresh and show live statistics.

That might work on RISC-V chips (like the ESP32-C3) which implement memory access while the CPU is running, but not on Xtensa ones. On Xtensa, either tracing has to be done via app_trace (which we use for SystemView) or you need to halt the CPU, for instance by stopping on a breakpoint.

In general, this looks more like a feature request which should be directed at https://github.com/espressif/idf-eclipse-plugin rather than IDF.

krupis, 2023-01-20:
> That might work on RISC-V chips (like the ESP32-C3) which implement memory access while the CPU is running, but not on Xtensa ones. On Xtensa, either tracing has to be done via app_trace (which we [...]

Thank you very much for your response. At least now we know!

I will try to get my hands on ESP32-C3 and test this out.

Additionally, can you clarify a little what do you mean :
> On Xtensa, either tracing has to be done via app_trace (which we use for SystemView). 

Does this need additional implementation on my program? Do I need to manually call app_trace functions?

igrr, 2023-01-20:
> Does this need additional implementation on my program? Do I need to manually call app_trace functions?

There are three options that I see:

1. Use SystemView tracing instead of the Eclipse plugin you have found. SystemView tracing is already supported in ESP-IDF. You can load the trace file into the SystemView utility to see the timeline of task execution. No implementation is needed on your part, just changing a few configuration options. The downside is that it's not the plugin you are asking about, it's another tool.
2. Still use SystemView on the chip side, but modify the plugin on the host side to interpret SystemView trace data, showing the same task list as in the plugin you have shown. This requires some work on modifying the plugin. (I think this is the most viable option, and what could be done in the feature request you have opened.)
3. Create a periodic timer, call FreeRTOS task list API from the timer callback to get the information about task run-time and stack size. Send this data to the host via app_trace API. Modify the plugin to get this data on the host side and present it in the same way as the plugin you have shown. A bit more work than in the previous option, but no dependency on SystemView.

zikalino, 2023-03-03:
Moved the issue to https://github.com/espressif/idf-eclipse-plugin/issues/712 so and closing here.

closed 2023-03-03 by zikalino