espressif/esp-idf #10102: BLE Scanner example is not scanning continuously

mohamed-jasim, 2022-11-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF-Version : 4.4.2
SoC : ESP32-C3
I've two esp32-c3 dev kit. In one of them I flashed gatt_server example and gatt_client example in the second. gatt server is advertising with 40ms interval and gat client is scanning with 60ms scan window and 60ms scan interval. Also scan duration 0 for infinite scanning. With this configuration gattc should find gatts 16 times in a second ideally. But it is taking more than 30 seconds to detect after the first scan result event. Need help in solving this.
Config
====
[image]

Logs
===
[image]

xiewenxiang, 2022-11-03:
@mohamed-jasim 
Please turn off `BLE Scan Duplicate Options` and try is again.

```c
make menuconfig -> component config -> Bluetooth -> Controller options -> BLE Scan Duplicate Options
```

mohamed-jasim, 2022-11-03:
@xiewenxiang BLE duplicate scanning was disabled by default in SDK config
[image]
But was enabled in scan params of example code. After disabling in code, the scan result was coming in expected interval.

xiewenxiang, 2022-11-03:
Yes, there is also a scan duplicate option in scan parameters.

mohamed-jasim, 2022-11-07:
Correct me if I'm wrong. In my understanding scan duplicate is applicable for a particular scan window. And this is to restrict multiple appearances of a device which is having shorter advertising interval in a bigger scan window. So this duplicate mac id filter enabled shouldn't delay the scan result event. In my case my advertiser is advertising in 1000ms and the scanner is detecting the advertiser in 30 seconds and more

mohamed-jasim, 2022-11-09:
I'm able to see multiple events for MAC IDs in same scan window even after enabling Duplicate filtering for mac id. Attaching the scan params, config and output logs
[image]
[image]

xiewenxiang, 2022-11-10:
If enable duplicate scan filtering, it's not mean the same device will be reported only once per scan window. It just means that the same device will be reported only once durning a certain time interval.

closed 2024-02-26 by Alvin1Zhang