espressif/esp-idf #10103: idf_component_register() doesn't seem to work with multiple Catch2 source files

ljden, 2022-11-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

CMakeLists.txt below results in a single executable that runs the test found in `test1.cpp`
```CMake
idf_component_register(SRCS test1.cpp)
find_package(Catch2 3 REQUIRED)
target_link_libraries(${COMPONENT_LIB} PRIVATE Catch2::Catch2WithMain)
```

### What is the actual behavior?

Test is not run when executable is run.
Adding an explicit executable adds the tests to the (separate) binary
Adding:
```CMake
add_executable(foo test1.cpp)
target_link_libraries(foo PRIVATE Catch2::Catch2WithMain)
```

### Steps to reproduce.

Install Catch2 v3
Create a basic test case in a cpp file with no main and set up files as below

### More Information.

```bash
vbox@vbox:~/component1/host_test$ ./build/linux_host_app.elf 
Randomness seeded to: 364657046
===============================================================================
No tests ran

vbox@vbox:~/component1/host_test$ ./build/esp-idf/main/foo 
Randomness seeded to: 101586356
===============================================================================
All tests passed (4 assertions in 1 test case)

```
```bash
vbox@vbox:~/component1/host_test$ cat CMakeLists.txt 
cmake_minimum_required(VERSION 3.8)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)

list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/freertos/")
list(APPEND EXTRA_COMPONENT_DIRS "../")
project(linux_host_app)
```
```bash
vbox@vbox:~/component1/host_test$ cat main/CMakeLists.txt 
if(NOT CMAKE_BUILD_EARLY_EXPANSION)
    # https://github.com/espressif/esp-idf/issues/9929#issuecomment-1270569206
    find_package(Catch2 3 REQUIRED)
endif()

idf_component_register()
target_link_libraries(${COMPONENT_LIB} INTERFACE Catch2::Catch2WithMain)

add_executable(foo test1.cpp)
target_link_libraries(foo PRIVATE Catch2::Catch2WithMain)
```
```bash
vbox@vbox:~/component1/host_test$ cat main/test1.cpp 
#include <catch2/catch_test_macros.hpp>

unsigned int Factorial( unsigned int number ) {
    return number <= 1 ? number : Factorial(number-1)*number;
}

TEST_CASE( "Factorials are computed", "[factorial]" ) {
    REQUIRE( Factorial(1) == 1 );
    REQUIRE( Factorial(2) == 2 );
    REQUIRE( Factorial(3) == 6 );
    REQUIRE( Factorial(10) == 3628800 );
}
```

ljden, 2022-11-04:
Thanks for the link. We don't really want to use v5.0 until it is formally released but this is not the first time the solution to a problem has been don't use v4.4... any chance you have an idea of the approx release date?

ljden, 2022-11-04:
Awesome, thanks for the update
For now I've cherry picked the commit https://github.com/espressif/esp-idf/commit/273633ee310fbc18b17edfaeae3f3121508e3b8d  into our fork of the repo and that seems to work well. Happy to call that good enough

closed 2022-11-04 by ljden

ljden, 2022-11-23:
@igrr how's the v5.0-rc1 going? Hopefully the v5.0 release is immanent 🤞🏼

igrr, 2022-11-23:
@ljden v5.0-rc1 is here: https://github.com/espressif/esp-idf/releases/tag/v5.0-rc1. We'll tag v5.0 next week if no blocking issues are reported.