espressif/esp-idf #10104: [Crash][v4.4.2][ESP32-S3] i2c_master_read_from_device() sometimes hits LoadProhibited at i2c_ll_get_intsts_mask()

chipweinberger, 2022-11-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

Should not crash.

### What is the actual behavior?

About 1 in ~50 invocations, my device crashes when calling i2c_master_read_from_device()

### Steps to reproduce.

Hard to give exact details. 
1. install i2c
2. call i2c_master_read_from_device() a bunch of times

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9936749/sdkconfig.txt)

### Debug Logs.

```plain
I (71243) pd stsafe impl: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (71253) pd stsafe impl: 00 b3 15 
I (71283) pd stsafe impl: BusRecv (5 bytes)
E (71283) pd stsafe impl: pd_stsafe_intf_BusRecv() address:0x40 data:0x3dec6a50 len:5 ESP_FAIL
I (71283) pd stsafe impl: BusRecv (5 bytes)
I (71293) pd stsafe impl: 78 78 78 78 00 
I (71293) pd stsafe impl: BusRecv (30843 bytes)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420e3501  PS      : 0x00060032  A0      : 0x8037c408  A1      : 0x3fca3500  
0x420e3501: i2c_ll_get_intsts_mask at /esp-idf/components/hal/esp32s3/include/hal/i2c_ll.h:277
 (inlined by) i2c_hal_get_intsts_mask at /esp-idf/components/hal/i2c_hal.c:69

A2      : 0x3fca4a8c  A3      : 0x3fca3530  A4      : 0x7e2803e8  A5      : 0x7e2803e8  
A6      : 0x7e2803e8  A7      : 0x3fca98fc  A8      : 0x3fca98fc  A9      : 0x00000000  
A10     : 0x00060123  A11     : 0x00060120  A12     : 0x7e2803e8  A13     : 0x3fcf58e0  
A14     : 0x00060123  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x420e34fe:0x3fca3500 0x4037c405:0x3fca3530 0x4037915d:0x3fca3570 0x420d729c:0x3fcf5950 0x42005e2d:0x3fcf5980 0x40386f2f:0x3fcf59b0 0x40389f4e:0x3fcf59e0
0x420e34fe: i2c_ll_get_intsts_mask at /esp-idf/components/hal/esp32s3/include/hal/i2c_ll.h:277
 (inlined by) i2c_hal_get_intsts_mask at /esp-idf/components/hal/i2c_hal.c:69

0x4037c405: i2c_isr_handler_default at /esp-idf/components/driver/i2c.c:491

0x4037915d: _xt_medint2 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1200

0x420d729c: cpu_ll_waiti at /esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182
 (inlined by) esp_pm_impl_waiti at /esp-idf/components/esp_pm/pm_impl.c:837

0x42005e2d: esp_vApplicationIdleHook at /esp-idf/components/esp_system/freertos_hooks.c:63

0x40386f2f: prvIdleTask at /esp-idf/components/freertos/tasks.c:3987 (discriminator 1)

0x40389f4e: vPortTaskWrapper at /esp-idf/components/freertos/port/xtensa/port.c:131
```

### More Information.

**Code:**
```
int32_t pd_stsafe_intf_BusInit()
{
    ESP_LOGI(TAG, "BusInit");

    i2c_port_t port = PD_I2C_PORT;
    int sdaGpio = pd_gpio_get(PD_GPIO_STSAFE_SDA);
    int sclGpio = pd_gpio_get(PD_GPIO_STSAFE_SCL);

    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = sdaGpio,
        .scl_io_num = sclGpio,
        .sda_pullup_en = true,
        .scl_pullup_en = true,
        .master = {
            .clk_speed = 100000,
        },
        .clk_flags = 0
    };

    esp_err_t err = i2c_param_config(port, &conf);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "i2c_param_config() %s", esp_err_to_name(err));
        return 1;
    }

    size_t txBufLen = 0; // I2C master doesn't need a buffer
    size_t rxBufLen = 0; // I2C master doesn't need a buffer

    err = i2c_driver_install(port, conf.mode, rxBufLen, txBufLen, 0);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "i2c_driver_install() %s", esp_err_to_name(err));
        return 2;
    }

    // success
    return 0;
}

int32_t pd_stsafe_intf_BusRecv(uint16_t address, uint8_t* data, uint16_t len)
{
    ESP_LOGI(TAG, "BusRecv (%u bytes)", len);

    i2c_port_t port = PD_I2C_PORT;
    esp_err_t err = i2c_master_read_from_device(port, 
        (uint8_t) 0x20, data, len, 1000 / portTICK_PERIOD_MS);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "pd_stsafe_intf_BusRecv() address:0x%02x data:%p len:%u %s",
            address, data, len, esp_err_to_name(err));
        return 4;
    }

    ESP_LOG_BUFFER_HEX(TAG, data, len);

    return 0;
}
```

kriegste, 2022-11-04:
Maybe related to https://github.com/espressif/esp-idf/issues/10079#issuecomment-1300401211 ?

chipweinberger, 2022-11-04:
certainly looks similar

chipweinberger, 2022-11-04:
The iram patch in https://github.com/espressif/esp-idf/issues/10079#issuecomment-1300401211 fixes it! 🎉

chipweinberger, 2022-11-04:
Thanks for the super fast reply @kriegste 

This is the fastest I've ever fixed an ESP-IDF crash =)

chipweinberger, 2022-11-04:
Closing this issue in favor of the existing one.

closed 2022-11-04 by chipweinberger