espressif/esp-idf #10107: classic bluetooth hid device doesn't reconnect

lishuai199502, 2022-11-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,i just use classic bluetooth with the example,esp-idf-v4.4.2\examples\bluetooth\bluedroid\classic_bt\bt_hid_mouse_device,
when I connect my android phone or my pc, it works well, but when I off my phone, the bluetooth disconnect, 
then I on my phone, it doesn't reconnect to esp32.
The same problem occurs in my PC, 
I want to know how to deal with this.

boblane1, 2022-11-07:
Hi @lishuai199502 ，

The ESP32 HID device set the SDP attribute `HIDReconnectInitiate` to `True`, it indicates that the HID device will be primarily responsible for connection re-establishment.  So if you want to reconnect to your phone or PC, you can call the function `esp_bt_hid_device_connect` at the user specific frequency.

lishuai199502, 2022-11-16:
@boblane1 Thanks very much for your reply, I've tested the method your show me, and it works well. Now I wonder know, how to distinguish the disconnect mode. For example, I turned my phone bluetooth off, then I turn on, esp32 should re-connect, but when I disconnect bluetooth by click disconnect button, esp32 shouldn't re-connect to.

boblane1, 2022-11-16:
@lishuai199502 ,

It seems that the Bluetooth stack can't distinguish the difference between the two procedures that you mentioned.

boblane1, 2022-11-18:
@lishuai199502 ,

According to the `Bluetooth HID Profile version 1.0`, when a `cabled` HID is plugged into a host by a user, the user creates both a “bond” between the devices and a 1:1 relationship. So, if you want the HID Device can be connected by other HID Host, you shall remove the virtually cabled device first. You can call `esp_bt_hid_device_virtual_cable_unplug`, but this will also remove the previous bonded HID Host information.

Please try the following patch.
[test.zip](https://github.com/espressif/esp-idf/files/10039467/test.zip)

lishuai199502, 2022-11-18:
@boblane1 
Hi, my bro. I've tested the patch and it works well, but another problem occured.
I have a HID keyboard moudle, it works as this, 
First, I pair my pc to the moudle and pair my phone to the module. After this, I turn off the three.
Second, I turn on my phone BT, when I turn on the moudle, it can connect to my phone automatically. Then I turn off them two.
Third, I turn on my PC BT, then turn on the moudle, also, it can connect automatically.
Forth, it confused me, I connect my phone to moudle, then I turn on my PC BT, and then I turn off my phone BT, the moudle also can connect to my PC automatically.  By contrast, I connect my PC to the moudle, then turn on my phone BT, it cann't connect automatically.
It seems that, the PC will try to connect my moudle every time my PC BT turn on.
meanwhile, I found that, I turn off my moudle, when I turn on my PC BT, it shows connected firstly, after a moment, it shows paired. But the PC BT list show my ESP32 alwayes connected no matter I turn on or off my ESP32.

https://user-images.githubusercontent.com/23136422/202719497-0db3a40c-f3af-44fc-b1bb-3bf25060a9aa.mp4

I don't know what happened between my PC and my moudle.
The video below shows that I turn on my moudle first then I turn on my PC BT, it connected automatically.
In the video, I click the device means it connect successfully.

https://user-images.githubusercontent.com/23136422/202721831-91f53ce0-2fc2-40ec-8065-5f4b7138e738.mp4

I want my ESP32 BT also can do like this, could give me a little more help.
Thanks my bro.

boblane1, 2022-11-21:
@lishuai199502 ,

>meanwhile, I found that, I turn off my moudle, when I turn on my PC BT, it shows connected firstly, after a moment, it shows paired. But the PC BT list show my ESP32 alwayes connected no matter I turn on or off my ESP32.

Like what I said before,
> The ESP32 HID device set the SDP attribute HIDReconnectInitiate to True, it indicates that the HID device will be primarily responsible for connection re-establishment. So if you want to reconnect [...]

so if you want to the HID Host to initiate the reconnect procedure, you can change the attribute `HIDReconnectInitiate` to False.
You can change
```c
result &= SDP_AddAttribute(handle, ATTR_ID_HID_RECONNECT_INITIATE, BOOLEAN_DESC_TYPE, 1, (uint8_t *)&bool_true);
```
to 
```c
result &= SDP_AddAttribute(handle, ATTR_ID_HID_RECONNECT_INITIATE, BOOLEAN_DESC_TYPE, 1, (uint8_t *)&bool_false);
``` 
in the function of `HID_DevAddRecord` in `/esp-idf/components/bt/host/bluedroid/stack/hid/hidd_api.c`.

The following attach is `Bluetooth HID Profile version 1.0`, you can learn some details about the implementation.
[HID_SPEC_V10.pdf](https://github.com/espressif/esp-idf/files/10052659/HID_SPEC_V10.pdf)

closed 2022-12-14 by espressif-bot