espressif/esp-idf #10110: ESP32 hangs on stack overflow

kewelspintly, 2022-11-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-DEVKITS

### Power Supply used.

USB

### What is the expected behavior?

Print error and reboot

### What is the actual behavior?

Enters panic handler and remains in hang state and requires a next power cycle to reboot.

### Steps to reproduce.

Unfortunately we cannot share the code here.

1) Initialize a task with low stack space.
2) When a stack overflow occurs, the code enters the panic_handler() function but does not execute it completely. 
Note : This does not occur on all stack overflows. Most of the time a stack overflow occurs the ESP32 prints the task in which the stack overflow occurred and reboots normally.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9928136/sdkconfig.txt)

### More Information.

Module : ```ESP32-WROOM-32E-8MB```
External Libraries : ```esp-aws-iot v3.1.x```

What we have noticed is, the ESP32 halts and goes into a state similar to reset as all GPIO's are in their reset state and stays there.
This happens on a stack overflow, however instead of rebooting the ESP32 gets stuck somewhere in the ```panic_handler()``` function, we presumed this after observing the gpio states as mentioned below

We have added two gpio_set_level() in the panic_handler() function in panic_handler.c,
Under normal resets both gpio 4 and 2, are set to logic 3.3v.
However when the esp32 goes in hang state only gpio 4 is set, and gpio 2 remains low.
```c
static void panic_handler(void *frame, bool pseudo_excause)
{
    panic_info_t info = { 0 };

    gpio_set_level(4, 1);
    /*
     * Setup environment and perform necessary architecture/chip specific
     * steps here prior to the system panic handler.
     * */
    int core_id = cpu_hal_get_core_id();

    // If multiple cores arrive at panic handler, save frames for all of them
    g_exc_frames[core_id] = frame;

#if !CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE
    // These are cases where both CPUs both go into panic handler. The following code ensures
    // only one core proceeds to the system panic handler.
    if (pseudo_excause) {
#define BUSY_WAIT_IF_TRUE(b)                { if (b) while(1); }
        // For WDT expiry, pause the non-offending core - offending core handles panic
        BUSY_WAIT_IF_TRUE(panic_get_cause(frame) == PANIC_RSN_INTWDT_CPU0 && core_id == 1);
        BUSY_WAIT_IF_TRUE(panic_get_cause(frame) == PANIC_RSN_INTWDT_CPU1 && core_id == 0);

        // For cache error, pause the non-offending core - offending core handles panic
        if (panic_get_cause(frame) == PANIC_RSN_CACHEERR && core_id != esp_cache_err_get_cpuid()) {
            // Only print the backtrace for the offending core in case of the cache error
            g_exc_frames[core_id] = NULL;
            while (1) {
                ;
            }
        }
    }

    // Need to reconfigure WDTs before we stall any other CPU
    esp_panic_handler_reconfigure_wdts();

    esp_rom_delay_us(1);
    SOC_HAL_STALL_OTHER_CORES();
#endif

    esp_ipc_isr_stall_abort();

    if (esp_cpu_in_ocd_debug_mode()) {
#if __XTENSA__
        if (!(esp_ptr_executable(cpu_ll_pc_to_ptr(panic_get_address(frame))) && (panic_get_address(frame) & 0xC0000000U))) {
            /* Xtensa ABI sets the 2 MSBs of the PC according to the windowed call size
             * Incase the PC is invalid, GDB will fail to translate addresses to function names
             * Hence replacing the PC to a placeholder address in case of invalid PC
             */
            panic_set_address(frame, (uint32_t)&_invalid_pc_placeholder);
        }
#endif
        if (panic_get_cause(frame) == PANIC_RSN_INTWDT_CPU0
#if !CONFIG_ESP_SYSTEM_SINGLE_CORE_MODE
                || panic_get_cause(frame) == PANIC_RSN_INTWDT_CPU1
#endif
           ) {
            wdt_hal_write_protect_disable(&wdt0_context);
            wdt_hal_handle_intr(&wdt0_context);
            wdt_hal_write_protect_enable(&wdt0_context);
        }
    }

    // Convert architecture exception frame into abstracted panic info
    frame_to_panic_info(frame, &info, pseudo_excause);

    gpio_set_level(2, 1);
    // Call the system panic handler
    esp_panic_handler(&info);
}
```

As for Logs, the ESP32 just stops printing out data over the UART, and does not reach the part where the the Backtrace is printed out.
However upon recreating the issue multiple times, we did observe the back trace printed once, as shown below... 
[image]

Also, here are some more verbose logs
[image]

Dazza0, 2022-11-09:
@kewelspintly When the ESP32 hangs after printing the register dump, does one of the WDTs kick in eventually to trigger a system reset? If so, the post system-reset PC values of each core could give a clue as to where the core's are stuck at.

kewelspintly, 2022-11-10:
@Dazza0 Thanks for the reply.
To answer your question, the WDT does not kick in ever, we have kept the esp32 ON in that state for more than 8 hours.
According to our observation the panic_handler function stop at line
```c
SOC_HAL_STALL_OTHER_CORES();
```
We tested this by putting a gpio_set after this line (which never got triggered).

By commenting out this line in the ```panic_handler()``` function, we have tested for about 2 days, and not faced the issue even once. Can this ```SOC_HAL_STALL_OTHER_CORES()``` be somehow causing the esp32 to get stuck, also it would be enlightening to us to know exact flow of the panic handler function, as in why the stall function is called in the first place?
Any input is appreciated.

kewelspintly, 2022-11-22:
Hi All, 

Just wanted to share an update on this.
I have been able to recreate this issue by using a simple example.
Please find the attached code.
[Basic_Project.zip](https://github.com/espressif/esp-idf/files/10065268/Basic_Project.zip)
In this example, every 9 seconds we invoke a panic due to a stack overflow and in about 10-20 mins the ESP32 gets hanged.

What we have noticed is this issue does not occur if WiFi is not initialized, also it happens faster if we print long strings of data.
We feel its similar to this issue https://github.com/espressif/esp-idf/issues/8033, however we do not call esp_restart().

brunohpg, 2022-12-06:
I have similiar problem. After some days the ESP32 completely hangs.

I checked it is not in download mode, interrupt watchdog is on and interrupt task is on. Both interrupts calls panic/abort.

It start when I have migrated from 3.x to 4.x.

I have updated the idf to latest 4.x release and I'm trying `[Basic_Project.zip](https://github.com/espressif/esp-idf/files/10065268/Basic_Project.zip)` of @kewelspintly.

I'll check if the RTC Watchdog reset system even when the CPU get stuck. @Dazza0 do you think this will prevent hangs?

brunohpg, 2022-12-07:
Good news:

Configuring the rtc watchdog to reset the RTC and the system, after hangs it restarts.

```
    rtc_wdt_protect_off();  // Disable RTC WDT write protection
    rtc_wdt_disable();
    rtc_wdt_set_length_of_reset_signal(RTC_WDT_SYS_RESET_SIG, RTC_WDT_LENGTH_3_2us);
    rtc_wdt_set_stage(RTC_WDT_STAGE0, RTC_WDT_STAGE_ACTION_RESET_RTC);
    rtc_wdt_set_time(RTC_WDT_STAGE0, 60000);
    rtc_wdt_enable();      // Start the RTC WDT timer
    rtc_wdt_protect_on();  // Enable RTC WDT write protection
```

It is certainly not the best solution. The system loses RTC and RTC memory. However, at least you don't lose access to the device.

This is interesting, RTC may be involved in the process.

My suspicion is that there are two exceptions at the same time, most likely one related to the UART, so one should go, however, the UART hangs at some point. Or both exception are IWTD.

It doesn't make much sense for both CPUs stop. To stop one CPU the other has to be active and vice versa. 
However, there is a Busy wait rule and both could crash if the exceptions are IWTD. Even if IWTD restarts, they both restart at the same time and wait together. Not sure if this is possible.

brunohpg, 2022-12-22:
@Dazza0 
I can reproduce his problem with the example he put. Maybe there is some version difference of esp-idf or something.
You have to wait a long time for it hangs. It only hangs after restarting many times! So better wait more time.

You asked me in another issue the PC after the restart, I'll see if I can.

However, I already say, RTC using RTC_WDT_STAGE_ACTION_RESET_SYSTEM does not reset the system!
It only resets RTC_WDT_STAGE_ACTION_RESET_RTC. Therefore, it seems to me that not even WDT can restart, except "force" one.

brunohpg, 2022-12-23:
Here is the moment when ESP hangs:

```
count: 10, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 9, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 8, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 7, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 6, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 5, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 4, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 3, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 2, printing a long message, The quick brown fox jumps over the lazy dog, The quick brown fox jumps over the lazy dog
count: 1, printing a long message, The quick browets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14876
ho 0 tail 12 room 4
load:0x40080400,len:3792
entry 0x40080694
[0;32mI (28) boot: ESP-IDF v4.4.3-231-g4e7d57da8b-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 08:43:38[0m
[0;32mI (30) boot: chip revision: 1[0m
[0;32mI (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (40) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (45) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (49) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (59) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (93) boot: End of partition table[0m
[0;32mI (97) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13b0ch ( 80652) map[0m
[0;32mI (142) esp_image: segment 1: paddr=00023b34 vaddr=3ffb0000 size=038fch ( 14588) load[0m
[0;32mI (148) esp_image: segment 2: paddr=00027438 vaddr=40080000 size=08be0h ( 35808) load[0m
[0;32mI (163) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=70b44h (461636) map[0m
[0;32mI (330) esp_image: segment 4: paddr=000a0b6c vaddr=40088be0 size=0cfe0h ( 53216) load[0m
[0;32mI (363) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (363) boot: Disabling RNG early entropy source...[0m
[0;32mI (374) cpu_start: Pro cpu up.[0m
[0;32mI (375) cpu_start: Starting app cpu, entry point is 0x40081290[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (391) cpu_start: Pro cpu start user code[0m
[0;32mI (391) cpu_start: cpu freq: 160000000[0m
[0;32mI (391) cpu_start: Application information:[0m
[0;32mI (395) cpu_start: Project name:     Basic_Project[0m
[0;32mI (401) cpu_start: App version:      1[0m
[0;32mI (405) cpu_start: Compile time:     Dec  6 2022 08:43:32[0m
[0;32mI (411) cpu_start: ELF file SHA256:  254c4dc2c82da712...[0m
[0;32mI (417) cpu_start: ESP-IDF:          v4.4.3-231-g4e7d57da8b-dirty[0m
[0;32mI (424) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (431) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (437) heap_init: At 3FFB7F98 len 00028068 (160 KiB): DRAM[0m
[0;32mI (444) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (450) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (456) heap_init: At 40095BC0 len 0000A440 (41 KiB): IRAM[0m
[0;32mI (464) spi_flash: detected chip: gd[0m
[0;32mI (467) spi_flash: flash io: dio[0m
[0;33mW (471) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (485) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
I (579) wifi:wifi driver task: 3ffc0b5c, prio:23, stack:6656, core=0
[0;32mI (579) system_api: Base MAC address is not set[0m
[0;32mI (579) system_api: read default base MAC address from EFUSE[0m
```

Here is the complete log:
[esp.zip](https://github.com/espressif/esp-idf/files/10294792/esp.zip)

kewelspintly, 2022-12-24:
@Dazza0

Thank you for your information on SOC_HAL_STALL_OTHER_CORES(). 
Also as @brunohpg mentioned you need to keep the device on full quite a while so that it undergoes a few restart for the hanging to occur.
Also to answer your query, 
No, we do not have the openocd/JTAG connected when running the test.
So after a few tests what we have noticed is that the hanging is more prominent when WiFi activity is high, hence attaching another example with iperf client running on the ESP32 when the stack overflow happens.
Attached below:
[iperf_crash.zip](https://github.com/espressif/esp-idf/files/10298821/iperf_crash.zip)

To get up and running with this example, please follow the steps below.

1. Change ```ESP32_WIFI_SSID``` and ```ESP32_WIFI_PASS``` from cmd_wifi.h file, to your WiFi configurations.
2. Change ```IPERF_IP_ADDRESS``` from cmd_wifi.h file to the IP address of the Laptop/PC running the iperf sever.
3. Finally run ipref server on the Laptop/PC using the following command : ```iperf -s -i 3```

Also please mention if you change any project configuration, or testing on a different IDF version. 

With the example given above the crash occurs very often, well within 10 minutes of running. We feel the ESP32 should not get hung if the watchdogs (task or interrupt watchdog) are enabled. After testing with the RTC watchdog turned on, the ESP32 enters the state of hanging but, the RTC watchdog eventually resets the chip. Can you please tell us how the RTC module is different compared to the other 2 cores, and why the RTC watchdog resets the chip when WDT0 fails to do the same.
How is the RTC WDT different from the rest of the watchdogs?

Also another thing we noticed was that any code run on the ULP keeps running even though both the cores were in hanging state.
Would love some insight on how the RTC module and the two cores operate.

@brunohpg thanks for the input, can you tell us after how long the issue occurs for you? Also can you try out the example attached above to see if the issue occurs quicker.

Will update with logs of example with RTC turned on, if needed.

Thanks in advance.

KaeLL, 2023-01-02:
> How is the RTC WDT different from the rest of the watchdogs?
Also another thing we noticed was that any code run on the ULP keeps running even though both the cores were in hanging state.
Would love some insight on how the RTC module and the two cores operate.

[Here's a start](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#figure.caption.7)

Dazza0, 2023-01-05:
@kewelspintly @brunohpg Here's a small update about our findings. We've been able to recreate the issue consistently, and it seems like the `SOC_HAL_STALL_OTHER_CORES()` is indeed causing the CPUs to get stuck. I've attached a small example (run on `v4.4.3` that can instantly recreate the issue). Steps to create:

1. `git checkout tags/v4.4.3`
2. Copy the code snippet below into any project
3. Disable the following configurations
    a. `CONFIG_ESP_INT_WDT=n`
    b. `CONFIG_ESP_TASK_WDT=n`
    c. All other configurations can be left as default
4. Build and flash the project to an ESP32. It might take a couple of restarts but the CPUs should eventually get stuck.

<details>
  <summary>Code Snippet</summary>

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/rtc.h"
#include "hal/wdt_hal.h"
#include "esp_cpu.h"
#include "esp_log.h"
#include "esp_heap_caps.h"

#define CORE0_TAG "core0"
#define CORE1_TAG "core1"

static volatile bool sync_flag_start = false;

static void core0_task(void *arg)
{
    volatile uint32_t *some_array = heap_caps_malloc(sizeof(uint32_t) * 16, MALLOC_CAP_8BIT);
    assert(some_array);

    while (1) {
        sync_flag_start = true;

        // Just loop access the allocated memory
        for (int i = 0; i < INT_MAX; i++) {
            for (int j = 0; j < 16; j++) {
                some_array[j] = j;
            }
        }
        ESP_LOGI(CORE0_TAG, "loop");
    }
}

static void core1_task(void *arg)
{
    volatile uint32_t *some_array = heap_caps_malloc(sizeof(uint32_t) * 16, MALLOC_CAP_8BIT);
    assert(some_array);

    // Wait for core0 to start
    while (!sync_flag_start) {
        ;

    }

    while (1) {
        for (int i = 0; i < 100; i++) {
            // Stall the other core
            esp_cpu_stall(0);
            // Access the allocated memory
            for (int j = 0; j < 16; j++) {
                some_array[j] = j;
            }
            // Unstall the other core
            esp_cpu_unstall(0);
        }
        ESP_LOGI(CORE1_TAG, "loop");
    }
}

void app_main(void)
{
    vTaskDelay(10);
    printf("Hello World\n");

    // Initialize RTC WDT to reset system after 20 seconds
    wdt_hal_context_t rwdt_context;
    wdt_hal_init(&rwdt_context, WDT_RWDT, 0, false);
    wdt_hal_write_protect_disable(&rwdt_context);
    wdt_hal_config_stage(&rwdt_context,
                         WDT_STAGE0,
                         20000 * rtc_clk_slow_freq_get_hz() / 1000,
                         WDT_STAGE_ACTION_RESET_SYSTEM);
    wdt_hal_enable(&rwdt_context);
    wdt_hal_write_protect_enable(&rwdt_context);

    xTaskCreatePinnedToCore(core1_task, "core1", 4096, NULL, 10, NULL, 1);
    xTaskCreatePinnedToCore(core0_task, "core0", 4096, NULL, 10, NULL, 0);
}
```
</details>

What's strange is that when the CPUs get stuck, `WDT_STAGE_ACTION_RESET_SYSTEM` has no effect, but `WDT_STAGE_ACTION_RESET_RTC` is still able to reset the system. I'll check with the hardware team as to why `esp_cpu_stall()` can cause the calling CPU to get stuck.

As for a quick workaround, commenting out `SOC_HAL_STALL_OTHER_CORES` in the panic handling code should be fine in most cases. Not stalling the other core during a panic should only be an issue in the edge case where both cores panic at the same time.

krjeschk, 2023-02-24:
@Dazza0 any news related to this topic? Is there any patch available?

idea--list, 2023-03-18:
Does someone know if ESP-IDF V5.0.1 is also affected?

AxelLin, 2023-05-09:
> @kewelspintly @brunohpg Here's a small update about our findings. We've been able to recreate the issue consistently, 

@Dazza0  Any update for the fix?

vshymanskyy, 2023-09-20:
Is there any practical difference between `RTC_WDT_STAGE_ACTION_RESET_SYSTEM` and `WDT_STAGE_ACTION_RESET_RTC`?
Can't find any info on this

AxelLin, 2023-12-25:
@Dazza0 

Do you have any follow up with https://github.com/espressif/esp-idf/issues/10110#issuecomment-1372168634 ?
You said you can reproduce the issue consistently, but this issue is not yet fixed in esp-idf.(it has been almost 1 year)

I do very worry about the case that WDT stop working and the device remains in hang state that requires a next power cycle to reboot.
Sometimes, the device is deployed in a very far away site.
The cost to re-power on a device is very high.
Which also means the device could stop working for quite long period until someone reset it.

@igrr

AxelLin, 2024-02-29:
> ### What is the actual behavior?
> Enters panic handler and remains in hang state and requires a next power cycle to reboot.

@Dazza0 @igrr 
Still no fix for this issue?

AxelLin, 2024-04-21:
@Dazza0 @igrr 

This issue has been marked as "Status: Selected for Development" for 1 year.
How is the status now?

closed 2025-02-23 by espressif-bot