espressif/esp-idf #10113: esp_bt_gap_set_cod

sueng66, 2022-11-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I failed setting bluetooth class device on major device class (always 0x00) and minor device class (always0x1f) , the only work is setting service class

BetterJincheng, 2022-11-07:
@sueng66 Please make sure the calling for `esp_bt_gap_set_cod` is before setting the device to `discoverable` mode.

sueng66, 2022-11-07:
yes I have, I do that just after `esp_bt_gap_register_callback` , and call `esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE)` after got `ESP_SPP_INIT_EVT` that definitly after calling `esp_bt_gap_set_cod`

BetterJincheng, 2022-11-07:
@sueng66  but I just tested on the example `bt_hid_mouse_device` with your code and it worked fine. Could you please describe the procedure to reproduce the issue? Thanks!

sueng66, 2022-11-07:
you are right I try with `esp_hid_device` example , it's could set the cod. My code is base on `bt_spp_initiator` here are my complete code :
```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include <unistd.h>
#include "driver/uart.h"
#include "driver/gpio.h"
#include "nvs.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_gap_bt_api.h"
#include "esp_bt_device.h"
#include "esp_spp_api.h"
#include "console_uart.h"

#include "time.h"
#include "sys/time.h"

#define GPIO_OUTPUT_IO_0    2
#define GPIO_OUTPUT_PIN_SEL  ((1ULL<<GPIO_OUTPUT_IO_0))

#define SPP_TAG "BT_STOREFORWARD"
#define SPP_SERVER_NAME "SPP_SISTEK_SERVER"
#define DEVICE_NAME "RPP02N"
#define NOCONNECT_PERIODE 100
#define CONNECT_PERIODE 750

#define SPP_SHOW_DATA 0
#define SPP_SHOW_SPEED 1
#define SPP_SHOW_MODE SPP_SHOW_SPEED    /*Choose show mode: show data or speed*/

unsigned blinkperiode = NOCONNECT_PERIODE;

void ledblink( void *param)
{
    printf("LED blink Task Started\r\n");
    while(1) {
        gpio_set_level(GPIO_OUTPUT_IO_0, 0);
        vTaskDelay(blinkperiode / portTICK_PERIOD_MS);
        gpio_set_level(GPIO_OUTPUT_IO_0, 1);
        vTaskDelay(blinkperiode / portTICK_PERIOD_MS);
    }
}

static const esp_spp_mode_t esp_spp_mode = ESP_SPP_MODE_CB;

static struct timeval time_new, time_old;
static long data_num = 0;

const esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE;
static const esp_spp_role_t role_master = ESP_SPP_ROLE_MASTER;
static const esp_bt_inq_mode_t inq_mode = ESP_BT_INQ_MODE_GENERAL_INQUIRY;
static const uint8_t inq_len = 30;
static const uint8_t inq_num_rsps = 0;
static const esp_spp_role_t role_slave = ESP_SPP_ROLE_SLAVE;

[... 466 lines cut ...]
    }
    ESP_ERROR_CHECK( ret );

    int r = bt_init();
    if(r<0)
        return;

    r = gap_init();
    if(r<0)
        return;

    r = spp_init();
    if(r<0)
        return;

    console_uart_init();

    printf("Tapper START\r\n");
    xTaskCreate(ledblink, "LED bliker", 2048, NULL, 10, NULL);
    while(1) {
        sleep(1);
    }
}

```

sueng66, 2022-11-07:
Done, I've inserted the `esp_bt_gap_set_cod(siscod, ESP_BT_INIT_COD)` between :

```
        esp_bt_dev_set_device_name(DEVICE_NAME);
        esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
```

closed 2022-11-07 by sueng66

BetterJincheng, 2022-11-07:
@sueng66  Thanks for your reporting. Turn to us any time.

sueng66, 2022-11-07:
the question is why setting cod must be done just before setting scan mode, why could not be done way before it (in my case  in init state) ?

BetterJincheng, 2022-11-07:
@sueng66 In user code `spp_init()` , some initialization will be done. 
In function `bta_jv_enable()` in stack, `cod.major` and `cod_minor` will be initialized.

```c
void bta_jv_enable(tBTA_JV_MSG *p_data)
{
    tBTA_UTL_COD   cod;

    tBTA_JV_STATUS status = BTA_JV_SUCCESS;
    bta_jv_cb.p_dm_cback = p_data->enable.p_cback;
    bta_jv_cb.p_dm_cback(BTA_JV_ENABLE_EVT, (tBTA_JV *)&status, 0);
    memset(bta_jv_cb.free_psm_list, 0, sizeof(bta_jv_cb.free_psm_list));

    /* Set the Class of Device */
    cod.major = BTM_COD_MAJOR_UNCLASSIFIED;
    cod.minor = BTM_COD_MINOR_UNCLASSIFIED;
    utl_set_device_class(&cod, BTA_UTL_SET_COD_MAJOR_MINOR);
}
```

So, when you set the cod before `spp_init`, it will be overwritten to default values.

A suggestion is that setting the cod after `spp_init()`, which is after the arrival of the event of `ESP_SPP_INIT_EVT`.

BetterJincheng, 2022-11-07:
@sueng66  We will improve the document. Thank!