espressif/esp-idf #10115: [Question] Stepper Motor Smooth Controller for More Than One Channel

homerliu1, 2022-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I modified the rmt step_motor example for more than one channel.
By repeating the following sequence,  
step_motor_create_rmt
step_motor_init 
step_motor_step
step_motor_deinit
4 motors were able to be driven one by one.

However, move than one motor cannot be driven simultaneously: 
for (int i = 0; i < 4; i ++) {
   step_motor_create_rmt, RMT_CHANNEL_0 + i, gpiopin+i,
    step_motor_init(motor[i])
}
...
wait for some commands 
for (int i = 0; i < 4; i ++) {
   step_motor_step(motor[i])
}

...
for (int i = 0; i < 4; i ++) {
   step_motor_deinit(motor[i])
}
...
I found inside step_motor_create_rmt function, 
rmt_register_tx_end_callback
may damage other rmt channels, cause the problem.

Any idea?

suda-morris, 2022-11-07:
Hi @homerliu1 I read the callback function in the example, seems all function calls are operating on per-motor's RMT channel. I guess it's because when multiple channels issue the interrupt almost at the same time, and in the interrupt handler, we always process the RMT channels in the same order (from channel 0->n).

BTW, can you try the example from the master branch or release/v5.0? this example has been cleaned up and uses the new RMT API, maybe the new example can offer the ability to drive multiple-step motors.

homerliu1, 2022-11-12:
Hi morris, I tried the master branch example according to your suggestion. 
xTaskCreate was used to generate 4 tasks, each one represents for one rmt motor channel.
It is working. Thank you for your support!