espressif/esp-idf #10116: Can't install tools with Python 3.11

persello, 2022-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### What is the expected behavior?

Following the instructions provided for installing ESP-IDF v4.4.2, I expect the tools to install successfully.

### What is the actual behavior?

Installation of tools fails with an error stating that `longintrepr.h` could not be found.

### Steps to reproduce.

```sh
sudo yum -y update && sudo yum install git wget flex bison gperf python3 python3-pip python3-setuptools cmake ninja-build ccache dfu-util libusbx
python3 --version # Shows version 3.11
mkdir esp
cd esp/
git clone -b v4.4.2 --recursive https://github.com/espressif/esp-idf.git
./esp-idf/install.sh esp32c3 # <~~ Error happens here
```

### Build or installation Logs.

```plain
[riccardo@thinkpad-p51 esp]$ ./esp-idf/install.sh esp32c3
Detecting the Python interpreter
Checking "python" ...
Python 3.11.0
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
WARNING: File /home/riccardo/.espressif/idf-env.json was not found. 
Creating /home/riccardo/.espressif/idf-env.json
Selected targets are: esp32c3
Installing tools: riscv32-esp-elf, openocd-esp32
Installing riscv32-esp-elf@esp-2021r2-patch3-8.4.0
Downloading riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-linux-amd64.tar.gz to /home/riccardo/.espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-linux-amd64.tar.gz.tmp
Done
Extracting /home/riccardo/.espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-linux-amd64.tar.gz to /home/riccardo/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0
Installing openocd-esp32@v0.11.0-esp32-20220411
Downloading openocd-esp32-linux-amd64-0.11.0-esp32-20220411.tar.gz to /home/riccardo/.espressif/dist/openocd-esp32-linux-amd64-0.11.0-esp32-20220411.tar.gz.tmp
Done
Extracting /home/riccardo/.espressif/dist/openocd-esp32-linux-amd64-0.11.0-esp32-20220411.tar.gz to /home/riccardo/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411
Installing Python environment and packages
pip 22.2.2 from /usr/lib/python3.11/site-packages/pip (python 3.11)
Creating a new Python environment in /home/riccardo/.espressif/python_env/idf4.4_py3.11_env
created virtual environment CPython3.11.0.final.0-64 in 2961ms
  creator CPython3Posix(dest=/home/riccardo/.espressif/python_env/idf4.4_py3.11_env, clear=False, no_vcs_ignore=False, global=False)
  seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
    added seed packages: pip==22.3, setuptools==65.5.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
Installing Python packages from /home/riccardo/esp/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
[... 396 lines cut ...]
  ERROR: Failed building wheel for gevent
Failed to build gevent
ERROR: Could not build wheels for gevent, which is required to install pyproject.toml-based projects
Traceback (most recent call last):
  File "/home/riccardo/esp/esp-idf/tools/idf_tools.py", line 1860, in <module>
    main(sys.argv[1:])
  File "/home/riccardo/esp/esp-idf/tools/idf_tools.py", line 1856, in main
    action_func(args)
  File "/home/riccardo/esp/esp-idf/tools/idf_tools.py", line 1577, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr, env=env_copy)
  File "/usr/lib64/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/riccardo/.espressif/python_env/idf4.4_py3.11_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/riccardo/esp/esp-idf/requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.
```

### More Information.

Fedora Workstation 37
```
uname -a
Linux thinkpad-p51 6.0.5-300.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Oct 26 17:24:18 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

The issue seems to be related to old versions of `gevent`. New versions of `gevent` can be installed correctly.
Installing ESP-IDF from the `master` branch works as expected.

dobairoland, 2022-11-06:
This needs more investigation, but it looks like this is caused by an old Cython issue (https://github.com/cython/cython/issues/4461) closed by a year ago.  There are also a couple of 3.11 issues in the gevent tracker: https://github.com/gevent/gevent/issues?q=is%3Aissue+is%3Aclosed

We will try to build wheels of gevent for 3.11 which would solve this issue.

KaeLL, 2022-11-08:
Fix ETA, workarounds besides using docker, anything?

dobairoland, 2022-11-09:
I'm sorry I cannot give an ETA because the root cause is outside of our jurisdiction. The proper fix will come with your Linux distribution's package updates.

The workarounds are:
- don't upgrade to Python 3.11 and use an older version,
- use the docker image: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html.

A workaround from our side will be to provide wheels but it is unknown to us right now how to fix those issues (there are others, the reported failure is just the first one which pops up).

BTW, what Linux distributions (and versions) are you using. Mine is a cutting edge rolling distribution and it still doesn't updated Python to 3.11.

KaeLL, 2022-11-09:
Clear Linux. We got Python 3.11 on 27/Oct

faulesocke, 2022-11-17:
You could also try upgrading to a newer gevent version. The error is reproducible with `pip3 install 'gevent<2.0,>=1.2.2'` but not with `pip3 install gevent`.

KaeLL, 2022-11-17:
Already known, as `master` isn't affected. Also, since that's the most straightforward option, I assume @dobairoland must have good reasoning not to take it.

faulesocke, 2022-11-17:
@dobairoland you mentioned the proper fix will come with my distro's packages. Can you briefly outline what needs to be done there? I can certainly do the required work.

dobairoland, 2022-11-18:
> @dobairoland you mentioned the proper fix will come with my distro's packages.

I realized later that gevent already provides wheels for Python 3.11. My mistake.

I'm sorry for the delay but the solution is harder than I thought. In order to lift the restriction on gevent, we need to lift it on gdbgui. And that needs the lifting of the restriction on pygdbmi which requires to backport fixes not just to ESP-IDF.

elwerene, 2022-11-23:
I have the same problem since updating to Fedora 37.

dobairoland, 2022-11-25:
Crosslinking https://github.com/cs01/gdbgui/issues/447. Later we can revisit this issue and support gdbgui together with Python 3.11. For now it won't be supported.

The workaround will be merged soon and linked to this issue.

zevv, 2023-02-28:
I hit this trying to build our idf 4.2 based software on latest Debian; from the above thread I was not able to deduce what the proper workaround for this is, apart from setting up a whole new old-debian-version-chroot for building. *Is* there a way around this?

mcdavsco, 2023-02-28:
> I hit this trying to build our idf 4.2 based software on latest Debian; from the above thread I was not able to deduce what the proper workaround for this is, apart from setting up a whole new old-debian-version-chroot for building. _Is_ there a way around this?

On the v4.3 branch, I applied commit [3974be7](https://github.com/espressif/esp-idf/commit/3974be7fec1ea6c529ecbee795c0152b42b61d55) referenced above to allow the tools installation to complete. Then I can compile no problem on Fedora 37 which ships with Python 3.11.

You may be able to perform something similar for the v4.2 branch.

zevv, 2023-02-28:
Thank you @mcdavsco , that's helpful!

dobairoland, 2023-02-28:
I'll try to backport the fix into the v4.3 and v4.2 branches.

kutukvpavel, 2023-02-28:
I'm running into this stuff after installing latest updates today on Debian 12 + VSCode extension (IDF v4.4.4).
First faced compilation failures and PEP668 consequences, then ultimately hit a wall with gevent wheels.
Tried reinstalling all the tools from scratch, and now extension's installer just gets stuck on "Installing Python virtual environment step", not doing anything:

```
bin/sh: 1: /home/paul/.espressif/python_env/idf4.4_py3.10_env/bin/python: not found
Command failed: "/home/paul/.espressif/python_env/idf4.4_py3.10_env/bin/python" -m pip install --upgrade --constraint "/home/paul/.vscode/extensions/espressif.esp-idf-extension-1.6.0/espidf.constraints.txt" --no-warn-script-location  -r "/home/paul/.vscode/extensions/espressif.esp-idf-extension-1.6.0/requirements.txt"
/bin/sh: 1: /home/paul/.espressif/python_env/idf4.4_py3.10_env/bin/python: not found
```

Where is this all coming from? Why is it trying to create 3.10 venv on a 3.11 system?

zevv, 2023-02-28:
> Where is this all coming from? Why is it trying to create 3.10 venv on a 3.11 system?

I ran into a mess as well, I think part of it was related to what was already in my `~/.espressif`. When I trow the whole dir away, `install.sh` runs clean until hitting the gevent issue.

dobairoland, 2023-02-28:
> Where is this all coming from? Why is it trying to create 3.10 venv on a 3.11 system?

I assume the update you run for the Debian updated Python from 3.10 and 3.11. Vscode works differently as the issue reported here. You will have run the Vscode setup again. The ESP-IDF installer won't help in this case. Please see the Vscode documentation.

kutukvpavel, 2023-02-28:
Ok, I removed both ./espressif and esp-idf directories, uninstalled VSCode extension, and then tried to install everything from scratch again. Extension installer still stuck with the same console output. No python_env folder is even created. That's probably a separate extension-related bug.

However, I managed to get IDF's install.sh working somehow. Prior to launching it I tried to install python3-gevent package system-wide, and obviously succeeded. Don't know if it has something to do with the end result, though. Then I wired 3.11 venv folder to VSCode extension.

All in all, I think I introduced following changes: 
 - Switched from IDF v4.4.3 to 4.4.4 during a clean reinstall
 - Installed gevent using APT (probably unrelated)
 - Found that VSCode extension has a separate setting pointing to the expected venv directory, and by default it expects python 3.10... Duh. Changed 0 to 1 and now I can compile.

Also I am now confused about the meaning of that drop-down on the VSCode installer screen, that lets you choose python path (`/usr/bin/python`, `/bin/python`, etc). What does it change, if python is always invoked out of a venv?

madskjeldgaard, 2023-03-07:
I'm having the same issue on MacOS 13.0.1 (22A400) on Apple Silicon if it's any help.

mykmelez, 2023-03-15:
System Python is 3.9.6 on both my Apple Silicon Mac running macOS Monterey 12.6.3 and my Intel Mac running macOS Ventura 13.2.1. However, I did experience this issue after linking Homebrew Python 3.11 (`brew link python`). I worked around it by unlinking Homebrew Python (`brew unlink python`).

malapropistic, 2023-03-20:
Thanks @mykmelez, that helped me get around this issue!

mikkeldamsgaard, 2023-03-24:
For me, I had to do these steps to make it work:
```
rm -rf ~/.espressif
brew link python@3.9
$IDF_PATH/install.sh
```

semihkaya0941, 2023-03-24:
@madskjeldgaard Did you find any development? Have you been able to find anything new about the topic? 

i am having the same problem after `./install.sh esp32` for esp-idf v4.4. I am getting such errors: 
`      src/gevent/_greenlet_primitives.c:216:12: fatal error: 'longintrepr.h' file not found
        #include "longintrepr.h"
                 ^~~~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gevent
Failed to build gevent
ERROR: Could not build wheels for gevent, which is required to install pyproject.toml-based projects`

Although I deleted and reinstalled many times, I always had problems in the same place.

dobairoland, 2023-03-24:
Please see the linked fixes in each release branch. If you are still experiencing this then you probably don't use the latest (bugfix) version of the release.

pkkrusty, 2023-06-08:
Ran into this problem today on MacOS 12.6.1, where the install script chokes on gevent.  Ran the command `brew unlink python` and script ran with no errors after falling back to system python of 3.9.6.

mattjperez, 2023-06-13:
Thanks @pkkrusty, this worked for me as well.

hermlon, 2023-06-16:
Just if anyone runs into this issue like me: the fix hasn't been published on the stable release tagged with `v4.4.2` etc yet, instead you have to checkout the release branches: `git checkout release/v4.4`. See how to switch to a release branch here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-a-release-branch