espressif/esp-idf #10120: VSCode ESP-IDF lvgl 例程编译报错

Yinke7, 2022-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

在VSCode上编译 lvgl 例程

### What is the actual behavior?

在VSCode上编译 lvgl 例程，编译失败

### Steps to reproduce.

1、打开VSCode插件 ESP-IDF ，
2、复制并创建例程 "lvgl"，
3、选择target 为 "esp32"，点击"build"，
4、开始编译。。。
5、结果编译不通过！log如下

### Build or installation Logs.

```plain
*  正在执行任务: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. 

-- Found Git: D:/ESP-IDF/espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/ESP-IDF/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/ESP-IDF/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/ESP-IDF/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] idf (4.4.2)      
[2/2] lvgl/lvgl (8.0.2)
CMake Error at D:/ESP-IDF/esp-idf/tools/cmake/build.cmake:454 (message):
  Traceback (most recent call last):

    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connection.py", line 174, in _new_conn
      conn = connection.create_connection(
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\util\connection.py", line 72, in create_connection
      for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
    File "socket.py", line 918, in getaddrinfo

  socket.gaierror: [Errno 11001] getaddrinfo failed

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):

    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connectionpool.py", line 703, in urlopen
      httplib_response = self._make_request(
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connectionpool.py", line 386, in _make_request
      self._validate_conn(conn)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connectionpool.py", line 1042, in _validate_conn
      conn.connect()
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connection.py", line 358, in connect
      self.sock = conn = self._new_conn()
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connection.py", line 186, in _new_conn
      raise NewConnectionError(

  urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection
  object at 0x0000017EEF2350D0>: Failed to establish a new connection: [Errno
  11001] getaddrinfo failed

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):

    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\requests\adapters.py", line 489, in send
      resp = conn.urlopen(
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\connectionpool.py", line 787, in urlopen
      retries = retries.increment(
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\urllib3\util\retry.py", line 592, in increment
      raise MaxRetryError(_pool, url, error or ResponseError(cause))

  urllib3.exceptions.MaxRetryError:
  HTTPSConnectionPool(host='components-file.espressif.com', port=443): Max
  retries exceeded with url: /6fdc1d19-9f86-4bb2-b0f2-09850aaa7ad5.tgz
  (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object
  at 0x0000017EEF2350D0>: Failed to establish a new connection: [Errno 11001]
  getaddrinfo failed'))

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 3, in <module>
      main()
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 126, in main
      args.func(args)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 39, in prepare_dep_dirs
      ComponentManager(args.project_dir).prepare_dep_dirs(
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 404, in prepare_dep_dirs
      downloaded_component_paths = download_project_dependencies(
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 128, in download_project_dependencies
      download_paths = fetcher.download()
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\fetcher.py", line 51, in download
      return self.source.download(self.component, self.managed_path)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\web_service.py", line 139, in download
      with requests.get(url, stream=True, allow_redirects=True) as r:
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\requests\api.py", line 73, in get
      return request("get", url, params=params, **kwargs)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\requests\api.py", line 59, in request
      return session.request(method=method, url=url, **kwargs)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\requests\sessions.py", line 587, in request
      resp = self.send(prep, **send_kwargs)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\requests\sessions.py", line 701, in send
      r = adapter.send(request, **kwargs)
    File "d:\ESP-IDF\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\requests\adapters.py", line 565, in send
      raise ConnectionError(e, request=request)

  requests.exceptions.ConnectionError:
  HTTPSConnectionPool(host='components-file.espressif.com', port=443): Max
  retries exceeded with url: /6fdc1d19-9f86-4bb2-b0f2-09850aaa7ad5.tgz
  (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object
  at 0x0000017EEF2350D0>: Failed to establish a new connection: [Errno 11001]
  getaddrinfo failed'))

Call Stack (most recent call first):
  D:/ESP-IDF/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "D:/VSCode_Pro/ESP-IDF/lvgl/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 ..”已终止，退出代码: 1。
```

suda-morris, 2022-11-07:
seems a network issue happened when downloading the LvLG library from the component registry. Can retry help?

Yinke7, 2022-11-07:
> seems a network issue happened when downloading the LvLG library from the component registry. Can retry help?
我已经重试过，从打开ESP-IDF插件的实例例程，再复制创建项目，再重新编译，编译结果log显示是一样的。

closed 2024-07-01 by suda-morris