espressif/esp-idf #10122: SOURCES of "__idf_xtensa" use a generator expression that depends on the SOURCES themselves

AndruePeters, 2022-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1626-g4b6d9c8ad3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### What is the expected behavior?

The project should build

### What is the actual behavior?

When running in a pure CMake environment: 
[image]

This also fails to compile with idf.py. But for some reason, idf.py doesn't pass in the toolchain? 

### Steps to reproduce.

Clone: `https://github.com/AndruePeters/esp_idf_cmake_issue`
Run with included `build-esp32.sh` that was copied from `examples/build_system/cmake/idf_as_lib/`

Also, try running with idf.py

### More Information.

This structure I'm using was working back in May of 2022. I have no idea what's happening here. Also, idf.py doesn't seem to pass the toolchain file when calling CMake? 

Can someone also explain better how `idf_build_process` works? I'm really not a fan of the structure of this macro.
There should be a better way where I can point esp targets. Also, should I have an idf_build_process per CMakeLists.txt file, or have it once at the top level? In the real codebase, there's a mix of libraries where some reference esp-idf libraries and others don't.

igrr, 2022-11-10:
@AndruePeters I think the problem you are running into is that we assume that the executable target is defined in the same directory as where idf_build_process is called. The workaround is to move `add_executable` call to the root CMakeLists.txt file.

The issue is described here: https://github.com/espressif/esp-idf/issues/7482.

igrr, 2022-11-10:
> Surprised it appeared to work a few months ago, but my memory might not be the best in that case.

We'll have to find the root cause of this issue; depending on the root cause, there may be more than one factor necessary to trigger this issue. Plus, possibly something has actually changed in the recent IDF versions. Since we don't have a test for such use case, it's possible that there is an actual regression. If this is not some kind of a fundamental issue which would require significant changes in the build system, we will fix it! Sorry for the trouble.

> There should be a way to expose the libraries without calling idf_build_process. But, what are the reasons for doing it this way? With a library like boost, you typically call find_package, list your requirements, and then can link against them target_link_libraries(target PRIVATE Boost::filesystem) and whatnot.

Thank you for the suggestion!

I think it's possible to wrap the generation of IDF build targets so that you don't have to call idf_build_process. However that would prevent one from passing some arguments to the IDF build system: things like SDKCONFIG, SDKCONFIG_DEFAULTS, COMPONENTS and IDF_TARGET. That in turn can be solved by introducing a "configuration package" ([Zephyr does that](https://docs.zephyrproject.org/3.0.0/guides/zephyr_cmake_package.html#zephyr-build-configuration-cmake-package)), or by passing them via global variables; but I personally don't think either of these options is cleaner or more user-friendly solution than a function (macro) call.  Will look into it a bit more; `find_package` support has been on our backlog for a while, perhaps we can finally do it in the next major release.

AndruePeters, 2022-11-20:
What is the purpose of SDKConfig? I don't see any documentation for it.

zikalino, 2023-11-29:
closing the ticket, as it's quite old and problem was already discussed and explained.

closed 2023-11-29 by zikalino

Ruanfc, 2024-01-01:
I'm having the same issue. Is there any branch where this problem is solved?