espressif/esp-idf #10125: Modbus Serial (RTU) assigns data from last recent poll instead current one

MichaelUray, 2022-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

aiThinker ESP32s

### Power Supply used.

USB

### What is the expected behavior?

The function `mbc_master_get_parameter()` should read the data of the characteristic defined in the current parameter and should not return the result from the last recent parameter poll.

### What is the actual behavior?

The function `mbc_master_get_parameter()` reads the value from its last reset parameter into the current one.

This is the code section from the official [Modbus serial master example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/serial/mb_master) from the ESP IDF.
It queries `input register`, `holding register` alternating

```
    { CID_INP_DATA_0, STR("Data_channel_0"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, 2,
                    INPUT_OFFSET(input_data0), PARAM_TYPE_FLOAT, 4, OPTS( -10, 10, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_0, STR("Humidity_1"), STR("%rH"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 0, 2,
            HOLD_OFFSET(holding_data0), PARAM_TYPE_FLOAT, 4, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_INP_DATA_1, STR("Temperature_1"), STR("C"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 2, 2,
            INPUT_OFFSET(input_data1), PARAM_TYPE_FLOAT, 4, OPTS( -40, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_1, STR("Humidity_2"), STR("%rH"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 2, 2,
            HOLD_OFFSET(holding_data1), PARAM_TYPE_FLOAT, 4, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_INP_DATA_2, STR("Temperature_2"), STR("C"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 4, 2,
            INPUT_OFFSET(input_data2), PARAM_TYPE_FLOAT, 4, OPTS( -40, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_2, STR("Humidity_3"), STR("%rH"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 4, 2,
            HOLD_OFFSET(holding_data2), PARAM_TYPE_FLOAT, 4, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
...
```

and prints out the result then via `ESP_LOGI()`.
```
                    err = mbc_master_get_parameter(cid, (char*)param_descriptor->param_key,
                                                        (uint8_t*)&value, &type);
                    if (err == ESP_OK) {
                        *(float*)temp_data_ptr = value;
                        if ((param_descriptor->mb_param_type == MB_PARAM_HOLDING) ||
                            (param_descriptor->mb_param_type == MB_PARAM_INPUT)) {
                            ESP_LOGI(TAG, "Characteristic #%d %s (%s) value = %f (0x%x) read successful.",
                                            param_descriptor->cid,
                                            (char*)param_descriptor->param_key,
                                            (char*)param_descriptor->param_units,
                                            value,
                                            *(uint32_t*)temp_data_ptr);
                            if (((value > param_descriptor->param_opts.max) ||
                                (value < param_descriptor->param_opts.min))) {
                                    alarm_state = true;
[... 69 lines cut ...]
E (5266) MB_CONTROLLER_MASTER: mbc_master_set_parameter(146): Master set parameter failure, error=(0x107) (ESP_ERR_TIMEOUT).
E (5276) MASTER_TEST: Characteristic #6 (Test_regs) write fail, err = 0x107 (ESP_ERR_TIMEOUT).
E (5526) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure.
E (5526) MB_CONTROLLER_MASTER: mbc_master_get_parameter(85): Master get parameter failure, error=(0x107) (ESP_ERR_TIMEOUT).
E (5536) MASTER_TEST: Characteristic #7 (RelayP1) read fail, err = 0x107 (ESP_ERR_TIMEOUT).
I (5746) MASTER_TEST: Characteristic #8 RelayP2 (on/off) value = OFF (0x0) read successful.
I (6446) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 0.000000 (0x0) read successful.
I (6646) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 1.000000 (0x3f800000) read successful.
I (6846) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 0.000000 (0x0) read successful.
I (7046) MASTER_TEST: Characteristic #3 Humidity_2 (%rH) value = 2.000000 (0x40000000) read successful.
I (7246) MASTER_TEST: Characteristic #4 Temperature_2 (C) value = 0.000000 (0x0) read successful.
I (7446) MASTER_TEST: Characteristic #5 Humidity_3 (%rH) value = 3.000000 (0x40400000) read successful.
I (7646) MASTER_TEST: #6 Test_regs (__)
E (7846) MB_CONTROLLER_MASTER: mbc_master_get_parameter(85): Master get parameter failure, error=(0x107) (ESP_ERR_TIMEOUT).
E (7846) MASTER_TEST: Characteristic #7 (RelayP1) read fail, err = 0x107 (ESP_ERR_TIMEOUT).
E (8056) MB_CONTROLLER_MASTER: mbc_master_get_parameter(85): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (8056) MASTER_TEST: Characteristic #8 (RelayP2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
I (8766) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 0.000000 (0x0) read successful.
I (8966) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 1.000000 (0x3f800000) read successful.
I (9166) MASTER_TEST: Characteristic #2 Temperature_1 (C) value = 0.000000 (0x0) read successful.
I (9366) MASTER_TEST: Characteristic #3 Humidity_2 (%rH) value = 2.000000 (0x40000000) read successful.
I (9566) MASTER_TEST: Characteristic #4 Temperature_2 (C) value = 0.000000 (0x0) read successful.
I (9766) MASTER_TEST: Characteristic #5 Humidity_3 (%rH) value = 3.000000 (0x40400000) read successful.
I (9966) MASTER_TEST: #6 Test_regs (__)
```

alisitsyn, 2022-11-07:
Hello @MichaelUray,

Unfortunately v4.4 freemodbus component will not work with the configuration used in your application because your other tasks blocks the normal processing of modbus tasks.

Please try to fix this using the instruction below to check:
1. Create the components folder in the root folder of your application.
2. Copy the component folder "espressif__esp-modbus" from the project here[ https://github.com/alisitsyn/modbus_su ... components](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/components) to your "components" folder.
3. Add the line to your root CMakeLists.txt file: [https://github.com/alisitsyn/modbus_sup ... sts.txt#L6](https://github.com/alisitsyn/modbus_support/blob/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii/modbus_master_rtu/CMakeLists.txt#L6).
4. Change the task priority of `9` to `8`.
5. `idf.py fullclean` or delete the build folder of your application manually then rebuild the project.
6. Make sure the slave response time (idf.py menuconfig, CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND) is set to maximum slave response time.
7. Rerun your project.

Please also take a look to the migration guide notes for modbus:
[https://github.com/espressif/esp-idf/bl ... esp-modbus](https://github.com/espressif/esp-idf/blob/master/docs/en/migration-guides/release-5.x/5.0/protocols.rst#esp-modbus)

If you still have issues with the update please send the log here. In this case it may require retries also to work in your configuration.
Thank you.

MichaelUray, 2022-11-07:
Hi @alisitsyn, thanks for your response.

> Unfortunately v4.4 freemodbus component will not work with the configuration used in your application because your other tasks blocks the normal processing of modbus tasks.

What do you mean with other tasks block the normal processing of modbus tasks?
I used the official [ESP IDF serial Modbus master example](https://github.com/espressif/esp-idf/blob/4b6d9c8ad317dc9e45f3c7699525dd9479f1f4c7/examples/protocols/modbus/serial/mb_master/main/master.c#L305) and it does not start any tasks.

It only calls `master_operation_func()` but not anything else.

```
void app_main(void)
{
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK(master_init());
    vTaskDelay(10);

    master_operation_func(NULL);
}
```
I don't see other tasks there which could influence the Modbus communication and it is the official example which I used for my tests which causes this issue.

alisitsyn, 2022-11-08:
Hello @MichaelUray,

The comment about tasks is related to your question here: https://esp32.com/viewtopic.php?f=13&t=30434

I do not have a chance to install your Slave simulator software and for now can just guess what happens on your side. Note that the different tools can use different float format and even float 32 bit can be different. The example uses the float32 little-endian byte swap format. Please send your simulator log (modbus requests and responses) from your host software. The log for your case should look like below:
[Communication log4.txt](https://github.com/espressif/esp-idf/files/9962822/Communication.log4.txt)

I also recommend you to follow the instruction above to update the component. The updated component will allow to flush responses from slave that come after response timeout.

MichaelUray, 2022-11-08:
> The comment about tasks is related to your question here: https://esp32.com/viewtopic.php?f=13&t=30434

This there was a different issue, caused by a [too low stack from another task](https://forums.freertos.org/t/assert-failed-xmbmasterrtureceivefsm-mbrtu-m-c-235-with-modbus-master-from-esp-idf/16127/6).

> Note that the different tools can use different float format and even float 32 bit can be different (modbus standard do not prescribe this). The example uses the float32 little-endian byte swap format.

I am aware of the little and big endian matter, but this is not the problem.
If I send all data for the holding registers as 0, the convertation does not matter, but the 0 end up in the input register and the input register data end up in the holding register (with correct float conversation).

Please find there the (debug) logs from the ESP32 as well as the log from the Modbus simulator.
[ESP32 Modbus Log.txt](https://github.com/espressif/esp-idf/files/9963947/ESP32.Modbus.Log.txt)
[Modsim Log.TXT](https://github.com/espressif/esp-idf/files/9963949/Modsim.Log.TXT)

alisitsyn, 2022-11-10:
@MichaelUray ,

1. The freemodbus v4.4 component does not contain required fixes in the command handler and in main FSM. It can cause the issue with the incorrect command handling if the response to previous command comes after start of new tarnsaction.
2. v4.4 version also share some resources (timer group as an example) for master and slave code and this prevents to use serial master and slave in the same application. This can cause the issue with the incorrect events tracking.
The fixes were provided by patch files and were not merged correctly on time to v4.4. These fixes will not be back ported because the component `freemodbus` was removed from esp-idf from v5.0.

In order to fix the issues please try to follow the procedure above and repeat your tests.
Thanks.

MichaelUray, 2022-11-12:
> 1. if the response to previous command comes after start of new transaction (timeout was happened on previous transaction due to incorrect maximum response time setting)

Thanks for pointing that out, to extend the timout helped to get rid of this issue.
In my opinion the Modbus library should check if the response function-code fits to its query function-code as well as its size.

I also updated to the other version which you recommended to have the buffer flushed after a timeout if it happens.

> These fixes will not be back ported because the component

Is it not the idea, to get things like this fixed within ESP IDF Service Period, in which V4.4 is still in?

alisitsyn, 2023-01-12:
@MichaelUray ,

Thank you for update.

> Is it not the idea, to get things like this fixed within ESP IDF Service Period, in which V4.4 is still in?

I agree with you it is not good idea but was an exception. However, the modbus component in esp-idf v4.4 has been updated using the subrepo and reflects the changes I mentioned above. Please try it and report if you still have the same issue.

MichaelUray, 2023-01-14:
Thank you for your response.
I am not able to test it anymore since I moved in the meanwhile over to the [modbus-esp-cpp](https://github.com/plasmapper/modbus-esp-cpp) library which is much more convenient to me.

closed 2023-01-14 by MichaelUray