espressif/esp-idf #10126: How to run UNIT TEST for a particular component

krupis, 2022-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I have been reading and learning about unit tests lately and wanted to setup a very simple project with my very first unit test.

I have followed these simple steps:

1. In my project, I have created a component called "test_component".

2. In this component directory, I have created a directory called "test" as explained in the esp.idf instructions.
3. In this "test" directory, I have test.c and CMakeLists.txt
4. I have enabled "Include Unity test fixture" in menuconfig

My test_component.c is as following:
[code]

#include <stdio.h>
#include "test_component.h"

int sum(int a, int b){
    return a+b;
}

[/code]

As you can see from above, the component contains 1 function that I want to run unit test for.

My test.c is as following:
#include "test_component.h"
#include "unity.h"

// test sum function
TEST_CASE("sum function test", "[sum]")
{
      TEST_ASSERT_EQUAL_INT(sum(1,2),3);
}

My test CMakeLists.txt:
[code]
idf_component_register(SRC_DIRS "."
                       INCLUDE_DIRS "."
                       REQUIRES unity test_component)
[/code]

My project structure is as following:
[image]

**My problem:**

It is not fully clear to me how can I run this test. Could someone help me understand how can I execute this test?

igrr, 2022-11-07:
Hi @krupis, please refer to this example: https://github.com/espressif/esp-idf/tree/master/examples/system/unit_test

igrr, 2022-11-22:
I'll close this as I think the question has been answered. Please leave a comment if something is unclear!

closed 2022-11-22 by igrr