espressif/esp-idf #10128: Using RMT as UART

StarOfSlytherin, 2022-11-07:
### Is your feature request related to a problem?

My application requires 5 UART ports. So I have a shortage of Hardware UART ports, Software UART would be helpful in my situation. I know software UART will have issues in high baud rate scenarios but my use case is 4800 & 9600 on software UART.

I found a library on using RMT as UART but it has a few drawbacks. The most important for me is the lack of support for Even parity. Here's the link to the library: https://github.com/naffej/esp32-rmt-uart

I have searched a lot for this and seen a few people looking for this feature on the ESP32 forums, Reddit etc. Hope you consider adding an example or a component as a feature.

### Describe the solution you'd like.

A Software Serial component or an example to use RMT as Software UART

### Describe alternatives you've considered.

I couldn't find any software solutions but I have considered using a Mux IC to switch multiple serial ports to a single GPIO pin. The issue with this is I'll have to de-init and re-init the UART driver to change the baud rate, parity etc.

suda-morris, 2022-11-08:
Thanks for the idea, we will consider adding such an example in the future. Implement the UART protocol via RMT is equivalent to writing an [encoder](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#rmt-encoder) to translate the user buffer into RMT data format. Actually we have already provided many of them (e.g. NEC, One-Wire), hope you can get inspired from [them](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt)

BTW, in some ESP series (e.g. ESP32C3, ESP32S3), we also provided [CPU instructions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/dedic_gpio.html) to manipulate the GPIO, which means you can implement the UART protocol by writing **assembly code**. We're currently preparing such examples internally.

StarOfSlytherin, 2022-11-08:
Thanks for the reply. I'll see if I can write a UART encoder but in any case can I expect the examples with v5.0 release?

StarOfSlytherin, 2022-12-05:
I just saw the dedicated GPIO examples. Is it restricted to RISC-V or will there be support for Xtensa CPUs later?