espressif/esp-idf #10129: esptool flash programming: Wait for completion of flash encryption (ESPTOOL-534)

M-Bab, 2022-11-07:


AdityaHPatwardhan, 2023-09-05:
Hi @M-Bab Thanks for the issue.

This is indeed a limitation related to flash encryption. The problem here is not esptool but the large time required for encryption of the flash contents. Please note that this increased time is due to the limitation of the Flash Encryption Hardware. The hardware can only encrypt 256 bits at a time and then write it to respective place. This bottleneck cannot be solved as of now considering it is a hardware bottleneck. This causes the problem of reliability as you have mentioned in the Flash Encryption process. When Flash Encryption is enabled the traditional way i.e. through the bootloader. Each partition is encrypted on the device (esp module) itself before writing to the flash. The time required to flash would increase as the firmware size is increased.

An alternative to this, we have added a documentation for a Host based workflow to enable the Flash Encryption for a module. It has its own set of conditions but with proper management of critical resources this workflow can be implemented at the factory for increased reliability and faster flashing time.
Please find the details at this [link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally)

I think this should resolve your issue.
Thanks,
Aditya

M-Bab, 2023-09-05:
@AdityaHPatwardhan Oh I don't have any problem with the time it needs for the flash encryption. This is totally fine!

My problem with the esptool is that it _returns as completed_ while there are ongoing processes in the controller. This could lead to bricked esp's, if they are reset or unplugged or similar.

This issue requests for a delay or an autodetection to ensure the flash encryption is actually completed by the time the esptool returns.