espressif/esp-idf #10131: WiFi prov suitable for WiFiMulti

podaen, 2022-11-07:
### Is your feature request related to a problem?

The configuration of the wifi network needs to be stored:out for WiFiMulti to connect to the best AP based on a given list and can only be done by IDF for security reasons.

### Describe the solution you'd like.

Now the [restore wifi](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_restorev), restores the first connected wifi network configuration. That is not a problem an can been overseen by setting evrything before or after `esp_wifi_start()`. Note: that this takes a week to configure. The problem is that the ssid and password are not been saved with the same encryption as the the first stored configuration and can be hacked out of the memory if it is not stored properly by default and if you save an other netwoork you need to save all other settings too. My proposal is to check the mac adress and save all other settings if it has been saved. And I think this is allready done. Now I don't see the option to save all networks in to memory. I am thinking of to make esp_wifi_prov() suitable to MultiWiFi.

### Describe alternatives you've considered.

The [documentation ](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/storage/nvs_flash.html?highlight=nvs)describes the ssid and password are encrypted, but doesn't tell how they are encrypted. It is also a long comment to do, so I suggest it has to be done by the api. 

### Additional context.

All the saved preferences of the wifi network found in the documentation.

out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv417esp_wifi_get_modeP11wifi_mode_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_restorev
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_scan_startPK18wifi_scan_config_tb
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_scan_get_ap_numP8uint16_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv428esp_wifi_scan_get_ap_recordsP8uint16_tP16wifi_ap_record_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_get_psP14wifi_ps_type_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_get_protocol16wifi_interface_tP7uint8_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv422esp_wifi_get_bandwidth16wifi_interface_tP16wifi_bandwidth_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_get_channelP7uint8_tP18wifi_second_chan_t
flash	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_countryPK14wifi_country_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_get_mac16wifi_interface_tAL6E_7uint8_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv424esp_wifi_get_promiscuousPb
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv431esp_wifi_get_promiscuous_filterP25wifi_promiscuous_filter_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv436esp_wifi_get_promiscuous_ctrl_filterP25wifi_promiscuous_filter_t
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv423esp_wifi_ap_get_sta_aidAL6E_K7uint8_tP8uint16_t
flash	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_country_codePKcb
out	https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_get_countryP14wifi_country_t

podaen, 2022-11-07:
I even don't know if it is erase it from the memory, if the connection has been established. :-)

Xiehanxin, 2022-11-08:
hi @podaen, thanks for your report, I think your need is restore a list of SSID&PASSWORD, and try to connect one of them. Maybe you can build a structure including the SSID, password and write a function, which can try and connect the APs one by one. 
If you have more needs, please let me know

podaen, 2022-11-08:
Yeah, but it doesn't look pretty. You will need to declare multiple objects (to many than needed) of structure with a fix size and it still will be not encrypted.

podaen, 2022-11-10:
If I want to use an structure, I just can use the Arduino.

I read the nvs out for the ssid I alwase get "-"

```
wifi_config_t wifi_config_stored;
static uint8_t wifi_flag = 0;
static EventGroupHandle_t wifi_event_group = NULL;

static void read_wifi_config(void)
{
    nvs_handle handle;
    esp_err_t err;
    static const char* NVS_NAME = "nvs.net80211";
    static const char* KEY = "sta.ssid"; 
    memset(&wifi_config_stored, 0x0, sizeof(wifi_config_stored));
    uint32_t len = sizeof(wifi_config_stored);
    err = nvs_open(NVS_NAME, NVS_READWRITE, &handle);
    if (err != 0) {
        ESP_LOGV("TAG", "nvs_get_blob('%s') failed: %s", NVS_NAME, esp_err_to_name(err));
        return;
    }
    err = nvs_get_blob(handle, KEY, &wifi_config_stored.sta.ssid, &len);
    if (err == ESP_ERR_NVS_NOT_FOUND)
    {
        ESP_LOGV("TAG", "nvs_get_blob('%s'): %s - check key", KEY, esp_err_to_name(err));
        wifi_flag = 0;
    }
    else if (err == ESP_OK)
    {
        ESP_LOGV("TAG", "nvs_get_blob ok");
        wifi_flag = 1;
        nvs_close(handle);
        printf("ssid:%s\n", wifi_config_stored.sta.ssid);
    }
}
```

Why is that?

podaen, 2022-11-11:
If I use the esp_wifi_get_config(), I need to initialize the wifi first. That's about 800 lines of code I did to day only to get my ssid and password. There must be a shorter way like this to get that. 

```
    const char* partitionname = "nvs";
    const char* name = "nvs.net80211";
    const char* key = "sta.ssid";
    const char* key1 = "sta";
    esp_err_t ret;
    nvs_entry_info_t nvs_info;
    nvs_iterator_t nvs_it;
    nvs_handle_t nvs_handle;
    uint8_t ssid[36];
    
    //ini wifi doesn't help

    nvs_flash_init();

    ret = nvs_open(name, NVS_READWRITE, &nvs_handle);
    //ret = nvs_open_from_partition(partitionname, name, NVS_READONLY, &nvs_handle);
     if (ret != ESP_OK) {
         ESP_LOGV("TAG", "Failed to open NVS('%s'), check key: %s", key, esp_err_to_name(ret));
         return;
     }
    size_t actual_len;
    ret = nvs_get_blob(nvs_handle, key, nullptr, &actual_len);
    if (ret != 0) {
        ESP_LOGV("TAG", "nvs_get_blob('%s'), check key: %s", key, esp_err_to_name(ret));
        return;
    }
    size_t len = sizeof(ssid);
    ret = nvs_get_blob(nvs_handle, key, ssid, &len);
    if (ret != 0) {
        ESP_LOGV("TAG", "nvs_get_blob('%s'), failed: %s", key, esp_err_to_name(ret));
        return;
    }
    ESP_LOGV("TAG", "result('%s')", ssid);
```

> Output: -

Xiehanxin, 2022-11-11:
hi, @podaen, so you don't just want to store the ssid and password simply, but want to restore the whole information in the NVS? my initial plan is when you try to connect to an AP, you need to send the SSID and PASSWORD to the wifi configure and such information you can input and need not be encryped.

podaen, 2022-11-11:
I was looking if I could get the ssid and pass. Now I know how the wifi works. Then I tried to get the credentials before initializing the wifi, what I know out of the documentation that it was not possible!

What I want is that the api can stores multi encrypted wifi configurations in the nvs partition that get its key in the encrypted partition. To get the last used config  `esp_err_t esp_wifi_get_config(wifi_interface_t interface, wifi_config_t *conf);` is still valid and to get the config on from a mac would be `esp_err_t esp_wifi_get_config(const char* mac, wifi_interface_t interface, wifi_config_t *conf);` after initializing the wifi. Every AP has an unique mac address and can be scanned for before initializing the wifi. I don't know what happens within the api, but I think this is very possible to do. It can always limit the amount of configurations that can be stored within the api. I hope this helps!

Xiehanxin, 2022-11-17:
hi @podaen,  sorry for late reply, I would like to confirm that you wants API to be able to safely hand over encrypted credentials, and could you tell me about your specific application scenario ? I will discuss with others then give you a reply

closed 2023-06-13 by Alvin1Zhang