espressif/esp-idf #10134: ESP32S3 OpenThreadBR does not advertise OMR prefix

ble0, 2022-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

OMR prefix is published via Router Advertisement of ESP32S3

### What is the actual behavior?

ESP32S3 does not advertise OMR prefix

### Steps to reproduce.

1. Follow the steps in https://openthread.io/guides/border-router/espressif-esp32 to setup ESP32S3 and ESP32H2
2. Configure the correct WiFi AP using menuconfig of ot_br project
3. Compile and load ot_br and ot_rcp
4. Wait for ICMPv6 Router Advertisement

### More Information.

[log.txt](https://github.com/espressif/esp-idf/files/9959290/log.txt)
[packet_cap.txt](https://github.com/espressif/esp-idf/files/9959294/packet_cap.txt) - rename it to .pcapng

1. I tried esptool.py erase_flash and flashing esp32s3 again but it does not work
2.

gytxxsy, 2022-11-08:
I have tested with esp32s3 and esp32h2 but did not encountered the problems mentioned above.

You can find packets in packets.txt (rename it to .pcapng). The Router Advertisement packets are No.51 No.148 No.362 No.548 No.758 No.952. The packets show that the RA packets are sent successfully and all of them have omr prefix (fd15:66a3:ece:1::/64). You can also use display filters 'icmpv6.type == 134' to find the Router Advertisement packets quickly.

I also check the packet_cap.pcapng. It looks like the Router Advertisement packet (No.2) has the OMR prefix (refer to omrprefix.txt and rename it to .png). I think it conflicts with "ESP32S3 does not advertise OMR prefix". Did I understand you wrong, or you want to say "ESP32S3 does not advertise OMR prefix after working for a while"？

The commit ID of esp-idf I used is 9da7deaa7cb51c5e29f1b4ef15614ac949cb8899. I think it is a great idea to try again with this commit. I hope that it can help you to solve the problem. 
[packets.txt](https://github.com/espressif/esp-idf/files/9961710/packets.txt)
[omrprefix.txt](https://github.com/espressif/esp-idf/files/9961727/omrprefix.txt)

ble0, 2022-11-08:
The commit is not present in v5.1-dev tag, I will try again using the master

ble0, 2022-11-14:
the issue is not observed in the master

closed 2022-11-14 by ble0