espressif/esp-idf #10135: [Discussion] Remove the support of ADC2 digital controller on ESP32-S3 and ESP32-C3

ginkgm, 2022-11-08:
### Is your feature request related to a problem?

To be modified.

bbinet, 2023-02-22:
Could you please explain in more details what's wrong with ADC2?
As we currently have boards for which GPIO05 is hardwired to measure external analog voltage, what is the actual risk if we continue to use ADC2 using CONFIG_ADC_ONESHOT_FORCE_USE_ADC2_ON_C3 option? 
Until now, we do not experience any issue with ADC2 here, and we would prefer to avoid a hardware redesign of the board...

bbinet, 2023-02-23:
Ok, so the only risk is to get some wrong values (raw value equals 0) from time to time?
So I could ignore values which equals 0, and I'll be fine, right?

Icarus113, 2023-02-24:
Hi, @bbinet , the 0 result is just one of the potential issues, it appears on some chips. Generally speaking, the result isn't reliable. 

> So I could ignore values which equals 0, and I'll be fine, right?

I would say no. We can't guarantee other results are correct.

bbinet, 2023-02-24:
Thank you @Icarus113 for this information.

Icarus113, 2023-03-27:
Hi watchers, the code changes for this issue have been merged in to all related releases.

If there's no other comments, I'm gonna close this issue in 1 day.

closed 2023-03-29 by Icarus113

moefear85, 2023-07-02:
so what this means is, if continuous sampling adc1 and adc2 simultaneously, then even if sampling starts fine... at any random moment, a spurious `sampling enable` may happen, that then causes the entire DMA controller to crash, hence making all read attempts after that fail?

If this is correct, then, if user code determines such a failed state has been entered, is it possible to simply reconfigure the adc and start over, or would it require a restart of the entire chip?

Icarus113, 2023-07-03:
> so what this means is, if continuous sampling adc1 and adc2 simultaneously, then even if sampling starts fine... at any random moment, a spurious sampling enable may happen, that then causes the entire DMA controller to crash, hence making all read attempts after that fail?

Actually I don't actually know if this will be one of the conditions. (I didn't meet such scenario yet).

> if user code determines such a failed state has been entered, is it possible to simply reconfigure the adc and start over, or would it require a restart of the entire chip?

The problem is that, there's no flag or similar thing that indicating issue happens. And you can't recover it by reconfigure adc or restart the whole chip. The issue is always there.

@moefear85

moefear85, 2023-07-03:
strange... either way please leave the functionality available if possible, ie don't deprecate it. I find it very useful and the results I'm getting seem fine. Hopefully it might be fixed in a future revision of the chip.

devitalii, 2024-01-11:
so, ok how do I actually use A5  analog pin under ADC1 ?