espressif/esp-idf #10136: assert function has non-standard behaviour

dbahrdt, 2022-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The commit https://github.com/espressif/esp-idf/commit/9afdf547481e4c5e8683f6efc8561a34c49083cc
 introduced a change to https://github.com/espressif/esp-idf/blob/4b6d9c8ad317dc9e45f3c7699525dd9479f1f4c7/components/newlib/platform_include/assert.h#L34

The assert function executes the function even if NDEBUG is set. However according to http://port70.net/~nsz/c/c11/n1570.html#7.2 the assert macro has the following behavior:

> [1](http://port70.net/~nsz/c/c11/n1570.html#7.2p1) The header [<assert.h>](http://port70.net/~nsz/c/c11/n1570.html#7.2) defines the assert and static_assert macros and refers to another macro,
> 
>          NDEBUG
> which is not defined by [<assert.h>](http://port70.net/~nsz/c/c11/n1570.html#7.2). If NDEBUG is defined as a macro name at the point in the source file where [<assert.h>](http://port70.net/~nsz/c/c11/n1570.html#7.2) is included, the assert macro is defined simply as
>          #define assert(ignore) ((void)0)

The esp-idf documentation even has a warning about side effects of functions executed in assert() macro.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/style-guide.html#assertions

igrr, 2022-11-08:
This looks like the same issue as reported in https://github.com/espressif/esp-idf/issues/2758.

Unfortunately we have missed this issue when planning the breaking changes in v5.0 release, and now it's a bit too late to make it.

We can consider providing some workaround to get the correct behavior on demand, and switch over to the correct behavior by default in v6.0.
I suggest to close this issue and continue the discussion in the linked one.

dbahrdt, 2022-11-10:
> This looks like the same issue as reported in #2758.

Kind of looks like it. Though the assert defined in #2758 seems to have a different definition.

> Unfortunately we have missed this issue when planning the breaking changes in v5.0 release, and now it's a bit too late to make it.
> 
> We can consider providing some workaround to get the correct behavior on demand, and switch over to the correct behavior by default in v6.0.

That would be nice since this behavior breaks existing code that was not written specifically for esp-idf.

> I suggest to close this issue and continue the discussion in the linked one.
That's fine by me.

closed 2024-08-26 by espressif-bot