espressif/esp-idf #10137: ESP32S3 USB CDC Communication

Darshnik253, 2022-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Currently working on ESP32S3 Dev Kit C1, it is working as a USB Host and another ESP32S3 Dev kit C1 as a USB Device.
Using example of CDC_ACM_HOST version 4.4.2, and trying to do usb cdc communication between them.

But the issue is, when i connected USB device with USB host then host recognized device and generate NEW_Device_Connected event and USB CDC communication perfectly works. But whenever i disconnect it and reconnect usb device, then device recognized by USB Host with VID and PID but USB CDC Communication not working.

Means if i already connected device with host before uploading code to host then usb cdc communication works fine. Problem occurs in communication when device not connected before code flashing. 

I have go through the given link for new device detection
https://github.com/tore-espressif/esp-idf/commit/0428efa4ad6fd9e4fc9e2abcf58afae331bba57b

Reference Code is below,

void new_dev_cb(usb_device_handle_t usb_dev) {
    new_dev_cb_called = true;
    const usb_config_desc_t *config_desc;
    const usb_device_desc_t *device_desc;

    // Get descriptors
    ESP_ERROR_CHECK(usb_host_get_device_descriptor(usb_dev, &device_desc));
    ESP_ERROR_CHECK(usb_host_get_active_config_descriptor(usb_dev, &config_desc));

    ESP_LOGI(USB_CDC_TAG, "New device connected. VID = 0x%04X PID = 0x%04X\n", device_desc->idVendor, device_desc->idProduct);
}

static void usb_event_cb(const usb_host_client_event_msg_t *event_msg, void *arg)
{
    switch (event_msg->event) {
    case USB_HOST_CLIENT_EVENT_NEW_DEV:
        ESP_LOGI(TAG, "New device connected");
        if (p_cdc_acm_obj->new_dev_cb) {
            usb_device_handle_t new_dev;
            if (usb_host_device_open(p_cdc_acm_obj->cdc_acm_client_hdl, event_msg->new_dev.address, &new_dev) != ESP_OK) {
                ESP_LOGI(TAG, "Couldn't open the new device");
                break;
            }
            assert(new_dev);
            p_cdc_acm_obj->new_dev_cb(new_dev);
            usb_host_device_close(p_cdc_acm_obj->cdc_acm_client_hdl, new_dev);
        }
        break;
    case USB_HOST_CLIENT_EVENT_DEV_GONE: {
        ESP_LOGI(TAG, "Device suddenly disconnected");
        // Find CDC pseudo-devices associated with this USB device and close them
        cdc_dev_t *cdc_dev;
        cdc_dev_t *tcdc_dev;
        // We are using 'SAFE' version of 'SLIST_FOREACH' which enables user to close the disconnected device in the callback
        SLIST_FOREACH_SAFE(cdc_dev, &p_cdc_acm_obj->cdc_devices_list, list_entry, tcdc_dev) {
            if (cdc_dev->dev_hdl == event_msg->dev_gone.dev_hdl && cdc_dev->notif.cb) {
                // The suddenly disconnected device was opened by this driver: inform user about this
                const cdc_acm_host_dev_event_data_t disconn_event = {
                    .type = CDC_ACM_HOST_DEVICE_DISCONNECTED,
                };
                cdc_dev->notif.cb((cdc_acm_dev_hdl_t) cdc_dev, &disconn_event, cdc_dev->cb_arg);
            }
        }
        break;
    }
    default:
        assert(false);
        break;
    }
}

Can anyone help me out on this?

tore-espressif, 2022-11-09:
Hello @Darshnik253 ,
I'm sorry it doesn't work for you. Unfortunately, from the description I can't tell where the problem might be.
Could you please share with us:
1. Your HW connection to USB device (Implemented by ESP32-S3)?
2. Log from both correct and incorrect behaviour
3. Code snippet of you `main` function

Thank you!

tore-espressif, 2023-03-27:
4. The problem might as well be that your USB device does not recongnize disconnection event correctly. If you have self-powered device, you must monitor VBUS : https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_device.html#self-powered-device

Feel free to reopen