espressif/esp-idf #10139: gevent build error

KaeLL, 2022-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

`install.sh` to work

### What is the actual behavior?

Errors below. Perhaps due to https://github.com/python/cpython/pull/28968?

### Steps to reproduce.

`release/v4.4`'s Getting Started with Python 3.11? "BTW, I use Clear Linux"

### Build or installation Logs.

```plain
Building wheels for collected packages: gevent
  Building wheel for gevent (pyproject.toml) ... error
  error: subprocess-exited-with-error

src/gevent/_greenlet_primitives.c:216:12: fatal error: longintrepr.h: No such file or directory
        216 |   #include "longintrepr.h"
            |            ^~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/lib64/ccache/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gevent
Failed to build gevent
ERROR: Could not build wheels for gevent, which is required to install pyproject.toml-based projects
```
```

igrr, 2022-11-08:
Thanks for the report, this is being investigated: https://github.com/espressif/esp-idf/issues/10116#issuecomment-1304840376

KaeLL, 2022-11-08:
Perfect.

closed 2022-11-08 by KaeLL