espressif/esp-idf #10140: Pthread mutex locking hangs forever

MayaPosch, 2022-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

When I call `phtread_mutex_lock()` I expect the function to return with a success code.

### What is the actual behavior?

Instead I see in the debugger that the backtrace ends at `xQueueSemaphoreTake` or `esp_crosscore_int_send_yield` depending on the run (see debug logs). 

### Steps to reproduce.

I'm running the same pthread-based code as used on Linux, BSD and other pthread platforms.

### Debug Logs.

```plain
Backtrace from GDB:

#0  0x400873fc in esp_crosscore_int_send_yield (core_id=0) at C:\Users\Maya\.platformio\packages\framework-espidf\components\esp_system\crosscore_int.c:144
No locals.
#1  0x40089854 in xQueueSemaphoreTake (xQueue=0x3ffbae60, xTicksToWait=<optimized out>) at C:\Users\Maya\.platformio\packages\framework-espidf\components\hal\esp32\include/hal/cpu_ll.h:39
        xEntryTimeSet = 1
        xTimeOut = {xOverflowCount = 0, xTimeOnEntering = 1202}
        pxQueue = 0x3ffbae60
        xInheritanceOccurred = 0
        __func__ = "xQueueSemaphoreTake"
#2  0x40082ccf in pthread_mutex_lock_internal (mux=0x3ffbae48, tmo=4294967295) at C:\Users\Maya\.platformio\packages\framework-espidf\components\pthread\pthread.c:620
No locals.
#3  0x40082d05 in pthread_mutex_lock (mutex=0x3ffb41cc <NymphRemoteClient::methodsMutex>) at C:\Users\Maya\.platformio\packages\framework-espidf\components\pthread\pthread.c:650
        res = <optimized out>
#4  0x400db256 in __gthread_mutex_lock (__mutex=0x3ffb41cc <NymphRemoteClient::methodsMutex>)
    at c:\users\maya\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\bits/gthr-default.h:761
No locals.
```

Another run:

```
#0  0x40089854 in xQueueSemaphoreTake (xQueue=0x3ffbae60, xTicksToWait=<optimized out>) at C:\Users\Maya\.platformio\packages\framework-espidf\components\hal\esp32\include/hal/cpu_ll.h:39
        xEntryTimeSet = 1
        xTimeOut = {xOverflowCount = 0, xTimeOnEntering = 2323}
        pxQueue = 0x3ffbae60
        xInheritanceOccurred = 0
        __func__ = "\024?@?ueSemaphmaphake"
#1  0x40082ccf in pthread_mutex_lock_internal (mux=0x3ffbae48, tmo=4294967295) at C:\Users\Maya\.platformio\packages\framework-espidf\components\pthread\pthread.c:620
No locals.
#2  0x40082d05 in pthread_mutex_lock (mutex=0x3ffb41cc <NymphRemoteClient::methodsMutex>) at C:\Users\Maya\.platformio\packages\framework-espidf\components\pthread\pthread.c:650
        res = <optimized out>
#3  0x400db256 in __gthread_mutex_lock (__mutex=0x3ffb41cc <NymphRemoteClient::methodsMutex>)
    at c:\users\maya\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\bits/gthr-default.h:761
No locals.
#4  std::mutex::lock (this=0x3ffb41cc <NymphRemoteClient::methodsMutex>) at c:\users\maya\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_mutex.h:103
        __e = <optimized out>
        __e = <optimized out>
```
```

### More Information.

This code is called from the FreeRTOS mutex implementation in the NPoCo project:
https://github.com/MayaPosch/NPoCo/blob/master/core/src/platforms/Mutex_FreeRTOS.h
https://github.com/MayaPosch/NPoCo/blob/master/core/src/platforms/Mutex_FreeRTOS.cpp

I compile the NPoco libraries for ESP32 separately using the same toolchain as the project. See the Makefiles in the NPoCo project folder.

This is part of a porting attempt of NymphRPC to the ESP32:
https://github.com/MayaPosch/NymphRPC/

What I find amazing in the GDB backtrace is that it shows that xQueueSemaphoreTake somehow comes from hal/cpu_ll.h:39. 

Last item is that although I am using ESP-IDF here, the project build and flashing is controlled via PlatformIO's pio.exe. This should to my knowledge not affect the build process (which should use ESP-IDF's CMake-based build tools), but it may be relevant.

Dazza0, 2022-11-09:
Hi @MayaPosch, `xQueueSemaphoreTake()` getting stuck usually just means that the pthread mutex (implemented using FreeRTOS mutexes) is already taken by another thread. If you try `pthread_mutex_trylock()` on the same pthread mutex, does it return `EBUSY`?

MayaPosch, 2022-11-09:
@Dazza0 When I use pthread_mutex_trylock() I did indeed receive a non-zero return code.

The odd thing is that I'm running this exact same code on Linux, etc. without issues. The code in question is completely clean in Valgrind (DRD, Helgrind, Memcheck). The mutex in question is definitely unlocked (after multiple lock/unlock operations by the main thread) by the time this locking attempt is made (in a new thread).

Is there some pecularity about mutexes on FreeRTOS and/or the pthread implementation in ESP-IDF that I am unaware of?

chipweinberger, 2022-11-09:
I'd recommend putting some log statement around and trying to figure out who might be holding the lock. I recommend this log statement, as it can be called anywhere.

```
// this is how we can log from ISR's etc
esp_rom_printf(DRAM_STR("str1234"));
```

0xjakob, 2022-11-10:
@MayaPosch What is the exact return value of `pthread_mutex_trylock()`, is it really `EBUSY` or something else?

> Is there some pecularity about mutexes on FreeRTOS and/or the pthread implementation in ESP-IDF that I am unaware of?

Both implementations should adhere to the POSIX standard, so theoretically, it shouldn't make any difference. The FreeRTOS scheduling is different then scheduling on Linux, however, and that *can* lead to different results if different threads are accessing the same lock.

MayaPosch, 2022-11-10:
I seem to have resolved the issue. The core of the problem appears to be that reentrant mutexes are not supported. Removing the second locking attempt (earlier in the same task) fixed the issue.

Is either `PTHREAD_MUTEX_RECURSIVE_NP` or `PTHREAD_MUTEX_RECURSIVE` supported by the pthread implementation? I cannot seem to find references to either in the documentation.

Dazza0, 2022-11-16:
@MayaPosch `PTHREAD_MUTEX_RECURSIVE` is supported. `PTHREAD_MUTEX_RECURSIVE` will cause the underlying FreeRTOS to call `xSemaphoreCreateRecursiveMutex()` instead of `xSemaphoreCreateMutex()`. You simply need to set the attribute using `pthread_mutexattr_settype()` before that attribute into passing `pthread_mutex_init()` when creating the mutex.

MayaPosch, 2022-11-19:
Thank you for that response. I can indeed see that `PTHREAD_MUTEX_RECURSIVE` is supported. In that case there must be something off in my code/cross-platform preprocessing...

Dazza0, 2022-11-25:
@MayaPosch A couple of things that could help you debug mutex race conditions are

- Use [GDB Jtag debugging](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html) to pause your application. You can then use GDB to print out the current mutex holder directly (FreeRTOS stores a mutex holder in `u.xSemaphore.xMutexHolder` member of the `struct QueueDefinition`
- [Application level tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html) to trace the order of what mutexes are taken/released by what pthreads.

I'm closing this issue as I don't think it's a bug in pthreads. If you run into any more pthread mutex issues, please feel free to reopen this, or open a new.

closed 2022-11-25 by Dazza0