espressif/esp-idf #10144: How to view SDK version through bin file?

hzhh110, 2022-11-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How do I view the SDK version through the bin file? I need to look at the SDK version to see the difference, because the code is the same, because the SDK version is different, the result is different.

hzhh110, 2022-11-09:
I deleted the old version, so I can't view it

ESP-Marius, 2022-11-09:
You should be able to do `esptool.py --chip TARGET image_info --version 2 PATH_TO_BIN`

and get some output which contains application info, e.g.: 

```
Application information
=======================
Project name: hello_world
App version: v5.1-dev-1702-g8d24aa1760-dirty
Compile time: Nov  9 2022 14:33:19
ELF file SHA256: cd448f98032a35b037c127d4941dc3dfc0d5452004dab22226ddb69410b018fb
ESP-IDF: v5.1-dev-1702-g8d24aa1760-dirty
Secure version: 0
```

The exact info included will depend on SDK version and kconfig settings though.

hzhh110, 2022-11-09:
> You should be able to do `esptool.py --chip TARGET image_info --version 2 PATH_TO_BIN`
>
> and get some output which contains application info, e.g.:
>
> ```
> Application information [...]

> ```
> 
> The exact info included will depend on SDK version and kconfig settings though.

Okay, I tried your order, but it didn't work

huangzhenhuideiMac-2:tools huangzhenhui$ sudo python esptool.py --chip esp32 image_info --version 2 /Users/huangzhenhui/Desktop/P153-V1.7.2.bin 
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: unrecognized arguments: --version /Users/huangzhenhui/Desktop/P153-V1.7.2.bin
huangzhenhuideiMac-2:tools huangzhenhui$ sudo python esptool.py --chip esp32 image_info --version /Users/huangzhenhui/Desktop/P153-V1.7.2.bin 
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: unrecognized arguments: --version
huangzhenhuideiMac-2:tools huangzhenhui$ sudo python esptool.py --chip esp32 image_info /Users/huangzhenhui/Desktop/P153-V1.7.2.bin 
esptool.py v2.8
Image version: 1
Entry point: 40082ed8
6 segments

Segment 1: len 0x4049c load 0x3f400020 file_offs 0x00000018 [DROM]
Segment 2: len 0x05f70 load 0x3ffbdb60 file_offs 0x000404bc [BYTE_ACCESSIBLE, DRAM, DMA]
Segment 3: len 0x09bdc load 0x40080000 file_offs 0x00046434 [IRAM]
Segment 4: len 0x15c1c0 load 0x400d0020 file_offs 0x00050018 [IROM]
Segment 5: len 0x11e60 load 0x40089bdc file_offs 0x001ac1e0 [IRAM]
Segment 6: len 0x00010 load 0x50000200 file_offs 0x001be048 [RTC_DATA]
Checksum: fb (valid)
Validation Hash: f51cb203c3efcacb09626665b7e7d3bde7f2794d6c3410c20db4d10320d0b79d (valid)
huangzhenhuideiMac-2:tools huangzhenhui$

ESP-Marius, 2022-11-09:
Ah, seems like the version` argument is quite new, so it might not be available in your version of esptool.

You could try using the latest esptool. Alternatively you can just look at the binary directly yourself as the image header should be the first thing in it, e.g.:

`hexdump -C -n 300 build/hello_world.bin`:

```
00000000  e9 05 02 10 28 11 08 40  ee 00 00 00 00 00 00 00  |....(..@........|
00000010  00 8f 01 00 00 00 00 01  20 00 40 3f 3c 7a 00 00  |........ .@?<z..|
00000020  32 54 cd ab 00 00 00 00  00 00 00 00 00 00 00 00  |2T..............|
00000030  76 35 2e 31 2d 64 65 76  2d 31 36 39 31 2d 67 65  |v5.1-dev-1691-ge|
00000040  30 38 35 63 66 64 33 31  30 2d 64 69 72 74 79 00  |085cfd310-dirty.|
00000050  68 65 6c 6c 6f 5f 77 6f  72 6c 64 00 00 00 00 00  |hello_world.....|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  31 35 3a 35 37 3a 32 33  00 00 00 00 00 00 00 00  |15:57:23........|
00000080  4e 6f 76 20 20 39 20 32  30 32 32 00 00 00 00 00  |Nov  9 2022.....|
00000090  76 35 2e 31 2d 64 65 76  2d 31 36 39 31 2d 67 65  |v5.1-dev-1691-ge|
000000a0  30 38 35 63 66 64 33 31  30 2d 64 69 72 74 79 00  |085cfd310-dirty.|
000000b0  58 99 40 fb db b1 2c 82  c5 4a 88 97 6c 0e ae 23  |X.@...,..J..l..#|
000000c0  01 70 91 b2 69 61 70 a3  36 84 4a 64 42 d9 28 a1  |.p..iap.6.JdB.(.|
000000d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
```

hzhh110, 2022-11-09:
hank you very much

hzhh110, 2022-11-09:
Thank you very much

closed 2022-11-09 by hzhh110