espressif/esp-idf #10149: Coprocessor0Disabled (rsr.exccause = 32) thrown when executing "lsi f0,a1,96" leading to errors in esp-cryptoauthlib

QlifeDeveloper, 2022-11-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release_1_1_0-82-g762291e

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32

### Power Supply used.

External 5V

### What is the expected behavior?

I specifically expect f0 to be loaded.
I expect that floating point operations are supported.

### What is the actual behavior?

Using a JTAG debugger I observe that a UserException is thrown when I try to single step over the "lsi f0,a1,96" instruction.
The rsr.exccause reports 32, equal to Coprocessor0Disabled exception.

The symptom in the system is that after haing executed the code comprising the "lsi f0,a1,96" instruction any communication using HTTPS endpoints will at some point fail with following error message:
"./components/esp-cryptoauthlib/cryptoauthlib/lib/calib/calib_execution.c:379:f0:atreceive - failed"

### Steps to reproduce.

1. Execute "lsi f0,a1,96"
2. Communicate using a HTTPS endpoint
3. Observe "./components/esp-cryptoauthlib/cryptoauthlib/lib/calib/calib_execution.c:379:f0:atreceive - failed"

### Debug Logs.

```plain
No debug information other than the error message is reported.
```

igrr, 2022-11-09:
Hi @QlifeDeveloper,
Coprocessor0Disabled exception is normal part of handling the first floating-point instruction which is executed by the task. The exception handler enables the coprocessor and sets up floating point registers, if necessary.

I guess it might be an issue with how OpenOCD handles such a stepping scenario. It doesn't know that the CoprocessorException will eventually return and therefore doesn't step out of it.

Instead of stepping over this instruction, please try to set a breakpoint on the next instruction and then "continue". In this case the exception should finish executing and the program will stop at the breakpoint on the next instruction.

QlifeDeveloper, 2022-11-10:
Hi Ivan
Thanks a lot for your reply.
It helps me to dig into the problem with this information.
The problem though is unfortunately not the debugging. The single stepping is only part of the process to understand the issue.

Main problem is we experiences in running code miscellaneous error messages from the cryptoauthlib like  "./components/esp-cryptoauthlib/cryptoauthlib/lib/calib/calib_execution.c:379:f0:atreceive - failed" after the code comprising the float instructions have been executed.
We can verify that the float operations work, but it seems they mess up things for the cryptoauthlib.
Experiments with omitting the float operations verify that the cryptoauthlib errors only appears when float operations are used.

Is there a way to force the compiler to use software floating point?
As I understand only one CPU holds the floating point coprocessor. Would it be an idea to place the task performing float operations on this CPU?

Thanks in advance!
Søren

mahavirj, 2022-11-11:
@QlifeDeveloper 

- Can you please share your application along with sdkconfig that recreates this problem? 
- Which IDF release you are using? Is it IDF v4.4 or v5.0? 
- Please share your hardware details, how you are interfacing secure element to ESP32.

QlifeDeveloper, 2022-11-15:
Hi Mahavirj
Thanks for your interest!

We have tried to make a demonstration application in Espressif IDF v4.4 but we have not succeeded in creating a project that produces the same error.
We are in the process of trying to identify the software module in our existing application that causes the error.

The hardware uses the ESP32-WROOM-32E module with an external ATECC608A chip configured exactly as on the ESP32-WROOM-32SE module.

It spells in the Espressif documentation under "Floating Point Arithmetic": "However the use of hardware acceleration leads to some behavioral restrictions in ESP-IDF FreeRTOS"
Do you know what these "behavioral restrictions" are?

We will return when we have more information.
Thanks in advance!
Best regards Søren

mahavirj, 2022-11-15:
(I think it refers to)  For SMP case, if unpinned task does any FPU operations then it gets pinned to specific core. 

https://github.com/espressif/esp-idf/blob/76caa64286b0e93207ad06f68830a223be49cc88/components/freertos/port/xtensa/xtensa_vectors.S#L913

But that should not affect application behavior or result in this problem (IMO).

mahavirj, 2022-12-05:
@QlifeDeveloper Any update on this?

closed 2022-12-14 by mahavirj

QlifeDeveloper, 2022-12-14:
Hi Mahavir

Sorry for not answering you.
We have changed the task prioritizing and CPU affinity so out code runs on CPU1 and espressif code on CPU0.
It seems to solve the problem.

Thanks for your attention!
Best regards
Søren

Closed #10149<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fespressif%2Fesp-idf%2Fissues%2F10149&data=05%7C01%7Csbh%40egoo.health%7C4034e49bbfa441ea0d4e08dadd87a32f%7Ccb0fe16ecb614e8fa723b9237b6852e5%7C0%7C0%7C638065871869161085%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=DIiHHsT6GosKgbh%2ByGy82R2eO5Li8OX1d1AhEKZiSeY%3D&reserved=0> as completed.

-