espressif/esp-idf #10150: Release v4.4.3 is from an unknown branch not release/4.4

pebender, 2022-11-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The subject says it all. Unlike previous 4.4 releases, v4.4.3 is not on the release/4.4 branch. In fact GitHub provides the ominous warning "This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository."

KaeLL, 2022-11-09:
@igrr what up with that?

igrr, 2022-11-09:
I think it was because the merge commit hasn't synced to Github yet. Sorry for the inconvenience, should be fixed now.

closed 2022-11-09 by igrr