espressif/esp-idf #10152: ESP32 UART RX interrupt CPU crash when plugging/unplugging USB

krupis, 2022-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32 WROOM 32UE custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

ESP32 not crash when USB->UART converter is plugged/unplugged whilst the device is powered ON.

### What is the actual behavior?

When USB->UART converter is plugged in to my board UART RX interrupts are being triggered and cause the CPU to crash.

### Steps to reproduce.

1. Power up the device
2. Wait for the device to initialise
3. Plug the USB->UART connector to the board that has GND, RX, TX pins)
4. Plug the other end of the connector to the PC USB.

The below is uart interrupt handler and uart setup function.
When an interrupt is fired, I send information to the queue where I parse the data.
```
static void IRAM_ATTR uart_intr_handle(void *arg)
{
    BaseType_t pxHigherPriorityTaskWoken = pdFALSE;
    uint16_t rx_fifo_len, status;
    uint16_t i=0;
    uart_info_s message;

    
    status = UART1.int_st.val; // read UART interrupt Status
    rx_fifo_len = UART1.status.rxfifo_cnt; // read number of bytes in UART buffer
    
    //urxlen_uart1 = rx_fifo_len;
    message.rx_length = rx_fifo_len;
    while(rx_fifo_len){
        //rxbuf_uart1[i++] = UART1.fifo.rw_byte; // read all bytes
        message.rx_data[i++] = UART1.fifo.rw_byte;
        rx_fifo_len--;
    }
    
    // after reading bytes from buffer clear UART interrupt status
    uart_clear_intr_status(DISPLAY_UART, UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);

    xQueueSendFromISR(uart1_queue, &message, &pxHigherPriorityTaskWoken ); // send to queue
    if( pxHigherPriorityTaskWoken )
    {
        portYIELD_FROM_ISR ();
[... 214 lines cut ...]
0x401b3753: _vfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1781 (discriminator 1)

0x401b38d1: vprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x401c7ff5: esp_log_writev at C:/Users/petrikas.lu/esp/esp-idf/components/log/log.c:200

0x40095d01: esp_log_write at C:/Users/petrikas.lu/esp/esp-idf/components/log/log.c:210

0x400e21d1: parse_received_data at C:\Users\petrikas.lu\Desktop\WORK\elstat\elstat\build/../components/BLE/BLE_custom.cpp:735 (discriminator 8)

0x400e2fa1: DISPLAY_UART_task at C:\Users\petrikas.lu\Desktop\WORK\elstat\elstat\build/../components/Display/Display.cpp:237

0x4009174a: vPortTaskWrapper at C:/Users/petrikas.lu/esp/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 7e25f2c45cee992d

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

### More Information.

I believe simillar issue occurs when I simply connect long cables to my RX/TX pin and leave cables floating as if they were antenna.. The noise coming from the cables can cause issues with the RX interrupt and cause the CPU to crash.

igrr, 2022-11-10:
Please check if your interrupt handler is clearing all the interrupt status bits not just `UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR`. The cause of the panic, Interrupt Watchdog Timeout, indicates that the interrupt handler runs over and over, which is typically the case when the interrupt handler does not manage to clear the interrupt status.

krupis, 2022-11-10:
> Please check if your interrupt handler is clearing all the interrupt status bits not just `UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR`. The cause of the panic, Interrupt Watchdog Timeout, [...]

Thanks for the response. Could you clarify to me how can I check if all interrupt status bits are cleared? I assume the only way would be to check via the debugger but I dont have debugger connected.

How can I ensure that all interrupt status bits are cleared? Should I Add more flags to my current function:
```
uart_clear_intr_status(DISPLAY_UART, UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);
```

igrr, 2022-11-10:
Since you are already checking the interrupt status register and saving it into `status` variable,

> I assume the only way would be to check via the debugger but I dont have debugger connected.

you can also do something like
```c++
esp_rom_printf("int status: 0x%08x\n", status);
```
this is going to print such a debug line every time the interrupt occurs, but probably okay for figuring which bit is not cleared.

> How can I ensure that all interrupt status bits are cleared?

you can clear the same bits which are set in the status register:
```c++
uart_clear_intr_status(DISPLAY_UART, status);
```

krupis, 2022-11-10:
@igrr 

```
uart_clear_intr_status(DISPLAY_UART, status);
``` 

seem to fix my issue!

I have also added esp_rom_printf as you have suggested and I can see the following:
```
int status: 0x00000080
I (09:26:21.380) BLE: command is not valid
```

Now I do not get spam debug message as before and the device does not crash.

I do not understand why most examples (maybe even esp_idf uart interrupt example) suggest using:
```
uart_clear_intr_status(DISPLAY_UART, UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);
```
instead of 
```
uart_clear_intr_status(DISPLAY_UART, status);
```

igrr, 2022-11-10:
> I do not understand why most examples (maybe even esp_idf uart interrupt example) suggest using:
> uart_clear_intr_status(DISPLAY_UART, UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);

I think this is a mistake. UART driver enables more types of interrupts than the two mentioned above:

https://github.com/espressif/esp-idf/blob/cc5640aef358545e2688f8cfeef5c8db73d887aa/components/driver/uart.c#L62-L66

Probably clearing just the two interrupt bits happens to work in most cases, since UART_INTR_RXFIFO_OVF, UART_INTR_BRK_DET,  and UART_INTR_PARITY_ERR do not occur in "normal" circumstances. As you have found, electrical noise can cause one of these types of interrupts to be generated.

If you saw this in IDF examples, could you please give a link to such usage? I haven't found where ESP-IDF examples in v4.4.1 would use uart_clear_intr_status, at all.

krupis, 2022-11-10:
@igrr 

I have found uart_clear_intr_status in esp\esp-idf\examples\peripherals\uart\uart_async_rxtxtasks

This is uart_async_rxtxtasks_main.c:
```
/* 
UART Interrupt Example
*/
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/uart.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "esp_intr_alloc.h"

#include "esp32/rom/uart.h"
#include <freertos/task.h>
#include "freertos/semphr.h"

SemaphoreHandle_t xSemaphore = NULL;

static const char *TAG = "uart_events";

/**
 * This example shows how to use the UART driver to handle UART interrupt.
 *
 * - Port: UART0
 * - Receive (Rx) buffer: on
 * - Transmit (Tx) buffer: off
 * - Flow control: off
 * - Event queue: on
 * - Pin assignment: TxD (default), RxD (default)
 */

#define EX_UART_NUM UART_NUM_1
#define PATTERN_CHR_NUM (3) /*!< Set the number of consecutive and identical characters received by receiver which defines a UART pattern*/

#define BUF_SIZE (1024)
#define RD_BUF_SIZE (BUF_SIZE)
static QueueHandle_t uart0_queue;

// Both definition are same and valid
//static uart_isr_handle_t *handle_console;
static intr_handle_t handle_console;

// Receive buffer to collect incoming data
uint8_t rxbuf[256];
// Register to collect data length
uint16_t urxlen;

#define NOTASK 0

// this task will wait for semaphore
void WAITING_TASK(void *pvParameter)
{
 char received_data[256];
 while(1){
 
 if(xSemaphoreTake(xSemaphore, portMAX_DELAY))
    {
    printf("semaphore received =%s\n",rxbuf);
    }

 }
}
/*
 * Define UART interrupt subroutine to ackowledge interrupt
 */
static void IRAM_ATTR uart_intr_handle(void *arg)
{
    BaseType_t pxHigherPriorityTaskWoken = pdFALSE;
    uint16_t rx_fifo_len, status;
    uint16_t i=0;
    
    status = UART0.int_st.val; // read UART interrupt Status
    rx_fifo_len = UART0.status.rxfifo_cnt; // read number of bytes in UART buffer
    
        while(rx_fifo_len){
        rxbuf[i++] = UART0.fifo.rw_byte; // read all bytes
        rx_fifo_len--;
    }
    
    // after reading bytes from buffer clear UART interrupt status
    uart_clear_intr_status(EX_UART_NUM, UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);

    //xQueueSendFromISR(uart0_queue, &rxbuf, NULL); // send to queue

    xSemaphoreGiveFromISR(xSemaphore,pxHigherPriorityTaskWoken);
    if (pxHigherPriorityTaskWoken) {
        portYIELD_FROM_ISR();
    }

 
}
/*
 * main 
 */
void app_main()
{
    xSemaphore = xSemaphoreCreateBinary();

    int ret;
    esp_log_level_set(TAG, ESP_LOG_INFO);

    /* Configure parameters of an UART driver,
    * communication pins and install the driver */
    uart_config_t uart_config = {
    .baud_rate = 38400,
    .data_bits = UART_DATA_8_BITS,
    .parity = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };

    ESP_ERROR_CHECK(uart_param_config(EX_UART_NUM, &uart_config));

    //Set UART log level
    esp_log_level_set(TAG, ESP_LOG_INFO);

    //Set UART pins (using UART0 default pins ie no changes.)
    ESP_ERROR_CHECK(uart_set_pin(EX_UART_NUM, 13, 26, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));

    //Install UART driver, and get the queue.
    ESP_ERROR_CHECK(uart_driver_install(EX_UART_NUM, BUF_SIZE * 2, BUF_SIZE * 2, 0, NULL, 0));

    // release the pre registered UART handler/subroutine
    ESP_ERROR_CHECK(uart_isr_free(EX_UART_NUM));

    // register new UART subroutine
    ESP_ERROR_CHECK(uart_isr_register(EX_UART_NUM,uart_intr_handle, NULL, ESP_INTR_FLAG_IRAM, &handle_console));

    // enable RX interrupt
    ESP_ERROR_CHECK(uart_enable_rx_intr(EX_UART_NUM));

    xTaskCreate(&WAITING_TASK, "WAITING_TASK", 2048, NULL, 1, NULL);

}
```

igrr, 2022-11-10:
Sorry, that doesn't seem to match the file which is present in v4.4.1 release of ESP-IDF. (Or earlier releases, in fact.):
https://github.com/espressif/esp-idf/blob/v4.4.1/examples/peripherals/uart/uart_async_rxtxtasks/main/uart_async_rxtxtasks_main.c

I found similar looking source code here: https://github.com/theElementZero/ESP32-UART-interrupt-handling/blob/master/uart_interrupt.c but that doesn't come from ESP-IDF repository.

krupis, 2022-11-10:
@igrr 

Oh my..
I might have actually modified the esp-idf example a while ago and accidentally saved on top of it.. My bad :) Anyways, thanks for helping me solve the issue!

closed 2022-11-10 by igrr