espressif/esp-idf #10154: Starting carrier wave and other RF tests from ESP32 IDF HAL since the RF test firmware doesnt work in CDC on the C3

0x0fe, 2022-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The RF test firmware does not work with the ESP32C3 in CDC mode, we can connect and send commands however we only get empty replies from the ESP32C3. The problem is that in some case the UART is not accessible, only USB is accessible.
So i would like to know wether there is a way to use the RF test function from the IDF HAL, at least setting the carrier wave, and if not wether there a CDC version of the RF test firmware for C3?
IF not it would be a good idea to make such ersion, it is not always possible to hae UART on test pads, it is often used for other function making it unusable for UART even if it was broken out on test pads.

[image]

AxelLin, 2022-12-17:
https://www.esp32.com/viewtopic.php?f=2&t=31092

AxelLin, 2023-05-07:
@0x0fe 

The response from support:
Sorry, C3 only has RF test tools and firmware controlled by UART0 at present.
Thank you very much for choosing Espressif's products.

So I think you can close this, it's not supported.

igrr, 2023-05-07:
Note that while not exactly a solution to the issue, https://github.com/espressif/esp-idf/tree/master/examples/phy/cert_test can be used on C3 over USB and does provide PHY-related test commands.