espressif/esp-idf #10155: Save Core Dump to flash while using external SPI RAM not possible.

ddomnik, 2022-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My goal is to store fatal error/crash information somehow. After looking at this issue how to make a custom panic handler #5163 , it is suggested to use the [esp_core_dump_get_summary](https://github.com/espressif/esp-idf/blob/master/components/espcoredump/include/esp_core_dump.h#L150)() function or the core dump function in general.
However, I have an external SPI RAM / PSRAM connected using the ESP S3.

As soon as I enable Hardware Settings > Support for external SPI-connected RAM, I cannot select the "Save core dump to FLASH"
> Core dumps to Flash are not available if PSRAM is used for task stacks.

In code, I create tasks via xTaskCreateStatic() but none of the arguments, nor the TaskHandle_t struct itself is explicitly saved in external RAM. So why is flash not selectable?

Is it somehow possible to save the Core Dump to flash or save some fatal error information in another way?

igrr, 2022-11-10:
Could you try disabling the option CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY? This should make the CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH option available.

ddomnik, 2022-11-10:
@igrr thank you for the fast reply. I am using commit 01d014c42d (idf 5.0). But can not find a flag in the menuconfig to set CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY. Commenting in out manually does not help either after reloading the sdkconfig.

igrr, 2022-11-10:
It should be under Component config > ESP PSRAM > SPI RAM config

[image]

([docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32s3/api-reference/kconfig.html?highlight=config_spiram_allow_stack_external_memory#config-spiram-allow-stack-external-memory))

ddomnik, 2022-11-14:
Thank you. It is sadly not available for commit 01d014c / tag v5.0-rc1 on the ESP32S3 module but for ESP32. The current master however supports it for the ESP32S3. Will this option be available with the 5.0 release?

closed 2022-11-15 by ddomnik