espressif/esp-idf #10156: espcoredump.py reports KeyError: 'current-thread-id'

xobs, 2022-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1626-g4b6d9c8ad3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I'm not sure what the correct behavior is. I enabled core dump over UART, and expected that idf_monitor would work

### What is the actual behavior?

When the Task watchdog is triggered, it prints a backtrace and halts. However, there is an error in idf_monitor:

```
E (532281) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (532281) task_wdt:  - IDLE (CPU 1)
E (532281) task_wdt: Tasks currently running:
E (532281) task_wdt: CPU 0: IDLE
E (532281) task_wdt: CPU 1: gdbc fd:56
E (532281) task_wdt: Aborting.
E (532281) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400559dd:0x3fcdae00 |<-CORRUPTED

ELF file SHA256: 764cbc95c5c71bd8

Initiating core dump!
I (12988) esp_core_dump_uart: Press Enter to print core dump to UART...
I (12988) esp_core_dump_uart: Print core dump to uart...
Core dump started (further output muted)
Received  45 kB...
Core dump finished!
espcoredump.py v1.3

Traceback (most recent call last):
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor.py", line 376, in <module>
    main()
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor.py", line 367, in main
    monitor.main_loop()
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor.py", line 162, in main_loop
    self._main_loop()
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor.py", line 268, in _main_loop
    super()._main_loop()
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor.py", line 207, in _main_loop
    self.serial_handler.handle_serial_input(data, self.console_parser, self.coredump,
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor_base\serial_handler.py", line 102, in handle_serial_input
    with coredump.check(line):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor_base\coredump.py", line 128, in check
    self._check_coredump_trigger_before_print(line)
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor_base\coredump.py", line 106, in _check_coredump_trigger_before_print
    self._process_coredump()
  File "C:\Users\Sean\esp\esp-idf\tools\idf_monitor_base\coredump.py", line 79, in _process_coredump
    coredump.info_corefile()
  File "C:\Users\Sean\.espressif\python_env\idf5.1_py3.10_env\lib\site-packages\esp_coredump\coredump.py", line 367, in info_corefile
    self.print_threads_info(task_info)
  File "C:\Users\Sean\.espressif\python_env\idf5.1_py3.10_env\lib\site-packages\esp_coredump\coredump.py", line 218, in print_threads_info
    threads, _ = self.gdb_esp.get_thread_info()
  File "C:\Users\Sean\.espressif\python_env\idf5.1_py3.10_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 100, in get_thread_info
    current_thread_id = result['current-thread-id']
KeyError: 'current-thread-id'
```

### Steps to reproduce.

1. Enable core dumps
2. Have it panic with one thread not issuing the watchdog

alekseiapa, 2022-11-21:
@xobs Thank you for informing us about the potential bug. Let us take a quick look at this problem.

alekseiapa, 2022-12-19:
@xobs we solved the issue here. The issue arises as a result of an expired timeout while waiting for a response from GDB. An additional check was added while printing the threads info.

dobairoland, 2022-12-19:
For future reference, the fix has been added in `esp-coredump==1.4.2`. You will upgrade to it automatically if if you rerun `install.ps1` or `install.bat` in your terminal.