espressif/esp-idf #10157: esp32s3使用micropython内存太大

Zhengqinze05, 2022-11-11:
### Is your feature request related to a problem?

我使用的是esp32s3的芯片，使用micro python，但使用micropython之后，内部内存会直接消耗掉一百多k，我有其他的功能需要消耗内存，这一百多k的内存让我使用其他功能时很难进展

### Describe the solution you'd like.

是否有方法可以减少ESP32micro python的内存，我看其他芯片板子本身并没有多少内存依旧可以使用micropython，ESP32S3有520k的内存，但实际使用却仅能使用200多k，再加上micro python，我再使用蓝牙和WiFi，基本把内存消耗完了，想再加新功能就很难了，请问是否有减少micro python内存的方法，或者可以将520k内存全部利用起来的方法

203Null, 2022-11-15:
建议添置PSRAM

Zhengqinze05, 2022-11-15:
> 建议添置PSRAM

感谢你的回复，我的板子是有8M的PSRAM的，gc_init()也是采用PSRAM申请的。但是一旦使用了micropython，系统开始就会减少了100K左右的IRAM，我不知道这是在哪个地方减少的，在程序刚启动没有其他操作的时候，我打印内存就已经少了100K的IRAM了，就像是代码中有micropython，在编译阶段就将内存分配了。

Zhengqinze05, 2022-11-17:
> 

我使用的是官方最新的Micropython

closed 2023-12-20 by dobairoland

hackers365, 2025-01-07:
根据idf内存优化文档，调整不需要的wifi和ble功能，我的项目使用了ble和espnow功能，wifi由120k降低至40k，ble由60k降低至30k