espressif/esp-idf #10158: How to disable prov new device but keep old ble mesh network working

jiabuda, 2022-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I am developing a ble mesh game between devices. Example like this, the host device choose to be a provisioner and the guest device choose to be a node. And the host device could choose to host a how many people room, such as two people room, so only one guest device could join this network and the game begin automatically, the third device will not be provisioned and just waiting in the air.

After review the example code, I choose to judge if the room is full in event `ESP_BLE_MESH_PROVISIONER_RECV_UNPROV_ADV_PKT_EVT`, when the room is full, function `esp_ble_mesh_provisioner_add_unprov_dev` will not be called, so the third device will not get provisioned and waiting. But this method is not elegant to solve this problem, I try to use `esp_ble_mesh_provisioner_prov_disable`, yet net idx and app idx will be erased and no msgs could be sent or received.

So if this is a method, I could disable the provisioning function elegantly, then the two device could still communicate and the host device will not receive unprovisioned device's advertisement package?

If some functions exist to disable the provisioning function, and how to enable it again?

buganini, 2022-11-13:
In your mesh provisioning callback, don't call `recv_unprov_adv_pkt()` when you don't want to provision a new device.

jiabuda, 2022-11-15:
> recv_unprov_adv_pkt

Hi buganini, i know this, yet i am finding a more elegant way to disable the provisioner function but keep the network working

Shane555, 2022-11-25:
@jiabuda maybe you can try esp_ble_mesh_provisioner_prov_disable()

forx157, 2024-03-04:
Hi, @jiabuda 
API `esp_ble_mesh_provisioner_prov_disable` maybe is the answer to your needs.

xd3011, 2024-05-13:
Sorry, I have the same question. I also had the problem of not calling recv_unprov_adv_pkt. However, packets are still received to this provisioner and I think it has affected packets transmitted between other nodes even though I increased the ttl index higher. Please help me fix it

closed 2025-03-31 by Alvin1Zhang