espressif/esp-idf #10159: 2-Stage Application Concept

franz-ms-muc, 2022-11-11:
### Is your feature request related to a problem?

not Really. 

### Describe the solution you'd like.

We want to extend the Application Bootup. 

now it is like https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/startup.html.

1. First stage bootloader
2. Second stage bootloader
3. Application startup executes

RilabsAutomotive, 2022-11-11:
Here is also a diagram of the startup flow , that is described above
[image]

igrr, 2022-11-11:
I think something similar has been implemented in a number of projects already, as follows:
- "Factory" partition is used as the "3rd stage" bootloader, which can download applications from various sources (Wi-Fi, SD card, etc)
- "OTA" partitions are used to store the actual applications

This enables the following workflows:
- Normal boot: 2nd stage bootloader loads the app from one of the OTA slots, the app starts running.
- Upgrade:
   - OTA app receives an event that triggers the upgrade.
   - It calls `esp_ota_set_boot_partition(factory_partition)` and restarts.
   - The 2nd stage bootloader loads the factory app because it is now set as the boot partition.
   - The factory app downloads the update from one of the interfaces and flashes it into an OTA slot.
   - The factory app calls `esp_ota_set_boot_partition(ota_partition)` and restarts.
   - The 2nd stage bootloader loads the app from the selected OTA partition.
- Fallback:
   - If an OTA app fails to boot after the upgrade, the 2nd stage bootloader can switch the default boot app to "factory", enabling recovery process. This can be done using CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE option.

For example, in one such project, the "factory" app would also present a menu on an LCD screen, letting the user to select the app ("game") to boot. The "factory" app also set up a webserver over Wi-Fi, letting users to update other apps ("games"). In your example, the "factory" app can get the update over CAN bus.

franz-ms-muc, 2022-11-11:
[FH] @***@***.******@***.***> this would mean: we integrate the Detection of the „Magic Packet“ into the Main App. So the Update can be started anytime, no need to flood the Bus with the Magic Packet. The Box can Boot, run 1 hour, then the Magic Packet is received by the normal App, and the following happens

[FH] this is done inside the app.

[FH] this could even be happening in case of Error, Timeout, no Binary received etc..

[FH] yes, really cool concept.

If both apps OTA1 and OTA2 are damaged: the Factory App allows reprogramming.

If the App is running, but the Part where the magic packet is received and the switch to the factory app is done is damaged, then we have a Problem. This can be avoided by good Testing,

and maybe inventing a 2nd way of getting into Factory, e.g. over Wifi or MQTT.

Cool concept!!

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

RilabsAutomotive, 2022-11-11:
Yes, we have thought about putting the FBL as Factory, but it will not be possible to update the FBL after that.
Normally, we want also to add a so called "Bootloader updater" or "Bootkiller", which is downloaded in OTA partitions and updates the FBL itself.
If we choose to use factory we would not be able to use the OTA APIs to update the FBL, cause it is factory and not writable.
Maybe it is possible to write it directly via SPI Flash driver, but would the SSBL be able to read the updated FBL/Factory correctly. If there is a header check, this would make the SSBL consider the new image as invalid.

igrr, 2022-11-11:
> cause it is factory and not writable.

There's actually no restriction on writing to 'factory' partition. I think this scheme can be extended to make the factory partition updatable. You can add another partition (ota2) which would contain the dedicated 'factory partition updater app'. If the main app gets an event that signals the factory partition update, it sets the boot app to ota2 and resets. The rest of the logic is similar.

You may use a customized 2nd stage bootloader to implement custom boot priorities, if necessary — please refer to custom bootloader examples for that.

igrr, 2022-11-22:
CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED is only necessary to overwrite:

- the bootloader
- the partition table
- the partition from which the application is currently running from.

So if you aren't trying to overwrite `factory` partition while running the app from the `factory` partition, this option should not be necessary.

igrr, 2023-01-09:
@franz-ms-muc it seems that the question has been answered, can we close the issue now?

franz-ms-muc, 2023-01-09:
Close

franz-ms-muc, 2023-02-02:
@igrr can you give a Hint to [https://www.esp32.com/viewtopic.php?f=13&t=31895&sid=7e8498b82177cfeb323f1ff006b10162](https://www.esp32.com/viewtopic.php?f=13&t=31895&sid=7e8498b82177cfeb323f1ff006b10162)

Thanks !

franz-ms-muc, 2023-02-02:
Outstanding good Support !

closed 2023-02-02 by franz-ms-muc