espressif/esp-idf #10160: HTTP and HTTPS server coexist

DCSBL, 2022-11-11:
### Is your feature request related to a problem?

I want to add an https rest-server, but due to backwards compatibly we require an http server. Is it possible to have an http and an https server at the same time?

Currently when running this code, it does not work:

```
httpd_config_t http_config         = HTTPD_DEFAULT_CONFIG();
httpd_ssl_config_t http_ssl_config = HTTPD_SSL_CONFIG_DEFAULT();
extern const unsigned char cacert_pem_start[] asm("_binary_cert_pem_start");
extern const unsigned char cacert_pem_end[] asm("_binary_cert_pem_end");
http_ssl_config.cacert_pem = cacert_pem_start;
http_ssl_config.cacert_len = cacert_pem_end - cacert_pem_start;

extern const unsigned char prvtkey_pem_start[] asm("_binary_priv_key_pem_start");
extern const unsigned char prvtkey_pem_end[] asm("_binary_priv_key_pem_end");
http_ssl_config.prvtkey_pem = prvtkey_pem_start;
http_ssl_config.prvtkey_len = prvtkey_pem_end - prvtkey_pem_start;

http_config.lru_purge_enable = true;
http_config.uri_match_fn     = httpd_uri_match_wildcard;
http_config.max_uri_handlers = max_num_uri_handlers;
http_ssl_config.httpd        = http_config;

ESP_LOGI(TAG, "Starting HTTP service...");
ESP_ERROR_CHECK(httpd_start(&http_server, &http_config));
ESP_ERROR_CHECK(httpd_ssl_start(&http_ssl_server, &http_ssl_config));  // ←- This line fails "error in creating ctrl socket (112)"

// Init root api
static const httpd_uri_t root = {"/api/?", HTTP_GET, http_get_root, NULL};

ESP_ERROR_CHECK(httpd_register_uri_handler(http_server, &root));
ESP_ERROR_CHECK(httpd_register_uri_handler(http_ssl_server, &root));
```

### Describe the solution you'd like.

Allow a http and https server to coexist

### Describe alternatives you've considered.

A self-chosen encryption layer over the http-layer, but that is bad practice IMO.

### Additional context.

It was tried before [here](https://www.esp32.com/viewtopic.php?t=3136), but is seems any progress has been stalled.

mahavirj, 2022-11-14:
@DCSBL 

This could be an issue with same `ctrl_port` number in both HTTP and HTTPS configurations.

> http_ssl_config.httpd        = http_config;

I suggest that you keep both configurations separate and not over-write one with another. SSL configuration will have the requirement for higher stack size and its better to use values from its default configuration macro.

Please try following patch that I created on `https_server` example in IDF:

```C
diff --git examples/protocols/https_server/simple/main/main.c examples/protocols/https_server/simple/main/main.c
index 123fcec711..2dd65775dc 100644
--- examples/protocols/https_server/simple/main/main.c
+++ examples/protocols/https_server/simple/main/main.c
@@ -156,9 +156,15 @@ static httpd_handle_t start_webserver(void)
         return NULL;
     }
 
+    httpd_handle_t http_server = NULL;
+    httpd_config_t h_conf = HTTPD_DEFAULT_CONFIG();
+    h_conf.ctrl_port = 32769;
+    ESP_ERROR_CHECK(httpd_start(&http_server, &h_conf));
+
     // Set URI handlers
     ESP_LOGI(TAG, "Registering URI handlers");
     httpd_register_uri_handler(server, &root);
+    httpd_register_uri_handler(http_server, &root);
     return server;
 }
```

DCSBL, 2022-11-14:
Thank you @mahavirj!

Changing the port number fixes the issue and both servers now run at the same simultaneously. For the long run; should this be fixed in ESP-IDF or just like this? 

Also thank you for the tip to keep the config separate, had not spotted that there was a change in the default config.

mahavirj, 2022-11-14:
@DCSBL We can fix the default configuration in IDF. I will raise an internal fix for this.

closed 2022-11-22 by espressif-bot

DCSBL, 2022-11-22:
Thank you for the fast solution 🚀!