espressif/esp-idf #10161: Link error in SPI when building for riscv

josesimoes, 2022-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm working on adding a build of [.NET nanoFramework](https://github.com/nanoframework) for ESP32_C3.

All good so far except that I'm running into an issue with SPI. 
```
[build]     rtc_iram_seg:          40 B         8 KB      0.49e:/nftools/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/driver/libdriver.a(spi_common.c.obj): in function `spicommon_periph_claim':
[build] E:/GitHub/esp-idf/components/driver/spi_common.c:108: undefined reference to `__atomic_compare_exchange_1'
[build] e:/nftools/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/driver/libdriver.a(spi_common.c.obj): in function `spicommon_periph_free':
[build] E:/GitHub/esp-idf/components/driver/spi_common.c:127: undefined reference to `__atomic_compare_exchange_1'
[build] collect2.exe: error: ld returned 1 exit status
[build] %
[build] ninja: build stopped: subcommand failed.
```

I'm suspecting this is because we are calling SPI API from C++ code.
It's worth mentioning that the exact same code builds fine when builing for ESP32 and ESP32_S2.

I've searched around and found other reports of similiar issues with riscv.
Adding them here, for reference in case that's usefull:
https://github.com/telegramdesktop/tdesktop/issues/17449
https://github.com/advancedtelematic/aktualizr/issues/1427

Thanks!

josesimoes, 2022-11-11:
Hey @igrr! I've tried that (on a clean build folder to be sure) and the error is the same...

igrr, 2022-11-11:
Interesting, could you please check that the missing symbol is present in the object file?
```
riscv32-esp-elf-nm build/esp-idf/newlib/CMakeFiles/__idf_newlib.dir/stdatomic.c.obj | grep __atomic_compare_exchange_1
```

(not sure what's the equivalent of pipe & grep on Windows, sorry — the point is to see if the symbol name is present in riscv32-esp-elf-nm output)

josesimoes, 2022-11-11:
I believe the equivalent would be:
`riscv32-esp-elf-nm build/esp-idf/newlib/CMakeFiles/__idf_newlib.dir/stdatomic.c.obj |  findstr  __atomic_compare_exchange_1`

WIth and without newlib in the `PRIV_REQUIRES` the symbol is NOT present. (I can paste here the riscv32-esp-elf-nm output if that's of interest).

igrr, 2022-11-11:
If the build is done by https://github.com/nanoframework/nf-interpreter/tree/main/CMake, then I think I might know where the issue is: your toolchain file https://github.com/nanoframework/nf-interpreter/blob/main/CMake/toolchain.riscv32-esp-elf.cmake doesn't specify correct compiler flags for ESP32-C3. There should be no `-mlongcalls`, but instead you should have `-march=rv32imc` (like [here](https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-esp32c3.cmake#L10-L12)).

The files above can be used to confirm whether that's indeed the cause.

josesimoes, 2022-11-11:
Thanks for looking into that! 
That was just a placeholder that was added when reworking the integration with IDF build system. 
Here's the working branch: https://github.com/josesimoes/nf-interpreter/tree/develop-esp32-c3

igrr, 2022-11-11:
Please check if adding `-march=rv32imc` to CMAKE_C_FLAGS  and CMAKE_CXX_FLAGS of your toolchain file helps.

josesimoes, 2022-11-11:
Here are the requested files:
* sdkconfig: https://gist.github.com/josesimoes/c8c7bba3472f9dbd6bf4957e6fa35e42
* output of riscv32-esp-elf-nm: https://gist.github.com/josesimoes/d9b86afd7fc86bd99680f42436a93093
* build/compile_commands.json: https://gist.github.com/josesimoes/67e4a406076bec806f517bc782ebf8e2

josesimoes, 2022-11-11:
> -march=rv32imc

@igrr brilliant!! That was it. It's building fine now!🥳 
Thaaaaank you very much. 😃

closed 2022-11-11 by josesimoes