espressif/esp-idf #10162: Calling esp_event_loop_create_default() in ESP32_C3 it's failing

josesimoes, 2022-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Calling `esp_event_loop_create_default()` it's failing on ESP32_C3 in .NET nanoFramework:
```
E (467) event: create task for loop failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40388cdd
file: "E:/GitHub/nf-interpreter/targets/ESP32/_common/targetHAL_Network.cpp" line 353
func: void nanoHAL_Network_Initialize()
expression: esp_event_loop_create_default()

abort() was called at PC 0x40388ce1 on core 0
Core  0 register dump:
MEPC    : 0x403830f2  RA      : 0x40388cec  SP      : 0x3fca5a30  GP      : 0x3fc91400
TP      : 0x3fc86e34  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fca5a6c  S1      : 0x3fca5a6c  A0      : 0x3fca5a6c  A1      : 0x3fca5a4e
A2      : 0x00000000  A3      : 0x3fca5a99  A4      : 0x00000001  A5      : 0x3fc9c000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca5a50  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fca5a30: 0x00000000 0x00000000 0x3fca5a6c 0x4038e42c 0x00000000 0x00000000 0x00000000 0x00000030
3fca5a50: 0x38333034 0x31656338 0x3fca5a00 0x3fc92814 0x3fca5a50 0x3fc92830 0x3fca5a4c 0x726f6261
3fca5a70: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x38383330 0x20316563
3fca5a90: 0x63206e6f 0x2065726f 0x00000030 0x3c100000 0x00000000 0x00000000 0x3fca5ae0 0x40388ce4
3fca5ab0: 0x00000804 0x00000008 0x3fca5ae0 0x42048c14 0x00000000 0x00000000 0xffffffff 0x00000000
3fca5ad0: 0x0000000e 0x00000001 0x3fca5b80 0x42045908 0x00000804 0x0000000c 0x00000000 0x40382952
3fca5af0: 0x00000009 0x00000000 0x00000000 0x4206cf7c 0x00000804 0x00000008 0xffffffff 0x3fcde330
3fca5b10: 0x00000010 0x0000000e 0x00000009 0x4206d404 0x00000000 0xffc1ffff 0x00000000 0x3fca5b34
3fca5b30: 0x3fcde124 0x003e0000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5b50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000002 0x3fc90f9c
3fca5b70: 0x00000000 0x00000000 0x3fca5b90 0x42047b12 0x00000000 0x00000000 0x3fca5bb0 0x4204546a
3fca5b90: 0x00000000 0x00000000 0x00000010 0x00000000 0x00000000 0x00000000 0x3fca5bc0 0x42044de8
3fca5bb0: 0x00000000 0x00000000 0x3fca5c0c 0x4205457e 0x00000000 0x00000000 0x00036000 0x3fca5e90
3fca5bd0: 0x00000000 0x00000000 0x00000000 0x4205458a 0x00000000 0x00000000 0x00000000 0x42053f02
3fca5bf0: 0x00000000 0x00000000 0x00000000 0x42051a38 0x00000000 0x00000000 0x00000000 0x01000032
3fca5c10: 0x00000000 0x00000000 0x00000000 0x4038b196 0x00000000 0x00000000 0x00000000 0x00000000
3fca5c30: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fca59b0 0x3fc9c294
3fca5c50: 0x3fc9886c 0x3fc9886c 0x3fca5c48 0x3fc98864 0x00000014 0x462fbc0c 0xe48e19fd 0x3fca5c48
3fca5c70: 0x00000000 0x00000005 0x3fca21ac 0x6e69616d 0x7361745f 0x4733006b 0x006f7df8 0x00000000
3fca5c90: 0x3fca5c40 0x00000005 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9cc38 0x3fc9cca0
3fca5cb0: 0x3fc9cd08 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420de7f8
3fca5cd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5cf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5d10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5d30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5d50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5d70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca5d90: 0x00000000 0x00000000 0x3f000000 0x00000054 0x00000000 0x3fca5da0 0x00000000 0x00000000
3fca5db0: 0x00000000 0x3fca5db8 0xffffffff 0x3fca5db8 0x3fca5db8 0x00000000 0x3fca5dcc 0xffffffff
3fca5dd0: 0x3fca5dcc 0x3fca5dcc 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff
3fca5df0: 0x00000000 0x0000000c 0x00000001 0x3fc9c294 0x3fca5df0 0x00000014 0x00000000 0x3fc9c294
3fca5e10: 0x3fc9c294 0x3fc9c294 0x3fca5e00 0x00000014 0x00000002 0x00000013 0x0000001f 0x00000000
```

Decoding the above:
```
0x00000000: ?? ??:0
0x00000000: ?? ??:0
0x3fca5a30: ?? ??:0
0x00000000: ?? ??:0
0x00000000: ?? ??:0
0x3fca5a6c: ?? ??:0
0x4038e42c: __assert_func at assert.c:41
0x00000000: ?? ??:0
0x00000000: ?? ??:0
0x00000000: ?? ??:0
0x00000030: ?? ??:0
0x3fca5a50: ?? ??:0
0x38333034: ?? ??:0
0x31656338: ?? ??:0
0x3fca5a00: ?? ??:0
0x3fc92814: __global_pointer$ at ??:?
0x3fca5a50: ?? ??:0
0x3fc92830: __global_pointer$ at ??:?
0x3fca5a4c: ?? ??:0
0x726f6261: ?? ??:0
0x3fca5a70: ?? ??:0
0x20292874: ?? ??:0
0x20736177: ?? ??:0
0x6c6c6163: ?? ??:0
0x61206465: ?? ??:0
0x43502074: ?? ??:0
0x34783020: ?? ??:0
0x38383330: ?? ??:0
0x20316563: ?? ??:0
0x3fca5a90: ?? ??:0
0x63206e6f: ?? ??:0
0x2065726f: ?? ??:0
0x00000030: ?? ??:0
0x3c100000: _flash_rodata_dummy_start at ??:?
0x00000000: ?? ??:0
0x00000000: ?? ??:0
0x3fca5ae0: ?? ??:0
0x40388ce4: esp_system_abort at esp_system.c:127
0x3fca5ab0: ?? ??:0
0x00000804: ?? ??:0
0x00000008: ?? ??:0
0x3fca5ae0: ?? ??:0
0x42048c14: nanoHAL_Network_Initialize() at targetHAL_Network.cpp:356
```

The call happens [here](https://github.com/josesimoes/nf-interpreter/blob/c945ed78bc18e6a8eed23bfcdef6fb81d79e2947/targets/ESP32/_common/targetHAL_Network.cpp#L353).

The same code it's working fine for ESP32 and ESP32_S2 and there have been no changes in a while.
Any hints please?

igrr, 2022-11-11:
@josesimoes please check the amount of heap memory before calling this function. Most likely the task creation fails due to insufficient heap memory.
https://github.com/espressif/esp-idf/blob/c09322430aa94d45d0e65422513a3e36826020b0/components/esp_event/esp_event.c#L491

josesimoes, 2022-11-11:
OK. Will do and report back. 

 Does this mean that the C3 series requires more heap than the other series? Just checking because we have the memory allocation fine tuned in nanoFramework.

igrr, 2022-11-11:
There might be some chip- and architecture-specific differences. I would suggest checking both  static memory usage using `idf.py size` (and `idf.py size-components` for more details) and the heap available at run time. But in general the C3 should provide more heap than the S2. Since your build process is quite customized, it is possible that something doesn't get placed into the correct memory region, that's the only reason i can think of. The output of idf.py size and heap size will provide some clue.

josesimoes, 2022-11-14:
That was it. After increasing the RAM reserved for IDF allocation that error is gone. Thanks again! 👍🏻 

Something that isn't working for me: getting the detailed output as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#size-summary-idf-py-size) when using the `size ` and `size-components`.

I get this:
```text
Running ninja in directory e:\github\nf-interpreter\build
Executing "ninja size-components"...
ninja: error: unknown target 'size-components'
ninja failed with exit code 1
```

I believe this is because the customized build that we are using. Can you please provide some guidance on how to have that output? Is there a CMake option or extra parameter that we could use to make this happen?

josesimoes, 2022-11-16:
@igrr completely off topic, but I wasn't sure what was the best channel to make this suggestion.
It seems that ESP3-C3 is NOT listed in the RISC-V website. Don't know if that's on purpose or something that has been overlooked... Suggest that you reach out the appropriate contacts internally.
https://riscv.org/exchange/?_sft_exchange_category=board,hardware

igrr, 2022-11-21:
> I believe this is because the customized build that we are using. Can you please provide some guidance on how to have that output? Is there a CMake option or extra parameter that we could use to make this happen?

In your project CMakeLists.txt file, you need to add linker flags for generating the .map file. For IDF build system projects, it is done here (in release/v4.4 branch): https://github.com/espressif/esp-idf/blob/1f63dc70b8cf2a91a08f9e6d395cb05ab17eaa59/tools/cmake/project.cmake#L452-L453

Once the .map file is generated, you can run `$IDF_PATH/tools/idf_size.py --archives <map file>`. IDF build system registers a custom command `size-components` here:
https://github.com/espressif/esp-idf/blob/1f63dc70b8cf2a91a08f9e6d395cb05ab17eaa59/tools/cmake/project.cmake#L477-L480

> It seems that ESP3-C3 is NOT listed in the RISC-V website.

Thank you, I'll try to find a contact who can help with this.

josesimoes, 2022-12-20:
@igrr just a quick follow-up to let you know that following your guidance above, I've successfully added the output of the size summary to nanoFramework build. Thanks!

I'm also closing this issue since the question has been answered and the nanoFramework build for C3 is working fine.

closed 2022-12-20 by josesimoes