espressif/esp-idf #10168: Bluetooth (A2DP) and light sleep with iPhone, can't return to light sleep

expresspotato, 2022-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

LyraT v4.3

### Power Supply used.

USB

### What is the expected behavior?

The CPU should return to light sleep, after the audio is paused. It does this with a Galaxy S22+, but not with an iPhone 5s, SE or X.

### What is the actual behavior?

The CPU remains active, by virtue of:

`
	// Dev board green LED will show when in light sleep or not
	gpio_set_direction(GPIO_NUM_22, GPIO_MODE_OUTPUT);
	gpio_set_level(GPIO_NUM_22, 1);
	gpio_sleep_sel_en(GPIO_NUM_22);
`

1. No locks appear to be held by **bt** after the audio is paused
2. No other locks appear to be held
3. The only running task with any appreciable consumption is **btController**

### Steps to reproduce.

1. Have a 32 kHz crysal or oscillator on your board
2. In menuconfig enable: power management on boot, bluetooth sleep clock 32 kHz crystal, etc.
3. Download the ADF example of A2DP from https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream
4. Enable light sleep in main through esp_pm_configure
5. Enable GPIO 22 to be an indicator for light sleep
6. Modify the example to destroy (or stop perhaps) the pipeline when the audio is suspended
7. Notice that when paused / resumed on an S22 (probably more Android phones as well) the, light sleep resumes
8. Notice on iPhone it does not

### Debug Logs.

```plain
entry 0x400806b4
I (29) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (29) boot: compile time 17:29:26
I (29) boot: chip revision: 3
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32: SPI Speed      : 80MHz
I (42) boot.esp32: SPI Mode       : QIO
I (46) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
[... 593 lines cut ...]
IDLE           	18687751		81%
IDLE           	22753113		99%
ipc1           	153434		<1%
hciT           	141451		<1%
BTU_TASK       	997284		4%
BTC_TASK       	953183		4%
Tmr Svc        	15		<1%
esp_timer      	5755		<1%
btController   	1270429		5%
ipc0           	76468		<1%
sys_evt        	24		<1%

Lock stats:
Name            Type            Arg    Active  
i2c_driver      APB_FREQ_MAX    0      0       
bt              APB_FREQ_MAX    0      1       
rtos1           CPU_FREQ_MAX    0      0       
rtos0           CPU_FREQ_MAX    0      1
```

### More Information.

It appears the following even is missing on iPhone, but is present on my S22+

W (33027) BT_HCI: hcif ssr evt: st 0x0, hdl 0x81, tx_lat 54 rx_lat 1188

expresspotato, 2022-11-30:
So nearly 3 weeks without comment, the best I can do is attach an example of this and note that I have also see the following line missing on iPhone, but present on Android:

**W (248283) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0**

Seems like something is preventing the bt controller from returning to sniff mode.

The attached source is from a much larger project, with the majority taken out. It should be run on the LyraT v4.3 board and has IO22 (green LED) set to display when in / out of light sleep. 

1. Build the project with idf.py build
2. Flash it
3. Reset the board
4. Pair with an **Android** phone and don't touch the screen for a few seconds (to avoid touch sounds being played)
5. Notice the ESP32 enters light sleep as expected
6. Play some music / youtube / etc
7. Notice the CPU is awake for the duration of the playback
8. Pause the source
9. Notice light sleep resumes
10. Reset the board
11. Pair with an **iPhone**
12. Repeat steps 5 - 9
13. Notice that light sleep doesn't resume until the device is disconnected

[pipeline_a2dp_sink_and_hfp.zip](https://github.com/espressif/esp-idf/files/10125703/pipeline_a2dp_sink_and_hfp.zip)

Light sleep is really important with our mass production project here as it is battery powered and plays audio from a device often.

BetterJincheng, 2022-12-01:
@expresspotato Please test again with following menuconfig:

Component config -> FreeRTOS -> Tick rate (100 -> 1000)

Change the tick rate from 100 to 1000.

expresspotato, 2023-01-06:
Hi @BetterJincheng thanks for the suggestion but it didn't make a difference... 

Anyone from @espressif?

AxelLin, 2023-01-07:
> @expresspotato Please test again with following menuconfig:
> 
> Component config -> FreeRTOS -> Tick rate (100 -> 1000)
> 
> Change the tick rate from 100 to 1000.

@BetterJincheng 
What's the point of this change?

expresspotato, 2023-01-30:
Hello anyone from Espressi (@espressif-bot), 

This functionality is pretty critical for our application, and iPhone is a huge user base that's easy to target bug fixes for (since the OS behaves nearly identically across all generations of device).

The above example demonstrates the easy to reproduce problem. Is anyone able to take a look / comment?

BetterJincheng, 2023-01-31:
@expresspotato Could you please provide a minimal project that reproduces the problem? This is very helpful for solving problems quickly.

BetterJincheng, 2023-01-31:
@expresspotato We have tested the example examples/bluetooth/bluedroid/classic_bt/a2dp_sink using an ESP32. Nothing wrong with IOS devices on light sleep. You can have a try. Please let me know the test result and we will provide you with some support if needed.

expresspotato, 2023-02-02:
Hello @BetterJincheng,

So using **v4.4.3** there was no difference using the code above, as in my example zip attached above in this thread, the audio control (via pipline stop / terminate, audio_hal_ctrl_codec(..., AUDIO_HAL_CTRL_STOP), is already stopped when `ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND` event comes. 

Using **v4.4.3** **with the tick change**, I can see that light sleep is possible, but it's not the same as on Android, the duty cycle is **9/10** vs Android's **1/10**. This is unlikely to net any power savings when compared to Android. Something is preventing it from having a shorter duty cycle even though the ESP32 requests SNIFF mode.

The ESP32 receives the mode change:

Android:
W (116346) BT_HCI: hci cmd send: sniff: hdl 0x81, intv(30 54)
W (116375) BT_HCI: hcif mode change: hdl 0x81, mode 2, intv 54, status 0x0

The key is the HCI SSR (Sniff Sub-Rating) event takes place. After this command the duty cycle returns to 10/100 on Android.
**W (116529) BT_HCI: hcif ssr evt: st 0x0, hdl 0x81, tx_lat 54 rx_lat 1188**

Which is not seen on iOS:
W (43321) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(30 54)
W (43337) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 54, status 0x0
...

Even though the mode change comes, no ssr event comes, and the duty cycle remains 9/10. I couldn't get this to compile on master, it seems ADF is not yet compatible with it yet, so the only version tested is v4.4.3. 

Included is a video of this behaviour.
1) Board is reset, Android is connected.
2) Audio is played, audio is paused.
3) Light sleep resumes, connection remains.

4) Board is reset, iPhone is connected (doesn't seem to matter which, X, 11, 12, 5S)
5) Audio is played, audio is paused.
3) Light sleep is possible, but the duty cycle is very high, there is no HCI SSR event.

[1:38 Video of Problem](https://www.youtube.com/watch?v=VPumqPYr7mI)

BetterJincheng, 2023-02-06:
@expresspotato 
ESP32 has requeseted for a sniff subrating of 22, but iPhone responded with a sniff subrating of 1. So ESP32 should wake to RX every 1*sniff_interval. It seems that the problem is not on the ESP32 side.

ESP32 lmp_sniff_subrating_req
[image]

iPhone lmp_sniff_subrating_res
[image]

Maybe you can increase the sniff interval to save power. However it will increase the latency. In file `bta_api.h`:
```
#define BTA_DM_PM_SNIFF4_MAX     54
#define BTA_DM_PM_SNIFF4_MIN     30
```

Maybe you can have a try.

```
#define BTA_DM_PM_SNIFF4_MAX     540
#define BTA_DM_PM_SNIFF4_MIN     300
```

Alvin1Zhang, 2023-03-22:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2023-03-22 by Alvin1Zhang

expresspotato, 2023-05-29:
Thank you @BetterJincheng, @Alvin1Zhang, the change you suggested worked perfectly 👍