espressif/esp-idf #10170: esp_restart_noos() called randomly

203Null, 2022-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]

esp_restart_noos() is called for some reason after a lot of gpio interrupts.

I called a lot of gpio_isr_handler_remove() and gpio_isr_handler_add within the interrupt routine. The signal comes in on the data pin. After the interrupt, I delete it so it won't trigger during the data transmission. It was readded after the routine. Not sure if this is the cause for esp_restart_noos() is called

中文：
不知道为啥esp_restart_noos()被调用了，导致芯片重启。

代码是为了处理一个芯片的信号，data被芯片上拉，esp中断data的上拉之后，我暂时通过gpio_isr_handler_remove() 把中断取消了，处理完再加上去。 不知道是不是这样的操作造成的问题。

Code: https://gist.github.com/203Null/1c7dbc395294592b4b4c555269355dea

203Null, 2022-11-13:
The version is ESP-IDF V4.4.3

Dazza0, 2022-11-16:
@203Null From the log you posted, it seems like the ESP is getting stuck somewhere inside the `esp_restart_noos()`, and the ESP is eventually getting reset by the RTC watchdog timer. `esp_restart_noos()` is called from the panic handler, which probably means something is causing a panic (e.g., a failed assert, or some runtime error), but the panic handler is getting stuck.

Just to double check, could you provide the list of GPIOs used the touchbar example you've provided?

203Null, 2022-11-16:
Thank you for replying. Usually when the program panic, it will always give me a call stack so I find this really weird. 

TouchData_Pin is GPIO_NUM_47;
TouchClock_Pin is GPIO_NUM_33;
Target is esp32-s3

Dazza0, 2022-11-16:
@203Null  Could you check if your current [ESP32-S3 module is using Octal SPI for its flash/psram](https://www.espressif.com/zh-hans/products/modules)? If so, then `GPIO_NUM_33` would conflict with the Octal SPI bus which might explain why the ESP is getting stuck in a panic.

203Null, 2022-11-17:
@Dazza0 The module I'm using right now is ESP32-S3-Mini-1-N8. It should have a build in flash without octal spi psram. GPIO_NUM_33 should be free to use (My old touch bar driver without interrupt works flawlessly)

Dazza0, 2023-11-21:
@203Null Apologies for the late follow-up.

I misread the log, `rst:0xc (SW_CPU_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)` indicates a software reset (e.g., `abort()` or `assert()` failure). I couldn't recreate your error when running locally. Only error I could find was a stack overflow, thus may double the stack size to `configMINIMAL_STACK_SIZE * 2` will resolve the issues.

If you could provide the full backtrace before the reset occurs, that will help us track down the issue.

I've double checked the `gpio_isr_handler_remove()` and `gpio_isr_handler_add()` functions, and there shouldn't be an issue with the way you are calling them. `gpio_isr_handler_remove()` will clear any pending interrupts before removing the handler.

If you have any more findings, please feel free to reopen this issue.

closed 2023-11-21 by Dazza0